system76/firmware-open #2: Trouble compiling serde

bdelgado1995, 2019-09-07:
I'm getting an error while doing 'cargo install cargo-xbuild' when it tries to compile serde. 

There are a sequence of errors similar to this one below. Ideas?

error[E0412]: cannot find type `NonZeroI8` in module `num`

System is: Ubuntu 18.04 64bit

brian@ubu18:~/system76$ rustc --version
rustc 1.35.0-nightly (9d71ec135 2019-03-10)

Log: 
brian@ubu18:~/system76$ cargo install cargo-xbuild
    Updating crates.io index
  Downloaded cargo-xbuild v0.5.15
  Downloaded 1 crates (24.9 KB) in 1.69s
  Installing cargo-xbuild v0.5.15
  Downloaded cargo_metadata v0.5.8
  Downloaded fs2 v0.4.3
  Downloaded error-chain v0.7.2
  Downloaded serde_derive v1.0.99
  Downloaded tempdir v0.3.7
  Downloaded libc v0.2.62
  Downloaded serde v1.0.99
  Downloaded rustc_version v0.1.7
  Downloaded toml v0.2.1
  Downloaded serde_json v1.0.40
  Downloaded semver v0.9.0
  Downloaded rand v0.4.6
  Downloaded proc-macro2 v1.0.3
  Downloaded walkdir v1.0.7
  Downloaded remove_dir_all v0.5.2
  Downloaded quote v1.0.2
  Downloaded unicode-xid v0.2.0
  Downloaded error-chain v0.11.0
  Downloaded ryu v1.0.0
  Downloaded syn v1.0.5
  Downloaded semver-parser v0.7.0
  Downloaded itoa v0.4.4
  Downloaded rustc-serialize v0.3.24
  Downloaded same-file v0.1.3
  Downloaded semver v0.1.20
   Compiling proc-macro2 v1.0.3
   Compiling unicode-xid v0.2.0
   Compiling syn v1.0.5
   Compiling ryu v1.0.0
   Compiling libc v0.2.62
   Compiling serde v1.0.99
   Compiling semver-parser v0.7.0
   Compiling itoa v0.4.4
   Compiling semver v0.1.20
   Compiling error-chain v0.11.0
   Compiling cargo-xbuild v0.5.15
   Compiling remove_dir_all v0.5.2
   Compiling rustc-serialize v0.3.24
   Compiling same-file v0.1.3
   Compiling error-chain v0.7.2
   Compiling rustc_version v0.1.7
   Compiling walkdir v1.0.7
error[E0412]: cannot find type `NonZeroI8` in module `num`
    --> /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/serde-1.0.99/src/de/impls.rs:2416:5
     |
2416 |     NonZeroI8,
     |     ^^^^^^^^^ help: a struct with a similar name exists: `NonZeroU8`

error[E0412]: cannot find type `NonZeroI16` in module `num`
    --> /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/serde-1.0.99/src/de/impls.rs:2417:5
     |
2417 |     NonZeroI16,
     |     ^^^^^^^^^^ help: a struct with a similar name exists: `NonZeroU16`

error[E0412]: cannot find type `NonZeroI32` in module `num`
    --> /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/serde-1.0.99/src/de/impls.rs:2418:5
     |
2418 |     NonZeroI32,
     |     ^^^^^^^^^^ help: a struct with a similar name exists: `NonZeroU32`

error[E0412]: cannot find type `NonZeroI64` in module `num`
    --> /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/serde-1.0.99/src/de/impls.rs:2419:5
     |
2419 |     NonZeroI64,
     |     ^^^^^^^^^^ help: a struct with a similar name exists: `NonZeroU64`

error[E0412]: cannot find type `NonZeroIsize` in module `num`
    --> /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/serde-1.0.99/src/de/impls.rs:2420:5
     |
2420 |     NonZeroIsize,
     |     ^^^^^^^^^^^^ help: a struct with a similar name exists: `NonZeroUsize`

error[E0412]: cannot find type `NonZeroI128` in module `num`
    --> /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/serde-1.0.99/src/de/impls.rs:2432:9
     |
2432 |         NonZeroI128,
     |         ^^^^^^^^^^^ help: a struct with a similar name exists: `NonZeroU128`

error[E0412]: cannot find type `NonZeroI8` in module `num`
   --> /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/serde-1.0.99/src/ser/impls.rs:493:5
    |
493 |     NonZeroI8,
    |     ^^^^^^^^^ help: a struct with a similar name exists: `NonZeroU8`

error[E0412]: cannot find type `NonZeroI16` in module `num`
   --> /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/serde-1.0.99/src/ser/impls.rs:494:5
    |
494 |     NonZeroI16,
    |     ^^^^^^^^^^ help: a struct with a similar name exists: `NonZeroU16`

error[E0412]: cannot find type `NonZeroI32` in module `num`
   --> /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/serde-1.0.99/src/ser/impls.rs:495:5
    |
495 |     NonZeroI32,
    |     ^^^^^^^^^^ help: a struct with a similar name exists: `NonZeroU32`

error[E0412]: cannot find type `NonZeroI64` in module `num`
   --> /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/serde-1.0.99/src/ser/impls.rs:496:5
    |
496 |     NonZeroI64,
    |     ^^^^^^^^^^ help: a struct with a similar name exists: `NonZeroU64`

error[E0412]: cannot find type `NonZeroIsize` in module `num`
   --> /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/serde-1.0.99/src/ser/impls.rs:497:5
    |
497 |     NonZeroIsize,
    |     ^^^^^^^^^^^^ help: a struct with a similar name exists: `NonZeroUsize`

error[E0412]: cannot find type `NonZeroI128` in module `num`
   --> /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/serde-1.0.99/src/ser/impls.rs:509:9
    |
509 |         NonZeroI128,
    |         ^^^^^^^^^^^ help: a struct with a similar name exists: `NonZeroU128`

   Compiling toml v0.2.1
error: aborting due to 12 previous errors

For more information about this error, try `rustc --explain E0412`.
error: Could not compile `serde`.
warning: build failed, waiting for other jobs to finish...
error: failed to compile `cargo-xbuild v0.5.15`, intermediate artifacts can be found at `/tmp/cargo-installBZUZ1C`

Caused by:
  build failed

seasonedgeek, 2019-09-19:
Try using rustc 1.34.0 instead.

See following url for details...
https://github.com/serde-rs/serde/issues/1574

bdelgado1995, 2019-09-23:
Thanks. Somehow I'm not seeing this error anymore, I set up a new Ubuntu 18.04 VM and even with rustc 1.35.0-nightly, it's able to compile. Seems this issue can be closed.

closed 2019-09-23 by bdelgado1995