system76/firmware-open #6: Handle case of having no bootable disk installed

brs17, 2019-09-25:
On a Darter Pro:

After having installed Pop!_OS 19.04 to a NVMe drive, I removed the drive. Upon booting the system, I was taken to an EFI shell that gives the message that there is nothing to mount. Going into the bios, I still see Pop!_OS 19.04, Linux Boot Manager, and an EFI shell as bootable options.

On proprietary firmware provided by upstream, in a similar situation, the system shows a message: `Default Boot Device Missing or Boot Failed. Insert Recovery Media and Hit any key Then Select 'Boot Manager' to choose a new Boot Device or to Boot Recovery Media`. In the bios menu there are no selectable options from which to boot.

I think it is important in this situation to make it clear to the user that there is no bootable media immediately available.

crawfxrd, 2020-01-29:
Current behavior is that of #53: empty list for "One Boot Time" and "Change Boot Order", "Boot Default" reloads the front page (so no user visible behavior).

With system76/edk2#7, a warning will be printed if no devices are found:

[image]

crawfxrd, 2020-04-20:
Will be resolved with #77.

closed 2020-05-31 by crawfxrd