system76/firmware-open #15: Feature request: Live update "One Time Boot" menu when new device is plugged in

brs17, 2019-09-28:
Similar to the proprietary INSYDE firmware, it would be nice for the "One Time Boot" menu to recognize when a new bootable device is connected/disconnected and update the boot options accordingly.

crawfxrd, 2020-02-14:
The newer edk2 code in `wip/uefipayloadpkg` seems to trigger an event on device insertion/removal which updates this list. Still have to exit and re-enter the form to see the change, though.

crawfxrd, 2020-07-16:
I'm not seeing a way to register a notify function for device topology changes, like `RegisterKeyNotify()` of `EFI_SIMPLE_TEXT_INPUT_EX_PROTOCOL`.

crawfxrd, 2020-10-30:
I've hacked up MdeModulePkg to get this working with DisplayEngineDxe. I'll need to investigate the behavior with firmware-setup.

crawfxrd, 2021-02-08:
Released in with 2021-01-19_d6d37c0.

closed 2021-02-08 by crawfxrd