system76/firmware-open #20: BIOS Menu: Display installed hardware?

brs17, 2019-10-01:
Currently the BIOS menu does not list the hardware currently installed. This is useful when trying to diagnose a hardware issue and unable to boot into an OS.

INSYDE firmware currently lists CPU, memory frequency, and memory size.

WatchMkr, 2019-10-01:
Our goal is extremely light weight and focused firmware - boot and get out of the way. 

Firmware version is a good idea but I don’t think we need or should have anything else in firmware.

For Pop, we’ll likely work to remove the need for a firmware setup screen altogether.

closed 2019-10-01 by brs17

brs17, 2019-10-01:
@WatchMkr the current INSYDE firmware we use does not offer the ability to change the boot order. Do we really need to offer that in our firmware?

WatchMkr, 2019-10-01:
Maybe not. What happens when there are two drives each with an OS. What handles choosing the default option?

brs17, 2019-10-01:
Well the Darter only has one drive option but that aside, default boot options can be configured with `bootctl set-default $ID`.

WatchMkr, 2019-10-01:
There would have to be a OS agnostic method. If a customer installs different operating systems they would not be able to choose the default boot device.