system76/firmware-open #21: BIOS Menu: Extra menu when changing boot order

leviport, 2019-10-01:
After selecting "Change Boot Order" from the main menu, I am taken to a menu with an option called "Change Boot Order", with two other options "Commit Changes and Exit" and "Discard Changes and Exit". To change or view the boot order, I have to hit Enter again, rearrange the boot items, hit Enter once more, then select "Commit Changes and Exit". This menu in between the main menu and the boot order list feels like an unnecessary extra step since I cannot actually change the boot order without going one level deeper into the menu.

crawfxrd, 2020-02-18:
Is the current implementation sufficient to resolve this?

leviport, 2020-02-19:
I think it could still be improved, but it is much better than it was when I opened this issue. I'll close it for now.

closed 2020-02-19 by leviport