system76/firmware-open #22: BIOS Menu: "Esc=Exit" on main menu does nothing

leviport, 2019-10-01:
On the main BIOS menu, an option is shown in the bottom bar that says "Esc=Exit", however hitting the escape key at the main menu does not exit the BIOS. Seeing that prompt makes me expect the escape key to exit the BIOS menu and begin booting normally. To boot normally, I have to select "Boot Default" instead.

crawfxrd, 2020-01-22:
Would the preferred behavior be to not show `Esc=Exit` on the front page, or have Esc do something else (boot default entry)?

brs17, 2020-01-22:
@crawfxrd I like the idea of `Esc` escaping the firmware menu and booting the default boot entry.

closed 2020-01-23 by jackpot51

crawfxrd, 2020-01-23:
This may actually be a problem, depending on user behavior. If they hold or repeatedly press Esc, they'll go into the setup and then immediately exit to the default boot.

jackpot51, 2020-01-23:
That's a good point @crawfxrd.

jackpot51, 2020-01-23:
@brs17 any thoughts? Should we revert the change and instead remove the Esc=Exit on the front page?

brs17, 2020-01-23:
Yeah, @crawfxrd that is a very good point. I think the reason some IBVs get away with it is by having a confirmation screen upon exit. Let's just get rid of `Esc=Exit`.