system76/firmware-open #23: Darter Pro: Variety of issues with bootable USB devices

brs17, 2019-10-01:
The Darter Pro fails to boot from a variety of USB devices. All of the below listed devices were recognized and mounted within the OS on all 3 USB ports.

For reference when referring to specific USB ports on the Darter Pro:
USB port 1: Port closest to the power button
USB port 2: Port next to USB port 1
USB port 3: Port on the other side of the machine

The following are the devices tried. They all had a bootable OS installed. The usb drive was plugged in while the machine was off and then the system was cold booted (checkbox checked if boot was successful): 
- [x] SanDisk Cruzer Blade 8GB
- [ ] SanDisk Cruzer Micro 8GB
    - This USB for some reason has a second "drive" a "SanDisk Cruzer CD Drive" at `/dev/sr0`.
    - Two options are therefore shown in the bios menu, USB Device and USB Device 
    - USB Device goes to a grub shell
    - USB Device 1 fails and goes back to the main BIOS menu. Very quickly an error message is shown (but practically impossible to read without the aid of a camera):

    - [image]

    - It is a little concerning to me that this error mentions pressing a key to continue but I was never given the opportunity to do so.

    - This happened connected to each usb port.

- [ ] SanDisk Ultra USB 3.0 32GB
    - All 3 USB ports booted the drive to a grub shell
- [ ] DataStick Pro by Centon
    - When connected to any of the USB ports, the drive was not recognized in the bios menu
- [ ] ASUS Republic of Gamers giveaway flash drive
    - When plugged in to the USB port 1 or 3, the drive was not recognized in the bios menu
    - When plugged into port 2, the system booted to a GRUB shell
- [ ] USB 3.0 System76 Pop!_OS Flash drive
    - All 3 USB ports booted the drive to a grub shell

This may be similarly related to https://github.com/system76/firmware-open/issues/10

jackpot51, 2019-10-04:
Errors of booting into a GRUB shell appear to only occur when the Pop!_OS recovery partition exists on a drive. This interaction is hopefully something we can solve before launch

jackpot51, 2019-10-07:
May be fixed by https://github.com/system76/edk2/commit/d8207169d2421a375e1d4e4e2182c8842dd0ddc2

brs17, 2019-10-07:
### Still not working
`ASUS Republic of Gamers giveaway flash drive` is now not recognized at all in the BIOS boot menu.
DataStick pro is not recognized.

### Working
M.2 enclosures connected via usb type c or type A now function properly.
USB 3.0 System76 Pop!_OS Flash drive now works.
SanDisk Cruzer Micro 8GB works.
 SanDisk Ultra USB 3.0 32GB

brs17, 2019-10-24:
I am seeing that both the USB 3.0 System76 Pop!_OS Flash drive and M.2 enclosures are having difficulty being recognized as boot devices, wile they finally do get noticed at all, it is quite inconsistent and potentially a cause of some of the issues described here: https://github.com/system76/firmware-open/issues/50

brs17, 2019-11-25:
I have noticed the usb devices are recognized quite consistently when <kbd>Esc</kbd> is pressed when the splash screen appears and not before (from previous machines, I have developed a habit of pressing <kbd>Esc</kbd> as soon as the system begins a reboot/starts booting.

enkidulan, 2020-06-17:
same here with lemur pro, usb 3 devices cannot be booted from.

JanZerebecki, 2020-08-03:
I have a few usb2 sticks and most do not show up as a device in the boot order settings on a galp4.

crawfxrd, 2020-09-22:
Error from edk2 when using a USB 3 device:
```
XhcControlTransfer: error - Device Error, transfer - 100
UsbEnumerateNewDev: failed to set device address - Device Error
ERROR: C40000002:V01011000 I0 C4D1F932-821F-4744-BF06-6D30F7730F8D 99AEF028
```

crawfxrd, 2023-01-17:
This hasn't been updated in a while. Open specific issues when new problems are found.

closed 2023-01-17 by crawfxrd

enkidulan, 2023-01-17:
I think it has been fixed some time ago. At least I can confirm that usb-3 devices can be booted from.