system76/firmware-open #24: Darter Pro: Hotkeys requiring multiple presses

leviport, 2019-10-02:
When using the Fn hotkeys to change volume or display brightness, the key will occasionally require repeated keypresses instead of being holdable. Being able to hold the key down to quickly adjust the volume or screen brightness is much more convenient than having to press the key a bunch of times. @brs17 was seeing this intermittently on the i5 Darter Pro, though I have not encountered it yet.

WatchMkr, 2019-10-02:
I'm experiencing this on me Galago on audio hotkeys. Brightness works but the slider motion is awkward,

WatchMkr, 2019-10-02:
Okay, now that I'm on the right firmware... volume up/down and brightness up/down are both working when I hold down the key but the scale moves awkwardly. 

Interestingly, holding down the volume keys did not work in the proprietary BIOS.

brs17, 2019-10-07:
Have not seen this issue recently. I will reopen this issue if the issue reappears.

closed 2019-10-07 by brs17