system76/firmware-open #29: System freezes when selecting `One Time Boot` from BIOS menu

brs17, 2019-10-03:
With firmware `2019-10-03_01c09f0`, the BIOS menu hangs when trying to select the `One Time Boot` option. A hard poweroff is required to get out of this frozen state.

brs17, 2019-10-03:
> With firmware `2019-10-03_01c09f0`, the BIOS menu hangs when trying to select either the `One Time Boot` or `Change Boot Order` option. A hard poweroff is required to get out of this frozen state.

Edited stating selecting `One Time Boot` or `Change Boot Order` both hang.

brs17, 2019-10-03:
This firmware was flashed by building firmware-open and flashing it using popsicle.

jackpot51, 2019-10-03:
This may have been caused by boot menu pollution which was fixed, but there were leftover entries on the system

I cleared the NVRAM and it appears to be working now.

jackpot51, 2019-10-04:
Related to #27

brs17, 2019-10-04:
Since this issue may not be only caused by #27, I am going to reopen this issue.

jackpot51, 2019-10-08:
Workaround in https://github.com/system76/firmware-update/commit/1db61997f32f394820aef53cf241b3842d4e5144

brs17, 2019-10-08:
Looks good on `2019-10-08_3869d9c`

closed 2019-10-08 by brs17