system76/firmware-open #36: USB devices should not have generic names

brs17, 2019-10-07:
On the current firmware, bootable usb devices are listed as `EFI USB Device`, `EFI USB Device 2`, etc. 

On INSYDE firmware, the drives are listed with unique names/descriptions (eg. SanDisk Cruzer). It is much easier to identify a device this way.

brs17, 2019-10-07:
Should make https://github.com/system76/firmware-open/issues/30 irrelevant.

jackpot51, 2019-10-07:
Probably fixed by https://github.com/system76/edk2/commit/a45f569149ad88145bb49106e8dafd068e7db3a0

brs17, 2019-10-08:
@jackpot51 so now I see full drive names in the one time boot menu. I notice that on proprietary firmware it also provides context such as `Linux Boot Manager` and `EFI Hard Drive` (eg. `Linux Boot Manager (Samsung SSD 960 EVO 250GB)`, `EFI Hard Drive (Samsung SSD 960 EVO 250GB)`). 

Would we want to add more information? I would imagine it would make recognizing devices a bit easier/human friendly. Maybe there's an even better way?

brs17, 2020-01-23:
For whatever reason fix for this seems to have regressed as both Pop!_OS branded generic flash drive and the m.2 enclosure are recognized as `EFI USB Device`

jackpot51, 2020-01-23:
USB devices have never been named, only directly connected NVMe and SATA devices

crawfxrd, 2020-01-24:
I found a commit from @MrChromebox's fork which should resolve this, replacing our `HackDevName()` with `BmBootDescription()` from `UefiBootManagerLib`.

[image]

jackpot51, 2020-01-25:
Woot

closed 2020-01-25 by jackpot51