system76/firmware-open #39: Galago Pro: Resume from suspend fails

brs17, 2019-10-08:
Running `2019-10-07_5d2e96c` (-dirty, because of commented out `"SMMSTORE".to_string(),`). When pressing the power button to resume from suspend, the lights on the machine light up as thought the system is resuming but I see no progress on the screen. I did not let the system sit long enough to see if it eventually does reinitialize, I will investigate first thing in the morning.

brs17, 2019-10-08:
Upon further examination, this issue can be recreated by:
- Boot system
- Reboot system
- Suspend system via <kbd>Fn</kbd>+<kbd>F12</kbd>
- Turn system back on via power button

WatchMkr, 2019-10-22:
I'm on 2019-10-07. The first few resume cycles were fast. Now I'm not getting a long delay before I'm back at GDM. It started after a long lid-closed suspend but just occurred again with a FN+F12 suspend and immediate power on.

brs17, 2019-10-30:
This issue seems to be resolved.

brs17, 2019-10-30:
New issue will be opened for suspend issues we are seeing on first install of OS

closed 2019-10-30 by brs17