system76/firmware-open #51: Boot menu adds new entries on every install

brs17, 2019-10-24:
Every time I install an operating system to the machine, a new efi entry is created.

This is the case if I install the same OS over and over again which results in potentially having numerous boot entries that all have the same text while pointing to nothing.

leviport, 2019-10-24:
I'm noticing the same thing on a Galago, however the behavior is a little different when I am installing Ubuntu 18.04 repeatedly. The Ubuntu entry never gets duplicated, but it will stay in the boot menu even if I wipe that Ubuntu drive and install Pop 19.10 over it.

brs17, 2019-10-31:
This has been resolved with https://github.com/system76/edk2/commit/ab4b7c35a826bd542ff23cc5c19baa0ba7423093 and prior 2 commits.

closed 2019-10-31 by brs17