system76/firmware-open #59: DMI Chassis version is not set

brs17, 2019-11-25:
Currently the DMI information for the chassis-version is `Not Specified`, it should be the model of the product (darp6, galp4,etc).

crawfxrd, 2020-09-11:
Is this value used for anything that would make it worth setting?

Can be done by overriding `smbios_chassis_version()` per board in coreboot.

brs17, 2020-09-11:
Historically we had set this, but I guess it isn't a huge need (since this issue has been open for as long as it has). I'll concede to @jackpot51 / @leviport 's judgement.

leviport, 2020-09-11:
I wonder if the Steam hardware survey looks at DMI info. If so, I could see making our products known in those surveys being a good thing. 

This is what I currently see with lemp9 when I get the hardware survey:
[image]

brs17, 2020-09-11:
Desktops have this information set in the dmi info, correct? Perhaps you could compare with that @leviport ?