system76/firmware-open #63: Feature request: boot galp4 from SD card

mokus0, 2019-12-09:
If someone can outline the overall path to making this work, I could even take a crack at implementing it myself. I’m not familiar with the internals of coreboot yet, but I am comfortable with low-level bare metal systems programming, driver development, etc.

hawkinsw, 2020-02-02:
If we could have this support, that would be great!

hawkinsw, 2020-02-02:
Like @mokus0 , I'd be willing to help implement!

crawfxrd, 2020-02-26:
With coreboot debugging enabled, I see
```
PCI: pci_scan_bus for bus 02
POST: 0x24
PCI: 02:00.0 [10ec/5287] enabled
PCI: 02:00.1 [10ec/8168] enabled
POST: 0x25
```
which corresponds to
```
# lspci
02:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTL8411B PCI Express Card Reader (rev 01)
```

crawfxrd, 2021-03-17:
Driver would need to be added to (possibly written) for edk2.

pr0xidian, 2021-05-18:
I accidentally opened a duplicate feature request (Sorry) I'd also love this feature on my Oryx Pro. I'm not a developer, but if there's anything I can do to help test or provide information I'm happy to!

mdeguzis, 2024-03-15:
Any update on this? This would be a much appreciated feature!