system76/firmware-open #65: Wording for Change Boot Order not clear

jacobgkau, 2019-12-10:
The wording for changing the boot order is a bit convoluted at the moment.

After selecting Change Boot Order twice to get into the re-ordering list, "Esc" is labelled as "Exit Entry" while "Enter" is labelled as "Complete Entry." This option of whether or not to save the changes is redundant with the Change Boot Order screen, which already has options to Commit Changes or Discard Changes.

If the user doesn't want to save their changes, they can select Discard Changes on the next screen, so Esc discarding changes may not be necessary at all. However, even with the current behavior, it would be nice if the Esc description was changed from "Exit" to something more descriptive like "Discard."

crawfxrd, 2019-12-11:
This is because the boot order list is editable. The "Complete Entry" and "Exit Entry" are updating the list in memory, not NVRAM. So they have different meaning from "Commit" and "Discard". I think "Update"/"Cancel" would be provide better context for such editable fields.

system76/firmware-setup#2 should also help with this confusion by keeping the user on the same page and switching focus to the list to modify it. See also: #21.

crawfxrd, 2020-07-24:
Resolved by: system76/edk2#15

Both in-memory and NVRAM mapping are updated when hitting Enter to finish modifying boot order. (Existing logic in BMM prevents updating NVRAM if the list is unchanged.)

closed 2020-07-24 by crawfxrd