system76/firmware-open #66: NVIDIA eGPU not initializing

jacobgkau, 2019-12-11:
Tested on a darp6 (customer experiencing the same issue on a galp4.) When we plug in an NVIDIA GPU in the Akitio Node, we do see it listed in lspci:

```
system76@pop-os:~$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Device 9b41 (rev 02)
06:00.0 VGA compatible controller: NVIDIA Corporation GP108 [GeForce GT 1030] (rev a1)
```

However, nvidia-smi does not see the GPU, and we are not able to load the NVIDIA driver:

```
system76@pop-os:~$ nvidia-smi
NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.
system76@pop-os:~$ sudo modprobe nvidia
modprobe: ERROR: could not insert 'nvidia': No such device
```

dmesg shows this output continuously repeated while the GPU is connected:

```
[  251.993332] nvidia-nvlink: Nvlink Core is being initialized, major device number 237
[  251.993840] NVRM: This PCI I/O region assigned to your NVIDIA device is invalid:
               NVRM: BAR0 is 0M @ 0x0 (PCI:0000:06:00.0)
[  251.993841] NVRM: The system BIOS may have misconfigured your GPU.
[  251.993845] nvidia: probe of 0000:06:00.0 failed with error -1
[  251.993863] NVRM: The NVIDIA probe routine failed for 1 device(s).
[  251.993864] NVRM: None of the NVIDIA devices were initialized.
[  251.993998] nvidia-nvlink: Unregistered the Nvlink Core, major device number 237
```

ahoneybun, 2019-12-20:
From new logs from a Razer Core X Chroma with a Sapphire Radeon Nitro+ RX 590 8GB GPU it may be dealing with the same issue:

`[  143.468521] ---[ end trace 997c68591ecbbf46 ]---
[  143.468937] amdgpu: probe of 0000:06:00.0 failed with error -22
[  143.468951] pci 0000:06:00.1: D0 power state depends on 0000:06:00.0
[  143.469000] snd_hda_intel 0000:06:00.1: enabling device (0000 -> 0002)
[  143.469166] snd_hda_intel 0000:06:00.1: Handle vga_switcheroo audio client
[  143.469168] snd_hda_intel 0000:06:00.1: Force to non-snoop mode
[  143.481490] input: HDA ATI HDMI HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:01.0/0000:04:00.0/0000:05:01.0/0000:06:00.1/sound/card1/input56
[  143.481534] input: HDA ATI HDMI HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:01.0/0000:04:00.0/0000:05:01.0/0000:06:00.1/sound/card1/input57
[  143.481570] input: HDA ATI HDMI HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:01.0/0000:04:00.0/0000:05:01.0/0000:06:00.1/sound/card1/input58
[  143.481600] input: HDA ATI HDMI HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:01.0/0000:04:00.0/0000:05:01.0/0000:06:00.1/sound/card1/input59
[  143.481630] input: HDA ATI HDMI HDMI/DP,pcm=10 as /devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:01.0/0000:04:00.0/0000:05:01.0/0000:06:00.1/sound/card1/input60
[  143.481681] input: HDA ATI HDMI HDMI/DP,pcm=11 as /devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:01.0/0000:04:00.0/0000:05:01.0/0000:06:00.1/sound/card1/input61`

0-alex-0, 2019-12-21:
So I'm the customer with galp4. I was able to get the eGPU working with a Thinkpad and Lubuntu on an USB. Other than hardware change, the BIOS had a configuration part for thunderbolt 3. I disabled it's security and allowed pre OS support for the port. I think coreboot is either blocking the thunderbolt 3 connection due to security or most likely unable to support it in preboot environment since it's pretty new and coreboot BIOS environment only lets me choose boot options

sldavidson, 2019-12-21:
I am the user of the Razer Core X Chroma and AMD RX 590. Like 0-alex-0, I was able to use the same hardware with a Dell XPS 13 running the same Kubuntu version. I had to go into the BIOS settings and change the Thunderbolt security settings before it worked.

jamalex, 2020-01-12:
I've had no luck on a darp6 with coreboot also. I've tried with PopOS 19.10, Ubuntu 18.04, and Ubuntu 19.10. The eGPU (purchased specifically to use with this laptop) is a Sonnet eGFX Breakaway Box 550, containing a MSI GAMING GeForce RTX 2070 8GB (it's been tested plugged into a Windows laptop, and it worked). When I plug it in to the Darter Pro, it spins up, and I'm asked to authorize the TB3 device. When I do, the fans spin down to normal levels as expected. It shows the name of the eGPU box. But it doesn't work, and I get the following in `dmesg`:

```
[ 5331.431094] thunderbolt 0-1: new device found, vendor=0x8 device=0x38
[ 5331.431099] thunderbolt 0-1: Sonnet Technologies, Inc. eGFX Breakaway Box 550
[ 5372.032458] pcieport 0000:02:01.0: pciehp: Slot(1): Card present
[ 5372.032463] pcieport 0000:02:01.0: pciehp: Slot(1): Link Up
[ 5372.373040] pci 0000:04:00.0: [8086:1578] type 01 class 0x060400
[ 5372.373205] pci 0000:04:00.0: enabling Extended Tags
[ 5372.373483] pci 0000:04:00.0: supports D1 D2
[ 5372.373488] pci 0000:04:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[ 5372.373620] pci 0000:04:00.0: 8.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s x4 link at 0000:02:01.0 (capable of 31.504 Gb/s with 8 GT/s x4 link)
[ 5372.373903] pcieport 0000:02:01.0: ASPM: current common clock configuration is broken, reconfiguring
[ 5372.373979] pci 0000:04:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[ 5372.374247] pci 0000:05:01.0: [8086:1578] type 01 class 0x060400
[ 5372.374413] pci 0000:05:01.0: enabling Extended Tags
[ 5372.374673] pci 0000:05:01.0: supports D1 D2
[ 5372.374678] pci 0000:05:01.0: PME# supported from D0 D1 D2 D3hot D3cold
[ 5372.375077] pci 0000:04:00.0: PCI bridge to [bus 05-24]
[ 5372.375099] pci 0000:04:00.0:   bridge window [io  0x0000-0x0fff]
[ 5372.375114] pci 0000:04:00.0:   bridge window [mem 0x00000000-0x000fffff]
[ 5372.375134] pci 0000:04:00.0:   bridge window [mem 0x00000000-0x000fffff 64bit pref]
[ 5372.375143] pci 0000:05:01.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[ 5372.375419] pci 0000:06:00.0: [10de:1f07] type 00 class 0x030000
[ 5372.375517] pci 0000:06:00.0: reg 0x10: [mem 0x00000000-0x00ffffff]
[ 5372.375560] pci 0000:06:00.0: reg 0x14: [mem 0x00000000-0x0fffffff 64bit pref]
[ 5372.375600] pci 0000:06:00.0: reg 0x1c: [mem 0x00000000-0x01ffffff 64bit pref]
[ 5372.375625] pci 0000:06:00.0: reg 0x24: [io  0x0000-0x007f]
[ 5372.375649] pci 0000:06:00.0: reg 0x30: [mem 0x00000000-0x0007ffff pref]
[ 5372.375931] pci 0000:06:00.0: PME# supported from D0 D3hot
[ 5372.376110] pci 0000:06:00.0: 8.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s x4 link at 0000:02:01.0 (capable of 126.016 Gb/s with 8 GT/s x16 link)
[ 5372.376304] pci 0000:06:00.0: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
[ 5372.376316] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[ 5372.376441] pci 0000:06:00.1: [10de:10f9] type 00 class 0x040300
[ 5372.376511] pci 0000:06:00.1: reg 0x10: [mem 0x00000000-0x00003fff]
[ 5372.377117] pci 0000:06:00.2: [10de:1ada] type 00 class 0x0c0330
[ 5372.377201] pci 0000:06:00.2: reg 0x10: [mem 0x00000000-0x0003ffff 64bit pref]
[ 5372.377262] pci 0000:06:00.2: reg 0x1c: [mem 0x00000000-0x0000ffff 64bit pref]
[ 5372.377515] pci 0000:06:00.2: PME# supported from D0 D3hot
[ 5372.377812] pci 0000:06:00.3: [10de:1adb] type 00 class 0x0c8000
[ 5372.377879] pci 0000:06:00.3: reg 0x10: [mem 0x00000000-0x00000fff]
[ 5372.378217] pci 0000:06:00.3: PME# supported from D0 D3hot
[ 5372.378753] pci 0000:05:01.0: PCI bridge to [bus 06-24]
[ 5372.378768] pci 0000:05:01.0:   bridge window [io  0x0000-0x0fff]
[ 5372.378781] pci 0000:05:01.0:   bridge window [mem 0x00000000-0x000fffff]
[ 5372.378795] pci 0000:05:01.0:   bridge window [mem 0x00000000-0x000fffff 64bit pref]
[ 5372.378800] pci_bus 0000:06: busn_res: [bus 06-24] end is updated to 06
[ 5372.378810] pci_bus 0000:05: busn_res: [bus 05-24] end is updated to 06
[ 5372.378837] pci 0000:04:00.0: BAR 15: no space for [mem size 0x18000000 64bit pref]
[ 5372.378840] pci 0000:04:00.0: BAR 15: failed to assign [mem size 0x18000000 64bit pref]
[ 5372.378844] pci 0000:04:00.0: BAR 14: no space for [mem size 0x01800000]
[ 5372.378846] pci 0000:04:00.0: BAR 14: failed to assign [mem size 0x01800000]
[ 5372.378849] pci 0000:04:00.0: BAR 13: assigned [io  0x2000-0x2fff]
[ 5372.378854] pci 0000:05:01.0: BAR 15: no space for [mem size 0x18000000 64bit pref]
[ 5372.378856] pci 0000:05:01.0: BAR 15: failed to assign [mem size 0x18000000 64bit pref]
[ 5372.378859] pci 0000:05:01.0: BAR 14: no space for [mem size 0x01800000]
[ 5372.378862] pci 0000:05:01.0: BAR 14: failed to assign [mem size 0x01800000]
[ 5372.378865] pci 0000:05:01.0: BAR 13: assigned [io  0x2000-0x2fff]
[ 5372.378871] pci 0000:06:00.0: BAR 1: no space for [mem size 0x10000000 64bit pref]
[ 5372.378874] pci 0000:06:00.0: BAR 1: failed to assign [mem size 0x10000000 64bit pref]
[ 5372.378877] pci 0000:06:00.0: BAR 3: no space for [mem size 0x02000000 64bit pref]
[ 5372.378880] pci 0000:06:00.0: BAR 3: failed to assign [mem size 0x02000000 64bit pref]
[ 5372.378882] pci 0000:06:00.0: BAR 0: no space for [mem size 0x01000000]
[ 5372.378885] pci 0000:06:00.0: BAR 0: failed to assign [mem size 0x01000000]
[ 5372.378887] pci 0000:06:00.0: BAR 6: no space for [mem size 0x00080000 pref]
[... cut ...]
[ 5372.456118] R13: 000056371e9d31b0 R14: 0000000000020000 R15: 000056371e9c38b0
[ 5372.456121] ---[ end trace bf511b76f200d1ae ]---
[ 5372.456130] nouveau: probe of 0000:06:00.0 failed with error -12
```

Note: I previously tried installing Nvidia drivers, blacklisting nouveau etc, and the only effects I managed to get were an unbootable system -- couldn't get past the login screen anymore (tried reversing installation of the drivers via recovery, but no luck). That's how I ended up switching from PopOS to Ubuntu. But same issue on both.

jamalex, 2020-01-12:
I was happy my new laptop would be using open source firmware, but this is preventing me from using the eGPU for the work I need to use it for, so at this point I'd very readily flash a standard BIOS on here if it got the eGPU working. Is that a possibility? And is this issue still being looked into?

jamalex, 2020-01-12:
I realized I hadn't installed `system76-driver` after installing Ubuntu 19.10. I did so, and also installed `system76-driver-nvidia`. I still get the same behavior, but `dmesg` now gives:
```
[   25.666612] thunderbolt 0-1: new device found, vendor=0x8 device=0x38
[   25.666615] thunderbolt 0-1: Sonnet Technologies, Inc. eGFX Breakaway Box 550
[   25.762535] pcieport 0000:02:01.0: pciehp: Slot(1): Card present
[   25.762539] pcieport 0000:02:01.0: pciehp: Slot(1): Link Up
[   26.102817] pci 0000:04:00.0: [8086:1578] type 01 class 0x060400
[   26.102948] pci 0000:04:00.0: enabling Extended Tags
[   26.103165] pci 0000:04:00.0: supports D1 D2
[   26.103166] pci 0000:04:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[   26.103280] pci 0000:04:00.0: 8.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s x4 link at 0000:02:01.0 (capable of 31.504 Gb/s with 8 GT/s x4 link)
[   26.103467] pcieport 0000:02:01.0: ASPM: current common clock configuration is broken, reconfiguring
[   26.114369] pci 0000:04:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[   26.114571] pci 0000:05:01.0: [8086:1578] type 01 class 0x060400
[   26.114715] pci 0000:05:01.0: enabling Extended Tags
[   26.114937] pci 0000:05:01.0: supports D1 D2
[   26.114939] pci 0000:05:01.0: PME# supported from D0 D1 D2 D3hot D3cold
[   26.115222] pci 0000:04:00.0: PCI bridge to [bus 05-24]
[   26.115237] pci 0000:04:00.0:   bridge window [io  0x0000-0x0fff]
[   26.115244] pci 0000:04:00.0:   bridge window [mem 0x00000000-0x000fffff]
[   26.115256] pci 0000:04:00.0:   bridge window [mem 0x00000000-0x000fffff 64bit pref]
[   26.115261] pci 0000:05:01.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[   26.115438] pci 0000:06:00.0: [10de:1f07] type 00 class 0x030000
[   26.115520] pci 0000:06:00.0: reg 0x10: [mem 0x00000000-0x00ffffff]
[   26.115553] pci 0000:06:00.0: reg 0x14: [mem 0x00000000-0x0fffffff 64bit pref]
[   26.115585] pci 0000:06:00.0: reg 0x1c: [mem 0x00000000-0x01ffffff 64bit pref]
[   26.115603] pci 0000:06:00.0: reg 0x24: [io  0x0000-0x007f]
[   26.115622] pci 0000:06:00.0: reg 0x30: [mem 0x00000000-0x0007ffff pref]
[   26.115852] pci 0000:06:00.0: PME# supported from D0 D3hot
[   26.116010] pci 0000:06:00.0: 8.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s x4 link at 0000:02:01.0 (capable of 126.016 Gb/s with 8 GT/s x16 link)
[   26.116060] pci 0000:06:00.0: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
[   26.116065] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[   26.116131] pci 0000:06:00.1: [10de:10f9] type 00 class 0x040300
[   26.116184] pci 0000:06:00.1: reg 0x10: [mem 0x00000000-0x00003fff]
[   26.116599] pci 0000:06:00.2: [10de:1ada] type 00 class 0x0c0330
[   26.116665] pci 0000:06:00.2: reg 0x10: [mem 0x00000000-0x0003ffff 64bit pref]
[   26.116714] pci 0000:06:00.2: reg 0x1c: [mem 0x00000000-0x0000ffff 64bit pref]
[   26.116913] pci 0000:06:00.2: PME# supported from D0 D3hot
[   26.117071] pci 0000:06:00.3: [10de:1adb] type 00 class 0x0c8000
[   26.117123] pci 0000:06:00.3: reg 0x10: [mem 0x00000000-0x00000fff]
[   26.117400] pci 0000:06:00.3: PME# supported from D0 D3hot
[   26.117836] pci 0000:05:01.0: PCI bridge to [bus 06-24]
[   26.117851] pci 0000:05:01.0:   bridge window [io  0x0000-0x0fff]
[   26.117860] pci 0000:05:01.0:   bridge window [mem 0x00000000-0x000fffff]
[   26.117874] pci 0000:05:01.0:   bridge window [mem 0x00000000-0x000fffff 64bit pref]
[   26.117879] pci_bus 0000:06: busn_res: [bus 06-24] end is updated to 06
[   26.117888] pci_bus 0000:05: busn_res: [bus 05-24] end is updated to 06
[   26.117914] pci 0000:04:00.0: BAR 15: no space for [mem size 0x18000000 64bit pref]
[   26.117917] pci 0000:04:00.0: BAR 15: failed to assign [mem size 0x18000000 64bit pref]
[   26.117921] pci 0000:04:00.0: BAR 14: no space for [mem size 0x01800000]
[   26.117923] pci 0000:04:00.0: BAR 14: failed to assign [mem size 0x01800000]
[   26.117928] pci 0000:04:00.0: BAR 13: assigned [io  0x2000-0x2fff]
[   26.117933] pci 0000:05:01.0: BAR 15: no space for [mem size 0x18000000 64bit pref]
[   26.117935] pci 0000:05:01.0: BAR 15: failed to assign [mem size 0x18000000 64bit pref]
[   26.117938] pci 0000:05:01.0: BAR 14: no space for [mem size 0x01800000]
[   26.117940] pci 0000:05:01.0: BAR 14: failed to assign [mem size 0x01800000]
[   26.117943] pci 0000:05:01.0: BAR 13: assigned [io  0x2000-0x2fff]
[   26.117951] pci 0000:06:00.0: BAR 1: no space for [mem size 0x10000000 64bit pref]
[   26.117954] pci 0000:06:00.0: BAR 1: failed to assign [mem size 0x10000000 64bit pref]
[   26.117957] pci 0000:06:00.0: BAR 3: no space for [mem size 0x02000000 64bit pref]
[   26.117959] pci 0000:06:00.0: BAR 3: failed to assign [mem size 0x02000000 64bit pref]
[   26.117962] pci 0000:06:00.0: BAR 0: no space for [mem size 0x01000000]
[   26.117964] pci 0000:06:00.0: BAR 0: failed to assign [mem size 0x01000000]
[   26.117967] pci 0000:06:00.0: BAR 6: no space for [mem size 0x00080000 pref]
[   26.117969] pci 0000:06:00.0: BAR 6: failed to assign [mem size 0x00080000 pref]
[   26.117972] pci 0000:06:00.2: BAR 0: no space for [mem size 0x00040000 64bit pref]
[   26.117974] pci 0000:06:00.2: BAR 0: failed to assign [mem size 0x00040000 64bit pref]
[   26.117978] pci 0000:06:00.2: BAR 3: no space for [mem size 0x00010000 64bit pref]
[   26.117980] pci 0000:06:00.2: BAR 3: failed to assign [mem size 0x00010000 64bit pref]
[... cut ...]
[   32.819779] NVRM: This PCI I/O region assigned to your NVIDIA device is invalid:
               NVRM: BAR0 is 0M @ 0x0 (PCI:0000:06:00.0)
[   32.819781] NVRM: The system BIOS may have misconfigured your GPU.
[   32.819788] nvidia: probe of 0000:06:00.0 failed with error -1
[   32.819830] NVRM: The NVIDIA probe routine failed for 1 device(s).
[   32.819831] NVRM: None of the NVIDIA devices were initialized.
[   32.820123] nvidia-nvlink: Unregistered the Nvlink Core, major device number 237
```

jackpot51, 2020-01-13:
We are working on new firmware to address this issue

jackpot51, 2020-01-16:
New firmware will be released within the next week that will allow using NVIDIA GPUs for CUDA and PCIE passthrough. Thunderbolt security is enabled by default and this makes using NVIDIA GPUs for graphics require custom configuration because the graphics stack loads before boltd authorizes the eGPU.

vthg2themax, 2020-01-16:
Does this disable Thunderbolt security, and make me more vulnerable to 
untrusted thunderbolt devices? Or do we get configuration options available 
in coreboot?

tkolo, 2020-01-16:
Is the fix NVIDIA specific?

jackpot51, 2020-01-16:
The only available option is to have Thunderbolt security enabled. The fix was to increase hotplug memory for the Thunderbolt bridge and should not be specific to NVIDIA.

jackpot51, 2020-01-17:
That's right

tkolo, 2020-01-24:
This now works for me perectly after BIOS update. Thanks guys!

zlgonzalez, 2020-01-25:
Hi,
  Can someone tell me what specific egpu model would work now that I've updated my BIOS on the Darter Pro?

Thanks,
Ron

vthg2themax, 2020-01-25:
I have updated the firmware of my coreboot Galago Pro, and cannot get my AMD RX 5700xt in my Sonnet 650 egpu working still. I can add logs if needed, or if you prefer to continue troubleshooting only with support.

tkolo, 2020-01-25:
I have Darter Pro (darp6) with Sonnet 650 and tested with borrowed AMD R9 290X. I've ordered AMD RX 5700XT now, we'll see how it works.

vthg2themax, 2020-01-26:
I also popped in another hard drive and installed Windows 10, and installed the drivers and rebooted. The device in device manager came up with a code 12 leading me to believe that there could still be another issue with the firmware.

vthg2themax, 2020-01-26:
I swapped out the GPU to an and rx560 and it also gives me a code 12 error saying there are not enough resources on the device. I do some web programming for my main job, so if you think this type of problem is something I could contribute to, please feel free to let me know where to start. I really believe in the idea of coreboot, and want to see coreboot achieve the same functionality of the out of box BIOS the device originally had.

tkolo, 2020-01-26:
I'm sorry if I'm asking a stupid question, but did you also test eGPU with different laptop? Perhaps that's what's broken.

vthg2themax, 2020-01-26:
Not a stuipd question at all! Unfortunately, this laptop was my first with eGPU capability, so I do not currently have another to test with. I guess this is why being a trailblazer is so difficult. I will have to see if I can get ahold of another one to use to help troubleshoot this, but it will be a while I imagine.

jackpot51, 2020-01-26:
We have AMD RX 5700XT's to test with, I will try it as well. The fix was to increase the amount of hotplug memory reserved for the Thunderbolt bridge and I found the amount required to run the RTX 2080 Ti, but perhaps AMD GPUs need even more hotplug memory reserved.

jackpot51, 2020-01-26:
I have changed the title to remove `NVIDIA` since that portion is solved. We do not need any customers to investigate AMD graphics card issues as we will investigate them internally.

vthg2themax, 2020-01-26:
Answering on a Sunday? Wow! Thank you for your dedication! I just figured I would offer to help, because this is a public project with positive implications for personal freedom on an end user's device.

tkolo, 2020-01-30:
My radeon 5700xt arrived today and indeed it doesn't work. You guys mentioned you'll test it as well, so I'll just wait.

tkolo, 2020-01-30:
Actually never mind, loading `amdgpu` driver with `vm_update_mode=3` fixed the issue :)

jackpot51, 2020-01-30:
I wonder why that would be required

jackpot51, 2020-01-30:
```
parm:           vm_update_mode:VM update using CPU (0 = never (default except for large BAR(LB)), 1 = Graphics only, 2 = Compute only (default for LB), 3 = Both (int)
```

vthg2themax, 2020-01-30:
Is that a setting in pop os that needs to be modified?

jackpot51, 2020-01-30:
Perhaps, I still have more investigation to do

tkolo, 2020-01-30:
My initial source for this was https://wiki.archlinux.org/index.php/AMDGPU#Freezes_with_%22[drm]_IP_block:gmc_v8_0_is_hung!%22_kernel_error
I didn't get exactly this error but what I had was also about something IP blocks so I decided it's wroth a try.

vthg2themax, 2020-01-31:
The notice about using the CPU to do some of the work instead of the GPU is definitely kind of a downside to this workaround, but at least it works. I just wish I knew how to help with this, I'm not very familiar with SDMA and hotplug memory type of programming pieces. Thanks for posting that tkolo!

vthg2themax, 2020-02-06:
So, I aplogize for being such a novice about this, but since we are using pop os,  we would need to use the kernel stub command for this then, correct?

ahoneybun, 2020-02-06:
Correct. 

sudo kernelstub -a "option"

tkolo, 2020-02-06:
Unfortunately since kernel 5.5 I'm unable to get this model to work at all. There's a null dereference exception somewhere deep within amdgpu's code that I'm not sure how to workaround. Fortunately for me I'm not using amdgpu anymore anyway becasue on 5.4 while it worked, I had a lot of issues with reverse PRIME mode, so instead I opted in for vfifo and passthrough to windows VM, which works perfectly.
I'd report it to kernel developers but truth be told I don't even know where/how.

vthg2themax, 2020-02-07:
Thanks for the update tkolo! Unfortunately for me, I am still stuck with this GPU. Is there a guide for creating my own firmware and loading it to test any changes? I would like to see if perhaps doubling whatever the memory hotplug change was seems to do anything. I really don't want to have to get another GPU, as AMD has non proprietary drivers which is a win for the open source community I think.

vthg2themax, 2020-02-07:
I tried following some guides to hooking up the egpu to a windows vm, would you be willing share some tips on how you did it with this laptop with coreboot firmware?

tkolo, 2020-02-07:
Took me longer than expected, but there you go https://gist.github.com/tkolo/b1f24130efec706bc66205badbca85cc

vthg2themax, 2020-02-07:
Thanks tkolo, I will test it out and report back on my results!

vthg2themax, 2020-02-10:
> My initial source for this was [...]

I did sudo kernelstub -a "amdgpu.vm_update_mode=3" and rebooted, and I was still getting the error message in 19.10 pop_os dmesg log. Was that what you were getting even when it worked?

vthg2themax, 2020-02-11:
I am still working through your excellent guide @tkolo , and am posting revisions here:
https://gist.github.com/vthg2themax/893319aec795ff6b242d19c8acb07410
Please let me know if I was supposed to help edit your gist file instead. I am not certain of the original intent.

tkolo, 2020-02-11:
I only wrote it as gist because I feel comfortable with markdown and it was the first markdown-enabled paste service that came to my mind. Feel free to maintain it though if you want to re-use it somewhere else.

vthg2themax, 2020-02-11:
I shall leave it out there, and maintain it as best as I can. If it would be of use to anyone else on the web, it will be worth preserving. System76 / Pop_OS! seems like some good people, but did not anticipate this type of usage when they starting working on the coreboot, I suppose? Either way perhaps @jackpot51 could shed some light on whether the in-progress guide could be put somewhere on their site, or if they are already working on something like this.

tkolo, 2020-02-11:
Also answering your earlier question (sorry, I forgot about it) actually neither putting `vm_update_mode` in kernel parameters nor in modprobe.d worked for me, I always had to manually unload `amdgpu` with `rmmod` and reload it again with `modprobe amdgpu vm_update_mode=3`. Not sure if that behavior was because of my misconfiguration or some deeper driver issue, I don't really care ever since I switched to vfio

vthg2themax, 2020-03-09:
@jackpot51 Can you comment on how things are going with this?

jackpot51, 2020-03-17:
I have not figured out why AMD GPUs are not working while NVIDIA GPUs are. It doesn't appear that increasing the reserved memory matters.

vthg2themax, 2020-03-17:
Do you know who I could ask around to try to get this working? Or anything I could look into to help out with figuring this out?

jackpot51, 2020-03-17:
There may be a maintainer listed for the amdgpu module who would be able to provide more information on the error

vthg2themax, 2020-03-26:
After contacting the amazingly helpful, and prompt,  folks with the amdgpu driver, they had me test the card on a standard desktop, and with a newer kernel to see any differences. The card worked fine on a standard dekstop, however on the newer kernel still had an error. The error message specifically mentions corruption and the helpful hint they have given is the following statement:
'The discovery data is region of vram that is populated at device power up by a micro-controller on the GPU.  It contains information about the device that the driver needs to load properly.'
So, I think that means that this issue will need to be solved in coreboot. I will read more about how to get setup with coreboot, but wanted to know of there is anything special I need to debug and deploy coreboot onto my laptop? I would like to make this firmware better. @jackpot51

vthg2themax, 2020-05-10:
vfio now appears to be broken on 20.04 Pop_OS. :-( @tkolo Just a warning friend.

alexr4535, 2020-09-27:
I also have a Razer Core and AMD RX 580, i am not able to get this working. on Pop_OS 20.04...
`lspci -k | grep -EA3 'VGA|3D|Display'` shows both i915 (intel graphics) and amdgpu kernel drivers in use. 
but actual performance is not good. lots of screen tearing, very slow, and ~13 fps on gaming benchmarks.

kocoman2, 2022-12-16:
can anyone please tell me what code is the "The fix was to increase the amount of hotplug memory reserved for the Thunderbolt bridge" in? thank you

crawfxrd, 2022-12-19:
The coreboot configs `PCIEXP_HOTPLUG_MEM` and `PCIEXP_HOTPLUG_PREFETCH_MEM`.

antonkulaga, 2023-01-16:
Guys, eGPU does not work with my lemur pro (11 addition) with Razer Core and Nvidia RTX3060. I tried pretty much everything I found on the internet. Has anybody managed to make it work? I bought lemur specifically to bundle it with egpu when I need to do deep-learning and safe charge otherwise, but it does not work :((((

jacobgkau, 2023-07-20:
lemp12 isn't initializing an RTX GPU in an Akitio Node Titan. dmesg output: [lemp12-egpu.txt](https://github.com/system76/firmware-open/files/12114651/lemp12-egpu.txt) (Just trying to plug it in after boot in order to use it for CUDA. `nvidia-smi` is never showing it.)

A lemp12 customer with a Razer Core X and a GTX 1080 Ti is seeing the GPU show up in `nvidia-smi` at first, but it drops off after trying to use it for anything, sometimes just after having run `nvidia-smi`. Mattermost thread: https://chat.pop-os.org/pop-os/pl/1db8qbbhkb8n3pdrryx1r3bfcc

rspOverflow, 2025-10-06:
So, I have a Razer Core X V2 and a AMD RX 7900XTX. I have connected it to my darp10 via Thunderbolt 4 and my system is having three problems:

1. It hangs on boot or reboot when I'm connected to the eGPU. Then after power cycle, it kicks me into the GRUB boot screen where I have to manually select to boot into Ubuntu (fyi: My Ubuntu version is 24.04.3 LTS).

2. Both `pci` and `pcieport` fail to "assign bridge window space"

3. amdgpu detects the failed assignment then tries something with the GPU's SMU (OverridePcieParameters?) and dies

luarpy, 2026-01-21:
Hi again,
I have tried to diagnose the source of the problem. I wanted to know if it was a problem with PopOS, other distributions, the firmware installed on my laptop, the external eGPU card, or the cable that came with it.

I tried launching the Mini Windows OS from the Medicat tool, but the same problem seems to keep recurring. There is no apparent detection of the device in the [Mini Windows live OS](https://medicatusb.com/docs/tools/live-operating-systems/mini-windows/). It could be my external PCIe card, but I do not have too much time to try until returning it.