system76/firmware-open #72: Incorrect GalagoPro 3 C has incorrect chip name in chip.txt

asahaf, 2020-02-24:
I was successfully able to flash my GalagoPro 3 C using Raspberry Pi. However, I had 5 hours of struggling trying to flash the built rom. spipi.sh kept telling me, chip not found. After digging and going though the script, I discovered that the chip name for **galp3-c** in _models/galp3-c/chip.txt_ is "**MX25L12805D**" where my chip name is "**GD25Q128C**". I had to change that in _models/galp3-c/chip.txt_ to be able to flash the rom.

Thanks,

jackpot51, 2020-02-24:
What processor do you have?

asahaf, 2020-02-24:
Intel(R) Core(TM) i7-8565U CPU @ 1.80GHz

Base Board Information
Manufacturer: System76
Product Name: Galago Pro
Version: galp3-c
Asset Tag: Not Specified

jackpot51, 2020-02-26:
The chip I believe can be either on galp3-c. The one we have here has the MX25L12805D chip. I will see if we can match either

asahaf, 2020-02-26:
Thanks Jeremy
I bought mine a week before Galago Pro 4 was announced. I believe the one I got was from the last batch, I can't tell from the serial number though.

crawfxrd, 2022-01-31:
The lemp10 and galp5 I have also have different chips that the original models that were used for `chip.txt`. May need to list all possible values in `chip.txt` and loop through them until flashrom detects the correct one.