system76/firmware-open #74: recent kernel upgrade breaks wifi on galp4

igalic, 2020-03-16:
After updating my kernel today from regular package upgrades to `5.3.0-7642-generic`, wifi can no longer be initialized.
I've upgraded the driver as well, by installing `backport-iwlwifi-dkms` with the same result.
Finally, i upgraded the firmware, [directly from Intel](https://www.intel.com/content/www/us/en/support/articles/000005511/network-and-i-o/wireless-networking.html), and it's still failing with the same error:

```
Mar 16 20:18:30 76ix kernel: iwlwifi 0000:00:14.3: TLV_FW_FSEQ_VERSION: FSEQ Version: 43.2.23.17
Mar 16 20:18:30 76ix kernel: iwlwifi 0000:00:14.3: loaded firmware version 48.4fa0041f.0 op_mode iwlmvm
Mar 16 20:18:30 76ix kernel: iwlwifi 0000:00:14.3: Detected Intel(R) Wireless-AC 9560, REV=0x354
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: Collecting data: trigger 15 fired.
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: Start IWL Error Log Dump:
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: Status: 0x00000000, count: -126140537
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: Loaded firmware version: 48.4fa0041f.0
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x3FBFCEBF | ADVANCED_SYSASSERT          
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xD6543CA1 | trm_hw_status0
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xD3693967 | trm_hw_status1
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xFDFAEDD8 | branchlink2
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xBEFDA0B2 | interruptlink1
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x004EBA44 | interruptlink2
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x3C575D01 | data1
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xED6EE2F9 | data2
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x6AAB0C97 | data3
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x4B14F44C | beacon time
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x7AC2E479 | tsf low
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xFD1F3FFA | tsf hi
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xB817DCDE | time gp1
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x8DDE7037 | time gp2
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x6328001C | uCode revision type
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xDEFEA9AE | uCode version major
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x194D7501 | uCode version minor
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x05556DFE | hw version
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x2D4A01B5 | board version
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x6E4890C4 | hcmd
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x2FBEE3F6 | isr0
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xBCC8697E | isr1
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x7179169D | isr2
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x9A8017E2 | isr3
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xDEACC9E3 | isr4
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x1BF0C63D | last cmd Id
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x4014CADB | wait_event
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x0E32A403 | l2p_control
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xFEB7ABFF | l2p_duration
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xB7AE7BDB | l2p_mhvalid
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x9C2F1E61 | l2p_addr_match
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x48A3FAE0 | lmpm_pmg_sel
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xFF877756 | timestamp
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xA95AFD4B | flow_handler
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: Start IWL Error Log Dump:
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: Status: 0x00000000, count: 7
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x201013F1 | ADVANCED_SYSASSERT
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x00000000 | umac branchlink1
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xC008CF5C | umac branchlink2
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x00000000 | umac interruptlink1
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x00000000 | umac interruptlink2
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x00000003 | umac data1
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x20000302 | umac data2
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x01300202 | umac data3
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x00000030 | umac major
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x4FA0041F | umac minor
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x00005CB0 | frame pointer
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xC0887F58 | stack pointer
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x00000000 | last host cmd
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x00000000 | isr status reg
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: Fseq Registers:
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x00000003 | FSEQ_ERROR_CODE
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x00260000 | FSEQ_TOP_INIT_VERSION
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x00000000 | FSEQ_CNVIO_INIT_VERSION
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x0000A384 | FSEQ_OTP_VERSION
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x940C8E60 | FSEQ_TOP_CONTENT_VERSION
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x4552414E | FSEQ_ALIVE_TOKEN
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x20000302 | FSEQ_CNVI_ID
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x01300202 | FSEQ_CNVR_ID
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x20000302 | CNVI_AUX_MISC_CHIP
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x01300202 | CNVR_AUX_MISC_CHIP
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0x0000485B | CNVR_SCU_SD_REGS_SD_REG_DIG_DCDC_VTRIM
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: 0xA5A5A5A2 | CNVR_SCU_SD_REGS_SD_REG_ACTIVE_VDIG_MIRROR
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: SecBoot CPU1 Status: 0x5c1d, CPU2 Status: 0x3
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: Failed to start RT ucode: -110
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: Firmware not running - cannot dump error
Mar 16 20:18:31 76ix kernel: iwlwifi 0000:00:14.3: Failed to run INIT ucode: -110
```

This makes it seem like it's a lower level.

igalic, 2020-03-18:
and, who would've thought, getting a kernel update from system76 ppa fixed it.

Is there a way to pin that and only have kernel updates come from there, not ubuntu?

jackpot51, 2020-03-18:
I feel you have Pop!_OS or the system76-driver installed you will always get our kernel from now on

closed 2020-03-18 by jackpot51

igalic, 2020-03-19:
should i have opened this issue in https://github.com/pop-os/system76-driver ?