system76/firmware-open #78: Enhancement Request: Fn key toggle in Firmware/BIOs Options

mtdeguzis, 2020-04-21:
Much like working on my office Macbook Pro and previously with some other laptops, I would like the option to toggle the Fn key functions. This could be toggled then in Pop!_OS possibly. I often find myself using the functions of the keys instead of F1, F2, and so forth.

phs, 2020-05-15:
I came to request exactly this, though I would hope it works in stock ubuntu as well.  Either a toggle in the firmware menu or a e.g. `fntoggle` to poke somewhere under `/sys` would be fine by me.

curiousercreative, 2021-01-19:
The above mentioned PR would be a good solution to this, but in the meantime I've found it quite easy to edit the keymap and flash yourself. You can follow the [instructions here](https://github.com/system76/ec/blob/master/doc/keyboard-layout-customization.md) and just invert the top row of keys in your custom layout. I've done this for myself along with a couple of minor layout tweaks on my galp5, you can see [my layout here](https://github.com/curiousercreative/ec/blob/fans/src/board/system76/galp5/keymap/apple.c)

vbatts, 2021-02-09:
yes please. I've just logged a system76 support issue about this as well.
I use all the Fn keys at the top, more than the F1, F2, ... keys. I very much wish to FnLock them to just be the volume/brightness, etc.
On the Lemur Pro, this would work seamlessly, except for the ~/` key, which i use more than play/pause. Also the Right Ctrl key, as a menu key?
Just FnLock the F* keys at the top. 

a common FnLock toggle is Fn+Esc, which would not conflict with anything.

rellimmot, 2021-10-06:
+1 Please add this feature with some level higher of priority. `Fn+Esc` would be great. This is a big annoyance.

crawfxrd, 2021-10-06:
`Fn+Esc` is already used to reset the EC config.

crawfxrd, 2021-11-13:
I hacked together a custom `K_FNLOCK` key in system76/ec@66282b940ca2 as an example. It's only implemented for the 15in_102 keyboard. On the default keymap `Fn+CapsLock` will behave as FnLock.

jacobgkau, 2021-12-16:
> @crawfxrd remaining value? I can't wait for fn-lock. Please ship it :-D

I think the question was what value Fn-lock would provide over simply remapping your function keys in the Keyboard Configurator app so the two layers are reversed. Can you answer that question?

mdeguzis, 2021-12-21:
It is much easier to provide a checkbox/toggle that flip flops the FN/function keys. It probably isn't even that much work to save a flipped configuration to a system file JSON or wherever it is stored, then load/unload the default/non-default. What If I change my mind? Like OSX, it's a better customer experience to toggle them, vs making someone painstakingly click on each button and put it back. Heck, I am not even sure I saw "reset keys" button in Keyboard Configurator?

crawfxrd, 2021-12-21:
> It is much easier to provide a checkbox/toggle that flip flops the FN/function keys.

It's easier to add a custom key for it than a separate config. With [`wip/fnlock-custom-key`](https://github.com/system76/ec/commits/wip/fnlock-custom-key) it's part of the keyboard layout data (allowing users to set it to any key they want) and OS-independent (allowing users to enable/disable it on any distro without requiring a special tool for it).

> It probably isn't even that much work to save a flipped configuration to a system file JSON or wherever it is stored

This is in firmware. There is no OS and no file system.

> What If I change my mind?

_Why_ would you change your mind though? If you make the decision that you use the alternate functions so much more that you would have them be the default, why would you need to change them back? What scenarios require regularly switching between F1-F12 and the alt functions?

ids1024, 2021-12-21:
I considered that it would be useful to have an option for this in Keyboard Configurator (https://github.com/pop-os/keyboard-configurator/issues/24) though I haven't gotten around to implementing it.

Mostly I'm just not sure how best to expose it in the UI, but I guess anything is better than nothing. It could just have a menu option to "Swap F keys" (not sure the best wording?) alongside import/export/reset layout.

@mdeguzis I guess that's more what you have in mind? A single option like macOS, though as I understand it the way macOS implements it is a bit strange since they basically violate the USB HID spec to make this controllable by the OS rather than firmware.

curiousercreative, 2021-12-21:
+1 for configurator menu option

mdeguzis, 2021-12-22:
@ids1024 , yes, "Invert Function Keys" may make sense. I exported my config in both cases in case I want to revert later on, or if I mess things up with selections.

bitdivine, 2022-10-22:
@crawfxrd 
> What scenarios require regularly switching between F1-F12 and the alt functions?
I typically have a gui on tty4 and a bunch of plain raw terminals.  Switching to a terminal uses a f[1-9] key, but when in the gui I use buttons such as volume up and volume down.  On my thinkpad the fn-lock gets used quite a bit.

Maybe the entire keyboard should be made replaceable on System76 systems.  The entire numpad mallarkey on the right may be useful for some but for me it just makes the keyboard off-centre, which makes for uncomfortable typing, and none of the numpad keys do things I actually need, such a num-lock.

jacobgkau, 2023-01-30:
> Maybe the keyboard should be made replaceable on System76 systems.

The keyboards are replaceable on almost all System76 systems: https://tech-docs.system76.com/ 

What you are asking for is for multiple models of keyboard compatible with each (15"+) system to be produced and offered, which is much different.

closed 2023-08-01 by jackpot51

vbatts, 2023-08-02:
> Closed #78 <https://github.com/system76/firmware-open/issues/78> as completed via system76/ec#263 <https://github.com/system76/ec/pull/263>.
> You are receiving this email because you commented on the thread.

Super. I would love to see this come to the lemp9!

crawfxrd, 2023-08-02:
A FnLock key has been added to the EC. But the decision was not to add it to the default keymaps, because there's no indicator on the laptops for it.

Once the firmware is released, the Keyboard Configurator will be updated to be able to add it to the custom keyboard layout.

PR #444 (2023-08-01): Update EC and coreboot
Brings in the changes to resolve:

- Resolves: #78
- Resolves: #393
- Resolves: #424