system76/firmware-open #80: galp4: Sometimes fails to power off

crawfxrd, 2020-05-16:
Sometimes on shutdown of a galp3-c or galp4, the board does not power off. The screen is left on and has the following displayed on the console:

```
sd-umount[14108]: Failed to unmount /oldroot: Device or resource busy
sd-umount[14109]: Failed to unmount /oldroot/dev/pts: Device or resource busy
sd-umount[14110]: Failed to unmount /oldroot/dev: Device or resource busy
shudown[1]: failed to finalize file systems, ignoring
reboot: Power down
```

thomas-zimmerman, 2020-06-24:
Customer has now seen this in the wild (47466) on a galp4 system (  i7-10510U (2x16GB) 1 TB Samsung 970 EVO Plus ).

thomas-zimmerman, 2021-02-23:
I have another customer that has been able to somewhat recreate the dead EC (53162): 
```whenever I close the laptop screen and the laptop goes into sleep mode without shutting down, the touchpad and webcam lose function when I re-open the device and the log-in screen appears. In order to regain function, I have to hard-cycle a restart using the power button (turning the computer off and on).```

This would be a failure to suspend that requires a hard reboot. i7-10510U, 64GB memory, Samsung 970 EVO Plus, should be on "2020-05-20_de5eab8" bios but I don't have verification of that (system was factory imaged January 2020).