system76/firmware-open #86: Test updates to enable i2c-hid and above 4GB allocation

jackpot51, 2020-06-04:
Please build and flash the latest version of firmware-open. Then run the following tests:

- `dmesg -l err` is empty
- `dmesg | grep i2c` shows messages about a i2c_hid touchpad being added, successfully
- `lspci -vvnn | grep '\[size=.*M\]' shows prefetchable memory behind bridge allocated above 4 GB (greater than 0x100000000)

For the following models:

- darp5
- darp6
- galp3-c
- galp4
- lemp9

jackpot51, 2020-06-04:
I have tested on darp6 and lemp9. Touchpad feels more responsive after updates to darp6.

crawfxrd, 2020-06-04:
Touchpad issue on galp3-c:

```
[galp3-c ~]$ dmesg | grep i2c
[    0.888824] i2c /dev entries driver
[    1.019675] i2c_hid i2c-PNP0C50:00: i2c-PNP0C50:00 supply vdd not found, using dummy regulator
[    1.019685] i2c_hid i2c-PNP0C50:00: i2c-PNP0C50:00 supply vddl not found, using dummy regulator
[    1.020302] i2c_designware i2c_designware.0: i2c_dw_handle_tx_abort: lost arbitration
[    1.020977] i2c_designware i2c_designware.0: i2c_dw_handle_tx_abort: lost arbitration
[    1.021565] i2c_designware i2c_designware.0: i2c_dw_handle_tx_abort: lost arbitration
[    1.022128] i2c_designware i2c_designware.0: i2c_dw_handle_tx_abort: lost arbitration
[    3.094810] psmouse serio1: synaptics: Your touchpad (PNP: PNP0f13) says it can support a different bus. If i2c-hid and hid-rmi are not used, you might want to try setting psmouse.synaptics_intertouch to 1 and report this to linux-input@vger.kernel.org.
```

Touchpad continues to function.

---

coreboot allocations above 4GB working.

```
Prefetchable memory behind bridge: 000000046e800000-000000046e9fffff [size=2M]
```

jackpot51, 2020-06-04:
So, i2c_hid is not working on galp3-c. I will look into it

jackpot51, 2020-06-04:
I found that i2c_hid is not supported on galp3-c or galp4. I have disabled it there. Please test with commit 7c0aa57

crawfxrd, 2020-06-05:
i2c-hid is now disabled in galp3-c, so behavior is unchanged. coreboot allocator v4 appears to be working.

I don't have a TB device to test with. I have a Thunderbolt panel under Privacy in g-c-c, but it claims "No Thunderbolt Support" (my oryp5 also claims the same thing). (Are these USB-C ports TB compatible?)

closed 2020-08-25 by jackpot51