system76/firmware-open #87: lemp9 - USB-C HDMI output from hub not consistent

jwhittem, 2020-06-05:
Having issues with HDMI via an Insignia USB-C Hub consistently.  Will comment with my open case with System76.com.

jwhittem, 2020-06-05:
I'm having issues getting consistent HDMI output from my USB-C hub, my work macbook works pretty well however.

Model is insignia NS-PUCHUB219

Here is some output from lsusb, lspci that might be helpful:

Bus 002 Device 007: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 002 Device 006: ID 05e3:0749 Genesys Logic, Inc.
Bus 002 Device 005: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04f2:b685 Chicony Electronics Co., Ltd Chicony USB2.0 Camera
Bus 001 Device 011: ID 2109:8888 VIA Labs, Inc.
Bus 001 Device 010: ID 1532:005c Razer USA, Ltd
Bus 001 Device 009: ID 1532:0214 Razer USA, Ltd BlackWidow Ultimate 2016
Bus 001 Device 008: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub
Bus 001 Device 004: ID 8087:0aaa Intel Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
00:00.0 Host bridge: Intel Corporation Device 9b61 (rev 0c)
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics (rev 02)
00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 0c)
00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model
00:12.0 Signal processing controller: Intel Corporation Comet Lake Thermal Subsytem
00:14.0 USB controller: Intel Corporation Device 02ed
00:14.2 RAM memory: Intel Corporation Device 02ef
00:14.3 Network controller: Intel Corporation Wireless-AC 9462
00:15.0 Serial bus controller [0c80]: Intel Corporation Serial IO I2C Host Controller
00:19.0 Serial bus controller [0c80]: Intel Corporation Device 02c5
00:19.2 Communication controller: Intel Corporation Device 02c7
00:1c.0 PCI bridge: Intel Corporation Device 02bd (rev f0)
00:1d.0 PCI bridge: Intel Corporation Device 02b0 (rev f0)
00:1f.0 ISA bridge: Intel Corporation Device 0284
00:1f.3 Audio device: Intel Corporation Device 02c8
00:1f.4 SMBus: Intel Corporation Device 02a3
00:1f.5 Serial bus controller [0c80]: Intel Corporation Comet Lake SPI (flash) Controller
01:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS522A PCI Express Card Reader (rev 01)
02:00.0 Non-Volatile memory controller: Sandisk Corp Device 5009 (rev 01)

jwhittem, 2020-06-06:
not consistently having an issue now, may reopen...

jwhittem, 2020-06-13:
Still having this issue.

metaphys, 2020-07-29:
I’m also having this issue

jwhittem, 2020-09-18:
This issue seems to resolve if I remove power from the USB-C hub for a few seconds, and reboot it.  Not 100%, but 85%+ of the time that seems to work.

metaphys, 2020-10-22:
it seems that my issue was a defective usb c hub. sorry :/

metaphys, 2020-12-03:
well after buying a new usb hub, it seems that the issue is still there, sorry again :/

Dygear, 2020-12-16:
@metaphys Was the hub a different make or model? It could be bad firmware on the hub. Hard to say for sure without testing across multiple models.

metaphys, 2021-01-12:
Indeed same model

metaphys, 2021-03-22:
trying with another model I had no issue

closed 2021-03-23 by jwhittem