system76/firmware-open #88: Integrated Nvidia support?

cognifloyd, 2020-06-05:
Is there somewhere I can follow the progress towards supporting NVIDIA with system76's coreboot / firmware-open?

[This blog post](https://blog.system76.com/post/186655523269/open-firmware-and-more-news-from-july) says work to support Nvidia is one of the next steps:
> The open firmware is now functional on the Gazelle when running on Intel graphics. This will not yet be integrated, however, as more work is necessary to get the NVIDIA graphics up and running.

I'm particularly interested in laptops with 17" screens, all of which have NVidia:

- serw11 (Serval WS 17")
- oryp5 (Oryx Pro 17")
- gaze15 (Gazelle 17")

(You might consider this issue a feature request. Otherwise sorry for posting this question here. The DM response to [my query on twitter](https://twitter.com/cognifloyd/status/1267533992791879681) was "We do not offer any Coreboot options with a 17" screen." and the NVIDIA work is "still in the development stage and we do not have an exact timeline at this point." I understand development takes time, but I'd love to watch the progress.)

jackpot51, 2020-06-06:
You can track the addw2 progress here: https://github.com/system76/firmware-open/pull/84

It is an NVIDIA model and we are working to support it with coreboot, firmware-open, and system76 ec. It is the first of many NVIDIA models which will support it. The most likely one that will be 17 inch is the gaze15.

cognifloyd, 2020-06-08:
Awesome. Thank you! I will close this and watch #84 then. :)

closed 2020-06-08 by cognifloyd