system76/firmware-open #89: lemp9 windows support

thomas-zimmerman, 2020-06-11:
Customer would like to have the following support while in windows10:

- [ ] Brightness display working from function keys

- [ ] Enable/Disable touch pad function key

- [ ] Battery charging status change when AC power plugged/unplugged

jackpot51, 2020-06-11:
I'm not sure how to enable the touchpad function key without writing windows code

jackpot51, 2020-06-11:
Fixed the other two, and prepared firmware updates

ahoneybun, 2020-07-13:
Any update on this firmware update now that it was reverted? Are we working on this in a new firmware update?

leviport, 2020-07-13:
We're still getting to the bottom of it. It's a tricky one.

ahoneybun, 2020-07-27:
Anything new on this one? Thank you.

closed 2020-08-10 by jackpot51

jackpot51, 2020-08-10:
The firmware has been released