system76/firmware-open #90: Targetting different models

Moxinilian, 2020-06-11:
Hello!

I've been wanting to tinker with BIOS stuff for a while now, and I have a Clevo L141CU laptop so I wanted to see if I could flash coreboot and System76 stuff on it as it looks really cool. I know this unit is very close to a Lemur Pro, so I wanted to ask before I order the flashing hardware how portable the firmware is, and if it isn't working out of the box if it is reasonable for a somewhat savvy end user to try and configure things themselves.

Thanks in advance!

crawfxrd, 2020-06-12:
The L141CU and lemp9 (L140CU) are compatible. It won't receive firmware updates (and firmware-update doesn't have the model whitelisted, so it would fail).

Ubuntu 20.04 can be used for the Pi SPI programmer. I'd expect a Raspberry Pi 4 to work too, but haven't tried one. Our docs will need to be updated.

If you're planning on flashing the EC, the Mega2560 *must* be compatible with the official Arduino model. Some models claim to be compatible, but use a different USB chip than the Arduino model (typically, CH340G instead of the ATmega16U2).

Moxinilian, 2020-06-12:
Thank for your answer. Indeed, after thinking about the update cycle, and considering I only wanted to do this to have Rust stuff in the firmware, I’m probably going to hold off for a bit. Thank you very much for your great work however.

closed 2020-06-12 by Moxinilian

c4software, 2020-09-23:
> The L141CU and lemp9 (L140CU) are compatible. It won't receive firmware updates (and firmware-update doesn't have the model whitelisted, so it would fail). For the lemp9 specifically, attempting to flash using the `./scripts/flash.sh` script will also attempt to flash the EC.
> 
> Ubuntu 20.04 can be used for the Pi SPI programmer. I'd expect a Raspberry Pi 4 to work too, but haven't tried one. Our docs will need to be updated.
> 
> If you're planning on flashing the EC, the Mega2560 _must_ be compatible with the official Arduino model. Some models claim to be compatible, but use a different USB chip than the Arduino model (typically, CH340G instead of the ATmega16U2).

Hi

Since it's from the same device, instead of opening a new issue, I'm asking here. I have a L140CU from clevo. 

Your explanation seems clear but I have two questions : 

- Flashing this device from efi is possible ? Or it's hardware locked ?
- The flashing guide have a lot of details, but no picture. As I see online I'm not the only one who want Coreboot on this device. Do you plane to make a detailed guide per device?

Thanks a lot

crawfxrd, 2020-09-23:
> Flashing this device from efi is possible ? Or it's hardware locked ?

It's possible if you have the correct binaries. firmware-open doesn't have the EFI application for flashing from the proprietary BIOS to coreboot though.

> The flashing guide have a lot of details, but no picture.

Of what? Pictures don't look like they would be useful for the information in the flashing doc.

> As I see online I'm not the only one who want Coreboot on this device.

Indeed. Upstream is working on supporting the bare Clevo model, which will also support the TUXEDO InfinityBook ([CB:45624](https://review.coreboot.org/c/coreboot/+/45624)).

> Do you plane to make a detailed guide per device?

No. The only useful per-board information would be noting the location of the flash chip.