system76/firmware-open #95: Error during build/x86_64-efi-pe/efi.img.tmp creation (darp6 model)

neox95, 2020-06-18:
I got that : 
[image]

mkfs.vfat command is not found. Perhaps sudo is needed ?

crawfxrd, 2020-06-18:
`dosfstools` needs to be installed. Have you run `scripts/deps.sh`?

neox95, 2020-06-18:
I have dosfstools installed and I use mkfs.vfat quite often

neox95, 2020-06-18:
[image]

crawfxrd, 2020-06-18:
`sudo` is not required since we are writing to a file and not a block device.

This is occurring in `scripts/pxestick.sh`. We don't override `PATH` so be able to find it. Does `which mkfs.vfat` find it?

neox95, 2020-06-18:
Seems to be a path problem with debian
[image]

neox95, 2020-06-18:
Note : I've just tried compiling within su and it works without problem

crawfxrd, 2020-06-19:
This should be sufficient:

```
PATH=$PATH:/usr/sbin ./scripts/build.sh darp6
```

The patched version of parted 3.2 in Debian will produce permission errors trying to read the SMBIOS entry point. These aren't significant (and sent to `/dev/null` in 3.3).

PR #97 (2021-03-05): Fix building on Debian
Debian does not export `/usr/bin` in `PATH` for users like every other distro so builds fail to find mkfs and parted.

Fixes: #95
closed 2021-03-05 by jackpot51