system76/firmware-open #98: Issue on DarterPro-6 / Galaga Pro

matrix8967, 2020-06-20:
Recently updated the firmware on my Darter Pro 6, and I'm having issues booting after that. There are a few other's having issues in a Reddit thread [at this link.](https://old.reddit.com/r/System76/comments/hcaz8l/need_help_after_firmware_update/) 

Summary here:

- Galaga Pro and Darter Pro 6 seem to be affected. (Coreboot issue?)
- Color Cycling Keyboard and System76 Splash Screen.
- 100% Fan Usage.
- Unable to boot from USB or Bios/UEFI Boot Menu.
- Keyboard lights eventually shut off.
- Firmware Update flasher reported success.
- Updated with Barrel Charger plugged in.

a1kemist, 2020-06-20:
This just happened to me too after updating the firmware on my Darter Pro 6 with the same exact symptoms you listed above.

Hopefully there is a fix or workaround discovered soon!

jackpot51, 2020-06-20:
Thanks for the report, I have pulled the firmware update. Please contact support so we can collect more information and get your systems fixed

matrix8967, 2020-06-20:
> Thanks for the report, I have pulled the firmware update. Please contact support so we can collect more information and get your systems fixed

You bet! I've got a support ticket in, and 2 or 3 others from that thread do as well. We're trying some dumb noob stuff - but I'm reading about flashing with a raspberry pi + spi. I've never done it - but it doesn't seem too far away from my understanding. 

Let me know if you need a brave lab-rat.

jackpot51, 2020-06-20:
Well, this is the worst time for something like this to happen.

Please try unplugging all batteries (main and CMOS) and ac adapter for at least 30 seconds. Plug in only the cmos battery, then the ac adapter, and then power on. It should get to the firmware logo in 30 seconds. If not, there is a problem. When that happens, press space and go to the systemd-boot menu. The fan should not be running full speed - if it is, there is a problem. Press e to edit and remove `quiet loglevel=0 splash`, then boot. Hopefully there are messages indicating the problem.

matrix8967, 2020-06-20:
> Well, this is the worst time for something like this to happen.
>
> Please try unplugging all batteries (main and CMOS) and ac adapter for at least 30 seconds. Plug in only the cmos battery, then the [...]

Yep - on it. I tried pulling batteries (and HD, different ram, etc) earlier today. but I don't think I booted with the battery + Cmos disconnected. 

Gonna skip to the log level step real quick and if it doesn't pan out, I'll go through the battery steps. 

Stand by...

jackpot51, 2020-06-20:
Thanks for your help

jackpot51, 2020-06-20:
What is your hardware configuration?

matrix8967, 2020-06-20:
> Thanks for your help

Happy to. 

No change on modifying the boot line(s) of any of the entries. Fans kick back up after a minute. The *only* change when modifying the boot line as you mentioned is  that the keyboard stops color cycling and stays blue. Where as the other times it continues to cycle before shutting the light off. 

Gonna start pulling the battery etc. 

I'm on `darp6` 4.9 GHz i7-10510U (1.8 up to 4.9 GHz - 8MB Cache - 4 Cores - 8 Threads) 32gb ram. 

Others in the thread are on darp6, and galaga -- seems (right now) to be limited to core boot users??

jackpot51, 2020-06-20:
Interesting that lemp9 users are not affected, probably something with the EC. What drive so you have?

matrix8967, 2020-06-20:
256gb nvme (samsung.) 

Can't boot USB (same issues, just hangs.)

jackpot51, 2020-06-20:
I will replicate your configuration and perform the update

matrix8967, 2020-06-20:
Awesome - I'll keep the thread open while I'm getting this thing opened up.

k32, 2020-06-20:
Same issue. Here's my config, if it helps:

```
Orders ship the last week of October
--
1× Pop!_OS 19.10 (64-bit) with full-disk encryption
1× 15" Matte FHD 1080p Matte FHD Display
1× Intel® UHD Graphics
1× 4.9 GHz i7-10510U (1.8 up to 4.9 GHz - 8MB Cache - 4 Cores - 8 Threads)
32 GB Dual Channel DDR4 at 2666MHz (2× 16GB)
1 TB NVMe Seq Read: 3,500 MB/s, Seq Write: 3,300 MB/s
WiFi up to 867 Mbps + Bluetooth 5
United States Keyboard
```

Additional info: I updated OS to 20 just before the firmware update.

pkactual, 2020-06-20:
> What is your hardware configuration?

DARP6, i7-10510U, 64GB, 1TB NVMe  (all from System76)

Tried most of the above too - additionally:

- If you let it wait 25m or so, it'll eventually get to decrypt and LAGGGGS keystrokes take a long clip. Then sits at Please unlock ..

- If you try to boot external media, it goes black for 15m or so, then System76 logo, then 13m or so it would go to decrypt instead of the bootable media

- Same booting old kernel too

One difference from Matrix, on this DARP6, it is entirely unresponsive to trying to spacebar into editing .. the first time it responds to anything after either ESC or F2 is when it finally gets to decrypt.

Lastly, I've been on 20.04 POP since release day.

matrix8967, 2020-06-20:
Reporting back that removing all batteries + editing the boot line(s) yields the same results. :/

matrix8967, 2020-06-20:
Whoa just got a bunch of text on the screen stand by...gotta play a game of logger real quick.

jackpot51, 2020-06-20:
Replicating your 256gb nvme, i7 darp6 config I flashed back to the prior firmware and then flashed the update again with no issues. I'll yry that a couple more times before swapping RAM

pkactual, 2020-06-20:
ln: /tmp/mountroot-fail-hooks.d/scripts/initi-premount/lvm2: No such file or directory
Nothing to read on input

Repeats that ... this is about 45m into boot cycle.

Then back to System76 logo rendered REALLY SLOW ...

matrix8967, 2020-06-20:
This looks like it's trying to boot...It's asking for luks pw Other than CPU Throttling (lines up with the fan speed I'm hearing...)

If I can get to the FS I'll pull logs. Grabbed some pictures of what I saw but not sure if there's anything good in it.

jackpot51, 2020-06-20:
That would help a lot, thanks

matrix8967, 2020-06-20:
After the luks unlock I got more `cpu clock throttled` / `temperature speed normal` messages, 

Now I'm waiting on `perf interrupt took too long lowering kernel.perf_event_max_sample_rate to 8500`

jackpot51, 2020-06-20:
Do you have a usb stick and the ability to run popsicle?

pkactual, 2020-06-20:
> After the luks unlock I got more `cpu clock throttled` / `temperature speed normal` messages,
> 
> Now I'm waiting on `perf interrupt took too long lowering kernel.perf_event_max_sample_rate to 8500`

You're getting more than I am - I can't get into systemd / space bar .. I'm considering yeeting the disk out of the drive to see if I can decrypt and read logs. You're on the Galaga and I'm on the Darter.

matrix8967, 2020-06-20:
> You're getting more than I am - I can't get into systemd / space bar .. I'm considering yeeting the disk out of the drive to see if I can decrypt and read logs. You're on the Galaga and I'm on the [...]

I'm also on darter.

matrix8967, 2020-06-20:
> Do you have a usb stick and the ability to run popsicle?

I've got other hardware I can use and get something going. Let me rummage for popsicle source. 

Specifically popsicle? `dd`or etcher or something alright?

jackpot51, 2020-06-20:
Duplicated ram config and flashed back and forth, still nothing

matrix8967, 2020-06-20:
> Duplicated ram config and flashed back and forth, still nothing

Weeeird. This specifically is "goldkey" ram @ 2666. Mar'20 assembly/build date. 

This thing's still just spitting CPU/Core temp errors now that I've unlocked the disk. :/

jackpot51, 2020-06-20:
Either etcher or popsicle or dd would work. Unzip usb.img from usb.zip below. Use something like etcher or popsicle to image it to a USB stick. Attempt to boot from the USB stick. It should take you through flashing the previous firmware, before whatever bug was introduced.

[usb.zip](https://github.com/system76/firmware-open/files/4807415/usb.zip)

pkactual, 2020-06-20:
> > You're getting more than I am - I can't get into systemd / space bar .. I'm considering yeeting the disk out of the drive to see if I can decrypt and read logs. You're on the Galaga and I'm on the Darter.
> 
> I'm also on darter.

Gah. My mistake .. although yours is behaving slightly differently than mine. That's probably not a great sign either way.

Hrmm. @matrix8967 @jackpot51 Mine is also a March build ..

It just went to all gray one more time and then shutdown. I tried to boot from external USB once already and it just ignored it but I'll try the above too.

matrix8967, 2020-06-20:
> Either etcher or popsicle or dd would work. Unzip usb.img from usb.zip below. Use something like etcher or popsicle to image it to a USB stick. Attempt to boot from the USB stick. It should take you [...]

"App error: NotFound" text spilled into the top left corner of the Firmware update screen.

jackpot51, 2020-06-20:
Can you send a picture?

matrix8967, 2020-06-20:
`f841b7c171e0108329edbae50a69f63e07cc4736b3733adff1d97471f79c2c81` 

was the sha256 of the zip file. Extracted images was ~51mb and 

`ff3d9fa8bd8707189be3895e440e34dcce8752bb6b972d5f27f3b859232547d4`

Picture incoming

matrix8967, 2020-06-20:
[image]
I'm so sorry you have to see your baby this way.

jackpot51, 2020-06-20:
I will prepare another zip file soon

pkactual, 2020-06-20:
@matrix8967 @jackpot51 Mine just ignores the external media - tried a different stick too. No joy.

I'm getting scolded for ruining weekend plans now so I'm off to lick my wounds. Matrix, you deserve more gold at the end of this too. What a mess.  :-/

matrix8967, 2020-06-20:
@pkactual Nah man, I'm happy to help. Honestly a weird and interesting problem - and I'm happy to work on it. I also have 0 pressure on me in the situation :joy: so ya know. 

Just donate to a cool charity or nonprofit, or buy something from a local business or use it to get your self outta the dog house. Reddit has all the money it needs. 

:+1: Later dude, I'll keep you updated.

matrix8967, 2020-06-20:
> I will prepare another zip file soon

While that's going on, I'm gonna try to boot (any distro) since I've made a little bit of progress here. Earlier today I could see the splash screens and then they'd also just go to a black screen.

jackpot51, 2020-06-20:
Ok, here's another one. If it works I think I know what is going on. NVRAM is full.

[usb.zip](https://github.com/system76/firmware-open/files/4807427/usb.zip)

matrix8967, 2020-06-20:
> Ok, here's another one. If it works I think I know what is going on. NVRAM is full.
> 
> [usb.zip](https://github.com/system76/firmware-open/files/4807427/usb.zip)

Awesome - can you hit me with a sum of the zip or the img so I know we have the same file while I get this written on a disk?

jackpot51, 2020-06-20:
```
a8fa3f763b3b0922339500ac462c36b0cce5d6803c9d12af63da2c434fa6193e  usb.img
```

matrix8967, 2020-06-20:
> ```
> a8fa3f763b3b0922339500ac462c36b0cce5d6803c9d12af63da2c434fa6193e  usb.img
> ```

We're back. :raised_hands:

matrix8967, 2020-06-20:
Touch and go there for a sec -- screen stayed black after the reboot for about a minute or so but then it woke right up, spash screen, luks, and everything's normal. 

I'd ask what clued you in to the NVRam thing -- but it's late. 

Do you want me to pull logs anyway?

jackpot51, 2020-06-20:
Logs won't help anymore. I just know that firmware updater in and out. The place where it stalled only could happen if NVRAM was inaccessible. Linux wouldn't boot likely for the same reason.

matrix8967, 2020-06-20:
"Where we're going - we don't need logs." :sunglasses: 

Right on man. Well. I'll close this I guess, thanks for jumping on it and seeing it through. I'll reference this in the reddit thread - would the same firmware apply to the Galagas etc? 

Or should I just lay low and wait for you all to push something else out?

jackpot51, 2020-06-20:
To summarize:

- Download https://github.com/system76/firmware-open/files/4807427/usb.zip
- Extract usb.img and verify with sha256sum:
```
a8fa3f763b3b0922339500ac462c36b0cce5d6803c9d12af63da2c434fa6193e  usb.img
```
- Use etcher or popsicle to image that to a USB flash drive, preferably USB 2.0
- Turn on your broken system with the drive plugged in, and the AC adapter plugged in
- Press ESC at the System76 logo to get to the firmware menu. Go to the boot menu.
- Boot from the USB drive if it shows up in the firmware boot menu. If not, please take a screenshot of the firmware boot menu with the drive plugged in, and note the drive make and model.
- The previous firmware will be installed by the firmware updater and the system will prompt to shut down
- When booting up the next time, it will have a blank screen for about 30 seconds. This is normal.
- After 30 seconds, the System76 logo should appear. Press ESC, go to the boot menu and select your internal drive
- Boot your OS and check for stability
- If at any point any instructions don't work please make a comment with a screenshot

jackpot51, 2020-06-20:
I am going to sleep now. Our team should be up tomorrow morning, Mountain Time.

k32, 2020-06-20:
Sadly, I can't boot from USB drive that I prepared using etcher. All I get is this: https://a.uguu.se/FQF3bGK1TQX0.jpg

Reboot Into Firmware Interface leads to the same menu.

ccozianu, 2020-06-20:
darp6 here, same symptoms, with new firmware processor and fan goes into a spin even at a grub prompt.

Tried a8fa3f763b3b0922339500ac462c36b0cce5d6803c9d12af63da2c434fa6193e  usb.img , but the boot menu doesn't recognize the USB as bootable.

This UEFI crap is absolutely frustrating (I couldn't boot any other USB drives like debian that are bootable on my other laptop).

phuctranvn, 2020-06-20:
darp6, i5-10210u 2x16GB_RAM 500GB_SSDnvme hang on system76 logo after firmware update.
Can not boot from live usb ubuntu/pop_os, can not open firmware interface too. But when i remove 1 RAM slot and then it can boot into pop. Now i only use 1 slot RAM, it hang on boot when use dual chanel.

jackpot51, 2020-06-20:
@k32 @ccozianu @phuctranvn please try removing all internal drives and all but one memory stick, then try to boot from the USB drive

pkactual, 2020-06-20:
@k32 Tap 'ESC' on boot - see if you see your USB in there.

@jackpot51 Yoinked internal drive and all but one memory stick, still won't see the USB as bootable. Tried a PNY, a Sony, a Sandisk, and a LEGO something my kid had that is detected as a PNY. No joy on any of them.

@matrix8967 Sincerely happy to see you're back up though!

jackpot51, 2020-06-20:
@pkactual can you send a screenshot of what happens if you let it boot with only the usb drive plugged in, and without pressing any keys?

leviport, 2020-06-20:
@pkactual Another thing to try -- are you hitting Esc as soon as it begins booting, or waiting until you see the System76 logo screen? I've found some devices occasionally take a little longer to initialize and won't show up in the boot menu if they aren't initialized. With those, I have the best success waiting to press Esc until the logo screen has been up for about a second and a half. Basically you want to hit Esc at the last possible second.

pkactual, 2020-06-20:
@jackpot51 If I press nothing - it just hands on the System76 logo screen. I tried swapping sticks too. I also tried the additional step Ubuntu's site suggests of erasing into FAT format before imaging the USB.

@leviport I got excited there for a second but no joy - tried all manner of countdown, I can get to three-on-thousand before ESC is no longer allowed but none of the three respond any differently then.

Going to snag another machine from the office just to make sure I can boot these USBs on _some_ device and come back.

k32, 2020-06-20:
@jackpot51 Thank you -- that worked! I removed one RAM module and SSD and after two attempts got to the flashing part. After it completed, I reinstalled the removed components and got to the GDM.

Nice work troubleshooting this!

pkactual, 2020-06-20:
One last thing before I go into the office to get another USB and test box - I tried dd instead of Etcher just in case. No different. Happy to hear it @k32 .. hopefully something will land and work on my end soon.

crawfxrd, 2020-06-20:
May be worth picking https://github.com/MrChromebox/edk2/commit/8c5f3e4bd84b5d366699d0541a477adb809faf5c to reduce the amount of NVRAM vars we write.

k32, 2020-06-20:
One thing that I forgot to mention is that prior to RAM and SSD removal suggested by jackpot51, I tried removing CMOS and internal batteries for some minutes. That alone didn't work, but maybe it somehow affected the end result.

jackpot51, 2020-06-20:
@pkactual you may want to try the battery removal like @k32 did

pkactual, 2020-06-20:
OK - we have recovery ... first off, thank you @matrix8967 from Reddit to here. Thank you @jackpot51 for your Friday night too.

@k32 pushed it over the edge, the battery was the missing step -

Old firmware back on ... booted in.

@jackpot51 Presumably this will be re-released shortly with a fix? Not that I'm trying it first.  ;-)

I would also like to note LEGO USB drives are hilarious.

[17 more replies]

closed 2020-09-22 by jackpot51