system76/firmware-open #104: Swap fn and ctrl key

Gavinok, 2020-07-03:
As a longtime thinkpad user I would love to be able to swap the left hand fn key with the ctrl key to match the original thinkpad layout.

leviport, 2020-07-06:
This most likely isn't a change we'd make across the board, but there is a way to customize your own keyboard map at the firmware level. This has recently been documented here: https://github.com/system76/ec/blob/master/doc/keyboard-layout-customization.md .  If you take those instructions for a test-drive, please let me know if you run into any trouble so I can improve it.

Gavinok, 2020-07-07:
awesome thanks

closed 2020-07-07 by Gavinok

rbreaves, 2020-07-24:
This is pretty awesome. I am the author of kinto.sh and will likely look into getting a system76 laptop at some point. If I do I will definitely compile a custom BIOS for it to have the key layout I want, which will give it a macbook key layout.