system76/firmware-open #109: Support wake on AC and wake on LAN

jackpot51, 2020-07-21:


crawfxrd, 2020-07-21:
The Insyde H2O BIOS enables Wake on PME, but it is not configurable on the oryp5.

jackpot51, 2020-09-22:
We discussed wake on AC before, and decided it was not necessary with the addition of wake on LAN. So I think this is done

closed 2020-09-22 by jackpot51

tomkinsc, 2022-03-01:
Joining this conversation a few years late, we would value wake-on-AC. We have a bit of an unusual use case: we tend to deploy System76 systems in always-on configurations to accompany scientific instruments in developing countries with unreliable electric grids. Laptops work well for settings like this because in addition to whatever UPS and generator are backing the power, the built-in battery buffers power too. Unfortunately there are times when the internal battery, UPS, and generator all run out. In such circumstances it would be helpful for the systems to be able to wake automatically when power returns. Wake-on-LAN does not fulfill the need because we would have to add another device with wake-on-AC to serve as a watchdog and issue the wake-on-LAN packet.