system76/firmware-open #111: Secure Boot support?

misuzu, 2020-07-25:
There should be a way to enable Secure Boot and [add user-generated platform keys to protect system from tampering](https://www.linuxjournal.com/content/take-control-your-pc-uefi-secure-boot).

crawfxrd, 2020-07-29:
This is not something we are looking to add. (I know it's at least blocked by changes that need to be made in our edk2 fork, which requires changes to the coreboot smmstore support in edk2.)

Verified Boot support may be added instead, but it's a low priority.

crawfxrd, 2020-12-01:
Work being done by 9elements to edk2 should make Secure Boot support feasible. We would need to switch to SMMSTOREv2 first.

ckorder, 2021-03-11:
> Verified Boot support may be added instead, but it's a low priority.

sadly, i considered buying one. 

any news related to this topic yet?

anything like pureboot rsync with usb stick of the boot partition 
would be awesome for the future. please make some kind of a collaboration.
i love the purism boot, hardware kill switches but hate the "low" quality of hardware components within purism products. system76 are the best casual gnu/linux "ultra" notebooks but not the best relating towards the security aspect.

crawfxrd, 2021-03-11:
I just started working on rebasing on edk2-stable202102 yesterday, and looking through [9element's fork](https://github.com/9elements/edk2) for what can be picked. If I can get the FV block protocol + SMMSTOREv2 working then I should be able to also pick the commits for SecureBoot support.

(I did try to enable vboot on one of the boards, but I'm not convinced I did it correctly.)

anweiss, 2021-11-01:
Thanks so much for this @crawfxrd! Can't wait to see this merged!

mrtrivia, 2021-12-29:
Hey, @crawfxrd!

I just bought a new Gazelle (a gaze16, IIRC). I'm mostly a Windows user, but, honestly, if I can't install Win 11 on this computer, I might eventually just switch to Pop_OS. (I had been using Kubuntu, but I really like Pop.)

For what it's worth, as a new and pretty satisfied Sys76 owner, I would love it if you would support Secure Boot. (I'm a lawyer, so I need to whistle up *some* kind of Windows encryption tout suite.) I can't be the only Sys76 owner who would like to install Win 11 (or who will need to) without having to resort to kludges.

The power, storage (two physical drives? _Even three_?), general flexibility--all with an under-the-arm form factor!--sold me on this computer. Now that I have it, I ain't sending home for lack of this one feature.

That said, it'd be a great upgrade.

Thanks for your interest in this!

Here-Be-Saoirse, 2022-01-05:
it looks like the latest build of open firmware added coreboot *measure boot*, not secure boot? I'm a little confused?

crawfxrd, 2022-01-05:
It was originally enabled for the update, but was disabled because of issues that came up.

- A user reported their machine was not in setup mode when they enabled it, and prevented them from booting their OS.
- edk2 crashes loading a signed systemd-boot binary.

Here-Be-Saoirse, 2022-01-24:
heya, any update on this? I'm hoping to be able to upgrade to Windows 11 and the system is just absolutely not having it?

Kantankerus, 2022-06-20:
Is it currently possible to enable secure boot in edk2.config and build your own firmware or does this have high risk of bricking your machine?

jackpot51, 2022-06-20:
I'd recommend having an external programmer before doing that: https://github.com/system76/firmware-open/blob/master/docs/flashing.md

crawfxrd, 2022-07-10:
I've never had a problem with leaving it in Setup Mode.

```
$ mokutil --sb-state
SecureBoot disabled
Platform is in Setup Mode
```

---

I was planning on updating edk2 to edk2-stable202205, but the Secure Boot implementation from Star Labs on that branch doesn't seem to work. And AFAICT the 9elements implementation has never been updated for newer edk2 releases.

bmilliron67, 2022-08-10:
I would also like to see SecureBoot support enabled.

brandsimon, 2022-10-02:
I think it is useful to show the hash of the Platform Key, if a different operating system than default (Microsoft trusted Platform Key) is the current platform key and secure boot is enabled.
Android smartphones have this feature for several years, but all laptops I know lack this important security feature. [0] 
Please keep in mind, that the screenshots are not fully up-to-date, devices show not only the first 8 digits, but the full root of trust hash. [1]
The reference source code is available here: [2]

[0] https://source.android.com/docs/security/features/verifiedboot/boot-flow#locked-devices-with-custom-root-of-trust
[1] https://issuetracker.google.com/issues/217720443
[2] https://source.codeaurora.org/quic/la/abl/tianocore/edk2/tree/QcomModulePkg/Library/BootLib/VerifiedBootMenu.c?h=LA.UM.8.9.1.r1-03800-QCS610.0&id=8e06dfd3ceeb323546d330e918d19c542d2daee2#n340

JanZerebecki, 2022-10-05:
@brandsimon That is very bad design, though some of it may be intentional. 1) Not affording the same protection when using non-default roots like Android does is broken. 2) Making it look like a warning induces warning fatigue, people should not need to redo trust decisions, computers should remember. (People always think that the Android is broken when they see that screen. Some think it won't boot due to the delay and no progress indicator. Had people turn it off before it could boot due to that.) 3) While it needs to remain possible for people to compare fingerprints manually and it is a good idea to have the hashes well accessible, it should be possible for a computer to do that, like with a phone or fob, which has the added benefit that you can do attestation (with your own secret, not one signed by the CPU maker) at the same time. 4) You should not use a default trust root that has signed non-public, non-logged code, some of it AFAIK now know malicious, Instead one should store a list of hashes of the code needed to boot the device. And then enroll our own root of trust that has a verifiable, reproducible signing policy and thus is more secure. (You can still include widely used trust roots in a disabled state for user convenience, to not let perfect be the enemy of improvement.)

Anyway first need to have secure boot support or something like it at all. And I personally think a firmware write lock https://github.com/system76/firmware-open/issues/158 is worth while on its own and more important.

FAYZER77, 2022-12-03:
So is there no way for me to build a firmware with secureboot on and leave it in setup mode so I don't have to return the system I just bought?

Darin755, 2022-12-14:
Is it possible to use a third party bootloader with more security?

JordanBoulan, 2023-03-06:
Is there any update on this? Is secureboot coming/still being worked on or has it been given up on?

crawfxrd, 2023-05-31:
The next firmware update will include Secure Boot support. Configuration from firmware is very limited; it's expected to manage it from the OS.

No ETA on when anything will get released. It's a lot of boards to go through.

MilesBHuff, 2023-07-27:
Worth noting, since this issue was opened when the latest System76 laptops had 10th gen Intel CPUs:  Flashing the latest system76/firmware-open will brick your computer if you are 10th gen or older (#393).  The last release for you would be [2022-07-27](https://github.com/system76/firmware-open/blob/master/CHANGELOG.md#2022-07-27), and the last safe commit would be b337ac66fb3b8d77880e81f35acf3c50072a85c2.
So you should wait for an official release;  don't try to flash this yourself unless you're 11th gen or later.  Note though that because 10th gen and lower is being deprioritized (#402), it may be a while before you get Secure Boot (if ever).

jacobgkau, 2023-07-27:
> Note though that because 10th gen and lower is being deprioritized (#402), it may be a while before you get Secure Boot (if ever).

oryp7, a 10th gen system, just got an official update last Friday, which includes Secure Boot support (and does not brick the machine.)

A number of other previous-generation systems are undergoing testing, and our buildchain still runs on all systems back to the ones Open Firmware initially shipped on; there is no specific generation cutoff being "deprioritized." Daniel was only referring to the idea of lowering an issue's priority after adding a warning about it.

We'll have to look into the current state of #393.

crawfxrd, 2023-07-28:
Secure Boot support is enabled in master.

Track #340 to see if a release (2023-03-22 or newer) has been made for your model.

closed 2023-07-28 by crawfxrd

Anyknown, 2024-01-29:
> Is it possible to use a third party bootloader for more security?

i was thinking Tails. Generate and mount onto a USB using another machine (for example a macbook) for Linux. Then utilize the persistent feature for the safeguard measures you want the Linux machine to have installed, veracrypt, VPN, etc. Would that work?