system76/firmware-open #121: [Hardware] PopOS! installation failure and configuration issues

azrea, 2020-08-15:
_First sorry for open such issue in this repository however i have no way to resolve below problems easily and popos support basically available for devices purchased customers. i also believe @jackpot51 @ids1024 capable to solve that for me :)_

**Let me clarify the whole story instead describe the issue. it may help to get better idea, just don't mind about unnecessary details.**

The device: HP 15T-au000 https://support.hp.com/us-en/product/hp-pavilion-15-au000-notebook-pc-series/10862158/model/11792506
Pre-installed OS - Windows 10 (It works properly with that device and latest version)

Recently i installed Ubuntu 20.04.01 - focal forsa (not dual boot), did not noticed any problems during installation and nearly for a day then the first issue appeared. wi-fi auto disconnect randomly (sometimes constantly) but i can visit to websites, watch videos on youtube etc (only installed web browsers, note taking apps during that time). Meantime i filed a bug report (it contain adapter details and basic `journalctl --follow` log https://bugs.launchpad.net/ubuntu/+source/network-manager/+bug/1891407 
I didnt had reason to completely shutdown device for several days except restart. anyway after the first shutdown then boot the device again, everything went normal till login screen and i tested both my root and non root accounts; just after entered password to login then next move screen become purple (desktop doesnt appear)
(i had no way to open terminal or connect to internet on login screen because those weren't enabled or even if i pressed certain key combinations then spam screen with other errors and i found on forums about same purple screen issues without proper solution then i thought to give up and they also have moved to popos)

Downloaded PopOS! https://pop-iso.sfo2.cdn.digitaloceanspaces.com/20.04/amd64/intel/11/pop-os_20.04_amd64_intel_11.iso
Flashed to new flash drive using etcher (by default, we have to use PopOS! try mode before install then i noticed `PCIe Bus Error: severity=Corrected, type=Physical Layer` spamming on screen but popos loaded well automatically then i followed steps on wizard, choose languages, partitioning etc then finally it stuck on "Configuring the system - 70%"

After checked on forums and according to many users experience then i thought iso(however not corrupted) or recent flashing issue. https://www.reddit.com/r/pop_os/comments/ejgvf4/pop_os_installation_stops_at_70_tried_many_times/

Downloaded new iso then plugged the drive to another port and flashed. now has another problem i can't get the pc into popos try mode(the default procedure) easily however because screen spamming with `PCIe Bus Error: severity=Corrected, type=Physical Layer`

[image]

For search, found below articles with several parameter options and solutions but i dont have much experience to adjust these solutions for solve my problems then please read below questions and details

https://www.reddit.com/r/pop_os/comments/cfxpz8/failed_to_install_popos_1904_because_pcie_bus/
https://www.reddit.com/r/linuxquestions/comments/9mq7m8/pcie_bus_error_when_installing_linux/
https://unix.stackexchange.com/questions/327730/what-causes-this-pcieport-00000003-0-pcie-bus-error-aer-bad-tlp
https://askubuntu.com/questions/874349/pcie-bus-error-severity-corrected
https://unix.stackexchange.com/questions/440778/pop-os-doesnt-boot-up-because-of-pci-e-bus-error
https://askubuntu.com/questions/771899/pcie-bus-error-severity-corrected

https://askubuntu.com/questions/19486/how-do-i-add-a-kernel-boot-parameter
https://askubuntu.com/questions/353502/is-it-possible-to-boot-without-adding-pci-nomsi-or-acpi-off-to-the-grub-comm
https://askubuntu.com/questions/911296/pcie-bus-error-ubuntu-16-04
https://askubuntu.com/questions/863150/pcie-bus-error-severity-corrected-type-physical-layer-id-00e5receiver-id
https://www.reddit.com/r/linux4noobs/comments/8zsyrh/pcie_bus_error_severitycorrected_typephysical/
https://askubuntu.com/questions/1163190/pcie-bus-error-severity-corrected-type-physical-layer-id-00e5receiver-id-how

1. I prefer solve that `PCIe Bus Error: severity=Corrected` error for load the popos for installation.
2. I want to solve that `Configuring the system - 70%` stuck issue. i dont know what cause for it but i dont want to re-format again and again (on my laptop has more important details in other partitions; if hard disk health goes bad then good luck for me. **that's why i also reach popos devs through this repo**)
3. I want to solve that `PCIe Bus Error: severity=Corrected` error for after use (means after finished installation and daily use)
Below article may helpful for me anyway https://itsfoss.com/pcie-bus-error-severity-corrected/
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1521173

Currently i can get pc to below screen (if didnt choose flashdrive through bios boot options but still the same thing just load from flashdrive) then the second screen possible to get by pressing "e"

[image]
[image]

```
GNU GRUB version 2.04

setparams 'Try or Install Pop_OS'
   set gfxpayload=keep
  linux /casper_pop-os_20.04_amd64_intel_debug_22/vmlinuz.efi boot=casper live-media-path=/casper_pop-os_20.04_amd64_intel_debug_22 hostname=popos username=po\p-os noprompt ---
  initrd /casper_pop-os_20.04_amd64_intel_debug_22/initrd.gz
```
https://www.reddit.com/r/linuxquestions/comments/9mq7m8/pcie_bus_error_when_installing_linux/
(according to first answer then i modified the line as i understood but then got blank screen and os didnt load)

2. The most import thing that i expect solution `Configuring the system - 70%` stuck issue.
(even i could reboot pc hundred time for 1st cause but i cannot format my disk like that)

crawfxrd, 2020-08-15:
This repo is for issues with System76 Open Firmware and related projects.

Please open an issue in a relevant [pop-os](https://github.com/pop-os/pop) project.

closed 2020-08-15 by crawfxrd

azrea, 2020-08-16:
> This repo is for issues with System76 Open Firmware and related projects.
> Please open an issue in a relevant [pop-os](https://github.com/pop-os/pop) project.

@crawfxrd sorry i didnt noticed then opened https://github.com/pop-os/pop/issues/1193
It seems popos team doesn't care much about opened issues in that repo but this's urgent cause for me :(