system76/firmware-open #127: darp6: Failed Firmware update

npmenard, 2020-09-18:
Yesterday I run a firmware update on my Darter Pro running Archlinux using system76-firmware-cli from the system76-firmware-daemon-git AUR package.
When scheduling the firmware update the only issue I ran into was that the script couldn't remove Boot entry 0001 (expected since it doesn't exist, my Archlinux boot on entry 0002) Rebooting and running the update worked without any issues, at the end the updater reported that the boot entry was removed and then the system rebooted.

This is when the issues started, at reboot the system booted the firmware updater again which then throw a Device Error and asked to press a key to reboot. For all subsequent reboot the BIOS would again boot the firmware updater **ignoring the ESC and SPACE keys**. At this point it was impossible to select another boot entry or boot from an USB key or access any BIOS menu.
I tried to unplugged battery and CMOS battery without any result, I removed one RAM module without any result finally I removed the SSD. Without the SSD the BIOS would cycle through the boot entries (first the firmware updater then Archlinux) and complain that it couldn't find the EFI executable. After cycling through the entries I was able to boot from a USB key (with PopOS) but still could not interact with the BIOS menu via ESC or SPACE key. In POPOs I used efibootmgr to remove the 1776 entry as to avoid booting into it, the program didn't report any errors but after rebooting the BIOS would still boot first into the firmware updater. I also tried to add boot entry (just for the sake of it) in this case efibootmgr reports a **Could not prepare Boot variable: Input/output error** and refuse to add the entry 

I plugged the SSD drive to another machine to remove the firmware updater EFI executable once back in the Darter the system would boot again Archlinux (after complaining about the firmware updater missing) **But** after I replaced the second memory stick back into the machine, the BIOS would happily load grub the the boot process would get stuck at loading ramdisk. and Thus would not boot, the same happened with my PopOS USB key and my archlinux USB key. 

I read about issue #98 and tried the darp6 USB image that was linked, however I ended up with [image]

Finally I booted into my PopOs usb key and build the firmware myself using firmware open, I first tried to boot into the usb.img generated but it would give me a DeviceError. I was however able to flash the BIOS using the flash script (but i didn't update the EC) The BIOS was successfully updated and I was able to reboot into it (2020-09-17_f10af76), the ESC and SPACE started to work again **BUT** I still can't boot into anything when both memory stick are in, and I still cannot add an entry using efibootmgr. The only way to boot my Archlinux is to drop in an EFI shell and manually load grubx64.efi

This was a long post but I wanted to give as much information as possible.

Summary:
1. Updated BIOS firmware using the cli tool under archlinux
2. After update **ESC** and **SPACE** stopped working and the system would continuously boot into the updater
3. Removing 1 RAM stick, SSD and disconnecting CMOS battery and battery allowed me to boot into a USB key, but the **ESC** and **SPACE** key still didn't work.
4. Putting the RAM stick back would not let me boot beyond the ramdisk stage of the boot process. So going forward I removed it
5. Tried solution fronm #98 that would not work either
6. Rebuild a BIOS image and updated it using the flashing tool (under a PopOS live) worked. **ESC** and **SPACE** worked again as expected, however I am still not able to manipulate boot entry using efibootmgr
7. If I put the second RAM stick back would still not let me boot beyond the ramdisk stage.
8. Right now my system works with one RAM stick out and if I manually boot to my system using an EFI shell.

Things I will try:
1. Cycle the RAM sticks and see if the boot issue happens regardless of order or position.
2. Flash the EC

ryanolton, 2020-09-21:
Darter Pro (darp6) updated today with 2020-09-03_9c310f0 and I have the **exact** same issue as shown in the screenshot above.

jackpot51, 2020-09-21:
I believe I know the issue, based on the Device Error message mentioned.

In testing this update, we flashed back and forth between firmware versions a number of times on a number of Darter Pros, with no issue. But I believe that updating without clearing NVRAM between the very first Darter Pro firmware and the latest Darter Pro firmware is not functioning correctly. Updating without clearing NVRAM between the previous version and the latest version is. So devices that have kept their NVRAM over a number of updates may be unable to write NVRAM variables.

I will provide a new usb.img that will fix this, and the firmware updater will detect this condition and clear the NVRAM if necessary.

jackpot51, 2020-09-21:
Here are the steps to follow to clear your NVRAM, which should fix this issue:
- Download [darp6-clear-nvram.zip](https://github.com/system76/firmware-open/files/5258302/darp6-clear-nvram.zip)
- Extract darp6-clear-nvram.img from darp6-clear-nvram.zip and verify with sha256sum:
```
71d6c7ecf6276ca0d4a3f5dae53a52cb6f0f254035a17388466aecd6515511db  darp6-clear-nvram.img
```
- Use etcher or popsicle to image that to a USB flash drive, preferably USB 2.0
- Remove all internal drives and all but one memory stick from the broken system
- Disconnect AC, main battery, and CMOS battery for 30 seconds, then reconnect them
- Turn on your broken system with the drive plugged in, and the AC adapter plugged in
- Press ESC at the System76 logo to get to the firmware menu. Go to the boot menu.
- Boot from the USB drive if it shows up in the firmware boot menu. If not, please take a screenshot of the firmware boot menu with the drive plugged in, and note the drive make and model.
- The previous firmware will be installed by the firmware updater and the system will prompt to shut down
- When booting up the next time, it will have a blank screen for about 30 seconds. This is normal.
- After 30 seconds, the System76 logo should appear. Press ESC, go to the boot menu and select your internal drive
- Boot your OS and check for stability
- If at any point any instructions don't work please make a comment with a screenshot

jackpot51, 2020-09-21:
A user downloading the firmware update from this point onwards will have their NVRAM cleared, possibly fixing the issue. If the update was downloaded before this comment was made, the instructions in https://github.com/system76/firmware-open/issues/127#issuecomment-696397324 should be used.

ryanolton, 2020-09-21:
@jackpot51 I have followed the steps detailed above and am now at a Firmware Updater screen that reads:

`App error: NotFound`

Thoughts?

jackpot51, 2020-09-21:
Could you please give me a screenshot?

ryanolton, 2020-09-21:
[image]

jackpot51, 2020-09-21:
Thanks, what did you use to flash the USB stick?

ryanolton, 2020-09-21:
Popsicle/USB Flasher, I think there was an update to this app today (maybe?).

jackpot51, 2020-09-21:
That should have been fine then. Normally, there would be some output before an error like this. Since there is none, I can guess where it crashed. 

Here is one that avoids reading and setting the boot override:
[darp6-clear-nvram-no-override.zip](https://github.com/system76/firmware-open/files/5258420/darp6-clear-nvram-no-override.zip)

shasum256 is as follows:
```
0edc4e7ae48baad6e005a197b68f852bd77e2ca05d8d5ea806e618c4fe5eb31e  darp6-clear-nvram-no-override.img
```

mdeneen, 2020-09-21:
Hi!  I attempted to do this, wrote the contents of the iso file to a sandisk cruiser USB memory stick using dd and followed the procedure as described. 

No USB entry shows up in the boot menu regardless of the port used. I let it boot from the default device and received this:

[image]

Thinking that maybe it wiped nvram, I inserted the memory module and the nvme storage but received the same error that I started with. 

Is dd okay to use?  I tried balena etcher in windows 10 as well.

ryanolton, 2020-09-21:
This version allowed the Firmware Updater to complete, however, now I'm back to my machine not progressing past the screen that prompts you to `Press ESC for Boot/Options/Settings, or SPACE for Pop!_OS Recovery`.

When I attempt to flash the machine again using this latest file, it tells me that `* All updates applied successfully *`. When I hit ESC and select my boot drive, the screen goes dark and never boots.

Do I need to repeat any steps - battery disconnect, etc. - in order to complete this process?

leviport, 2020-09-21:
@ryanolton Does selecting your normal boot device from the boot menu also stick at the logo screen?

@mdeneen Did you use the most recent USB image that was posted here? https://github.com/system76/firmware-open/issues/127#issuecomment-696410060

`dd` should be fine, yeah

mdeneen, 2020-09-21:
Where are the one time boot items stored, if anywhere?  Is this the nvram we are trying to clear?  I'm asking because pop is and the removed nvme drive are in the list and I'm wondering if I must wait longer with all power sources disconnected.

mdeneen, 2020-09-21:
I used the previous image. Let me grab the newest one and try again.

mdeneen, 2020-09-21:
The second image is flashing over spi.

mdeneen, 2020-09-21:
I am now in the same situation as @ryanolton, stuck on a Press ESC... menu. I tried, for giggles, a USB keyboard but it did not work. 

If I put the USB stick back in I can boot from that but the only boot option in One Time Boot is "Sandisk Cruzer Fit"

ryanolton, 2020-09-21:
@leviport I can no longer get to the Boot menu, hitting ESC brings up a screen that lets me choose from:

- Pop!_OS (pop_os-current)
- Pop!_OS (pop_os-oldkern)
- Pop!_OS Recovery
- Reboot Into Firmware Interface

Selecting the last option leads to a loop back to this same screen, and subsequent time selecting the last option causes the machine to reboot.

leviport, 2020-09-21:
@mdeneen It sounds like it cleared what it needed to then. Can you now try the first image that was posted? Since the second one didn't read and set the boot override, that might be what it's stuck on.

@ryanolton Huh, that's the systemd-boot menu. I would think if you were seeing that, you'd be able to get to the OS. Do the other options get you anywhere? From the recovery partition, we could probably fix the bootloader.

mdeneen, 2020-09-21:
@leviport using the first image reports "App error: NotFound"

ryanolton, 2020-09-21:
@leviport the first three options lead to a blank screen and the fans ramping up. Nothing seem to happen. I could have sworn this happened with the last firmware update about three months ago, and reverting to the previous (May-ish) firmware was the only thing that worked. I tried going back to that firmware and the Firmware Updater won't let me proceed.

mdeneen, 2020-09-21:
Also, with the nvme storage inserted again, the boot menu says "Warning: No bootable media found" but perhaps this is expected.

leviport, 2020-09-21:
Alright, I guess I just need to keep trying to replicate the same issue you all are seeing. I've flashed it back and forth with different versions of firmware a couple dozen times and it just keeps working. Just one of those situations when I'm trying to break a machine and it won't break for me, ha.

Jeremy had an appointment, but he'll be back to help in a couple hours to keep working on this.

ryanolton, 2020-09-21:
@leviport I assume you reviewed #98? This _feels_ like the same issues as back then.

leviport, 2020-09-21:
I've reviewed it, yep. The solution is still going to look different between these two issues since we already fixed the issue without having to revert, so we just have to figure out how to get any machines back up and running that were updated right away.

Since you went through that issue though, did you get a chance to try any of the hardware things in there, like pulling the main and CMOS batteries for a few minutes, or booting with only one stick of RAM?

ryanolton, 2020-09-21:
> I've reviewed it, yep. The solution is still going to look different between these two issues since we already fixed the issue without having to revert, so we just have to figure out how to get any [...]

I did originally, yes, but haven't in awhile. I just pulled the CMOS and battery and that didn't fix. Booting with only one stick of RAM **did** allow it to boot all the way into the OS. What does this tell you?

jackpot51, 2020-09-21:
If possible, it would help to detail the configuration of your computer. The CPU, RAM, and storage. We will use this to try and reproduce the issue.

Now that it booted with one stick, try adding the other again and let me know what happens

ryanolton, 2020-09-21:
> If possible, it would help to detail the configuration of your computer. The CPU, RAM, and storage. We will use this to try and reproduce the issue.
> 
> Now that it booted with one stick, try adding the other again and let me know what happens

All the hardware come from System76:
- 4.9 GHz i7-10510U
- 32 GB Dual Channel DDR4 at 2666MHz (2× 16GB)
- 500 GB NVMe (970 Evo Plus)

Adding the second RAM stick causes it to no longer boot, the screen just remains stuck at` Press ESC for Boot/Options/Settings, or SPACE for Pop!_OS Recovery`. If I press SPACE and select any of the listed Pop!_OS options, the screen goes blank and nothing happens.

jackpot51, 2020-09-21:
Thanks, I will replicate your configuration and run some tests. For now, I would recommend running with one RAM stick. I will post a comment when I have something else to test

mdeneen, 2020-09-21:
I, too, was able to boot with just one memory module inserted. It still tells me that there is no bootable media but will boot if you pick the first menu option. 

My memory was not from System76, but is Crucial 16GB x2 dual channel 2666. 

Storage is 1tb nvme samsung 970 evo.

jackpot51, 2020-09-21:
Thanks. Looks like the problem is specific to dual channel 2x16 GB 2666 MHz memory. I have that here and will install it in my darp6 along with a Samsung 970 EVO and run tests.

jackpot51, 2020-09-22:
I have reproduced the issue and am working to solve it. For now, removing all but one RAM stick and flashing the darp6-clear-nvram-no-override.img file appears to get users back to a bootable system

WatchMkr, 2020-09-22:
Comment to link restore directions for affected customers:

- Download [darp6-clear-nvram-no-override.zip](https://github.com/system76/firmware-open/files/5259064/darp6-clear-nvram-no-override.zip)
- Extract darp6-clear-nvram-no-override.img from darp6-clear-nvram-no-override.zip and verify with sha256sum:

0edc4e7ae48baad6e005a197b68f852bd77e2ca05d8d5ea806e618c4fe5eb31e  darp6-clear-nvram-no-override.img

- Use etcher or popsicle to image that to a USB flash drive, preferably USB 2.0
- Remove all internal drives and all but one memory stick from the broken system
- Disconnect AC, main battery, and CMOS battery for 30 seconds, then reconnect them
- Turn on your broken system with the drive plugged in, and the AC adapter plugged in
-  Press ESC at the System76 logo to get to the firmware menu. Go to the boot menu.
-  Boot from the USB drive if it shows up in the firmware boot menu. If not, please take a screenshot of the firmware boot menu with the drive plugged in, and note the drive make and model.
-  The previous firmware will be installed by the firmware updater and the system will prompt to shut down
-  When booting up the next time, it will have a blank screen for about 30 seconds. This is normal.
-  After 30 seconds, the System76 logo should appear. Press ESC, go to the boot menu and select your internal drive
- Boot your OS and check for stability
- If at any point any instructions don't work please make a comment with a screenshot

mdeneen, 2020-09-22:
@WatchMkr it might be worth mentioning that these instructions do not say when to put the internal drive(s) back in.

zmitrok, 2020-09-22:
I am getting a slightly different error message in a loop after the firmware update:

[image]

talpah, 2020-09-22:
Also having a similar issue. After the firmware upgrade I was stuck in the "Firmware Upgrader" boot screen saying only "Booting from item 1776", and there was an error message in the top left of the screen "App error: DeviceError".
I've manged to get past that by following WatchMkr's instructions [here](https://github.com/system76/firmware-open/issues/127#issuecomment-696491476).
After that was done, trying to boot normally gives the error in the screenshot below. 
Hit `Esc` to enter the boot menu but now the "Change Boot Order" menu shows nothing, and the only way I'm able to boot is to select "ubuntu" in the "One Time Boot" menu. Oh, and that only works if I remove one RAM module; if both RAM modules are present, the screen goes black and nothing happens.
After booting, manually running `grub-install` gives the following:
```
Installing for x86_64-efi platform.
grub-install: warning: Cannot set EFI variable BootOrder.
grub-install: warning: efivarfs_set_variable: writing to fd 7 failed: Input/output error.
grub-install: warning: _efi_set_variable_mode: ops->set_variable() failed: Input/output error.
grub-install: error: failed to register the EFI boot entry: Input/output error.
```
I checked the mounts, and efivarfs is there:
```
efivarfs on /sys/firmware/efi/efivars type efivarfs (rw,nosuid,nodev,noexec,relatime)
```
There is a "BootOrder-8be4df61-93ca-11d2-aa0d-00e098032b8c", size 8 bytes in there. I can provide the full contents of that mount point if needed.

[image]

jaypeetee, 2020-09-22:
I have a darp6 and updated the firmware for my system as recommended by the email notification yesterday. It appeared to flash the firmware successfully. However, when I power it on, it either sits at a black screen, or shows the ESC or Space bar options, and goes no further. If I choose either of those it goes to a blank screen. Identical to what others have experienced above. Any solution yet?

jackpot51, 2020-09-22:
Anyone who still has problems, please try following the instructions above but using this USB image:
[darp6-old-firmware.zip](https://github.com/system76/firmware-open/files/5261613/darp6-old-firmware.zip)

jaypeetee, 2020-09-22:
@jackpot51 We need to take the laptop apart in order to do this?

acaloiaro, 2020-09-22:
Can you please pull this firmware update from the repos to prevent further bricking customers' computers? Keeping this flawed firmware live after confirming this issue is irresponsible.

ryanolton, 2020-09-22:
> Anyone who still has problems, please try following the instructions above but using this USB image:
> [darp6-old-firmware.zip](https://github.com/system76/firmware-open/files/5261613/darp6-old-firmware.zip)

@jackpot51 is this for those of us using two RAM sticks?

jackpot51, 2020-09-22:
@acaloiaro The firmware update was pulled yesterday. Some users may have already downloaded it before it was pulled, but applied it later.

@ryanolton yes, that should allow use of two RAM sticks and also fix the Ubuntu issue mentioned above

@jaypeetee yes, the laptop must be taken apart.

ryanolton, 2020-09-22:
@jackpot51 the **darp6-old-firmware.zip** firmware worked and my darp6 is up with both RAM sticks running firmware version `2020-05-20_de5eab8`

jackpot51, 2020-09-22:
I believe the steps can be simplified:

- Download [darp6-old-firmware.zip](https://github.com/system76/firmware-open/files/5261613/darp6-old-firmware.zip)
- Extract darp6-old-firmware.img from darp6-old-firmware.zip
- Use Popsicle or Etcher to flash darp6-old-firmware.img to a USB stick
- Remove the M.2 drive from the darp6
- Plug in the USB stick to the darp6
- Boot the darp6 and perform the firmware update
- When it shuts off, remove the USB stick and put the M.2 drive back into the darp6
- Boot into your OS again. Update packages to make sure it does not offer the firmware update again

jaypeetee, 2020-09-22:
This worked for me. Thank you for the help.

> I believe the steps can be simplified:
>
> * Download [darp6-old-firmware.zip](https://github.com/system76/firmware-open/files/5261613/darp6-old-firmware.zip)
> * Extract darp6-old-firmware.img from [...]

jackpot51, 2020-09-22:
The definitive reason the darp6 was failing to boot is due to putting the Thunderbolt controller into 64-bit memory space. This required the use of an extra MTRR (https://en.wikipedia.org/wiki/Memory_type_range_register). When also using an NVMe drive and two large RAM sticks, this causes the MTRRs to run out. The last MTRR is used indirectly for EFI variable handling. So EFI variables stop working when this occurs.

For users affected, it is recommended to first revert to the old firmware as detailed here: https://github.com/system76/firmware-open/issues/127#issuecomment-696759847

New firmware will be provided in an automatic update with the MTRR allocation fixed. This new firmware will also include open EC, touchpad improvements, Windows support improvements, and the firmware configuration screen - things we haven't been able to release for the darp6 customers since May.

closed 2020-09-22 by jackpot51

mdeneen, 2020-09-22:
@jackpot51 Thanks for the explanation, Jeremy, it is much appreciated.  I'll try to revert to older firmware tonight.

Out of curiosity how large is "large" when it comes to RAM?  I assume that multiple MTRRs are required as the capacity of the module increases, and I'm at 50% of the capacity supported by darp6.  Are EFI variables read/written from userspace or just from the BIOS firmware?

jackpot51, 2020-09-22:
@mdeneen if you have two sticks, each one needs at least one MTRR allocated to it. Larger sticks could potentially need more than that. EFI variables are read and written by the firmware and the Linux kernel.

eebutler, 2020-09-22:
Thanks for the help! I was able to implement the fix without removing the M.2 drive, which made the fix even simpler.

npmenard, 2020-09-22:
Thanks a jackpot51, the image didn't work initially so I had to revert to d645e62 and then do the update.

Let me know if you want me to test a firmware before release!

Thanks

acaloiaro, 2020-09-22:
For those who find this later, at least on a darp6, only the following steps were necessary:

- Download [darp6-old-firmware.zip](https://github.com/system76/firmware-open/files/5261613/darp6-old-firmware.zip)
- Extract darp6-old-firmware.img from darp6-old-firmware.zip
- Use Popsicle or Etcher to flash darp6-old-firmware.img to a USB stick
- ~Remove the M.2 drive from the darp6~
- Plug in the USB stick to the darp6
- Boot the darp6 and perform the firmware update
- When it shuts off, remove the USB stick ~and put the M.2 drive back into the darp6~
- Boot into your OS again. Update packages to make sure it does not offer the firmware update again

mdeneen, 2020-09-23:
I can confirm that I was able to revert to the previous firmware and everything is functioning as it was before.

matrix8967, 2020-09-23:
Just posting that I got a `sha256sum` of `8f055c591d623dfb040383cc0a644c453e3403c4ca2659c175443f18dae52924` on the `darp6-old-firmware.img` mentioned in [this post](https://github.com/system76/firmware-open/issues/127#issuecomment-696759847).

`echo "8f055c591d623dfb040383cc0a644c453e3403c4ca2659c175443f18dae52924 *darp6-old-firmware.img" | sha256sum --check
darp6-old-firmware.img: OK`

Minor detail for all this so just posting if anyone needed a sanity check because I'm pedantic.

I'm blessed with spare hardware and good backups, so I shoot first and never ask questions when it comes to `darp6` firmware updates. If there's ever public testing, I'm always down to get weird.

bryanpaget, 2020-09-23:
> For those who find this later, at least on a darp6, only the following steps were necessary:
>
>     * Download [...]

This worked for me!

jackpot51, 2020-09-23:
For any users here who would want to beta test the firmware update that fixes the issue that caused this, please follow the instructions in https://github.com/system76/firmware-open/issues/130