system76/firmware-open #129: gaze15 open firmware disables touchpad

caputmat, 2020-09-23:
Updated my gaze15 to open firmware on 9//21/2020. Flash was successful, everything went smoothly. Computer boots fine and works fine, except my touchpad no longer works. Dooesn't even appear in `xinput list`:
```
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech M570                           	id=11	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ System76 ACPI Hotkeys                   	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=9	[slave  keyboard (3)]
    ↳ Intel HID events                        	id=13	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Chicony USB2.0 Camera: Chicony          	id=12	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=10	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=14	[slave  keyboard (3)]

```
```
lsusb:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04f2:b685 Chicony Electronics Co., Ltd Chicony USB2.0 Camera
Bus 001 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 004: ID 8087:0aaa Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```
```
lspci:
00:00.0 Host bridge: Intel Corporation 10th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) (rev 02)
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics (rev 05)
00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 02)
00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model
00:12.0 Signal processing controller: Intel Corporation Comet Lake PCH Thermal Controller
00:14.0 USB controller: Intel Corporation Comet Lake USB 3.1 xHCI Host Controller
00:14.2 RAM memory: Intel Corporation Comet Lake PCH Shared SRAM
00:14.3 Network controller: Intel Corporation Wi-Fi 6 AX201
00:15.0 Serial bus controller [0c80]: Intel Corporation Comet Lake PCH Serial IO I2C Controller #0
00:15.1 Serial bus controller [0c80]: Intel Corporation Comet Lake PCH Serial IO I2C Controller #1
00:17.0 SATA controller: Intel Corporation Device 06d3
00:1d.0 PCI bridge: Intel Corporation Device 06b6 (rev f0)
00:1f.0 ISA bridge: Intel Corporation Comet Lake LPC Controller
00:1f.3 Audio device: Intel Corporation Comet Lake PCH cAVS
00:1f.4 SMBus: Intel Corporation Comet Lake PCH SMBus Controller
00:1f.5 Serial bus controller [0c80]: Intel Corporation Comet Lake PCH SPI Controller
01:00.0 VGA compatible controller: NVIDIA Corporation Device 1f99 (rev a1)
01:00.1 Audio device: NVIDIA Corporation Device 10fa (rev a1)
02:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTL8411B PCI Express Card Reader (rev 01)
02:00.1 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 12)
```
Running Pop_OS 20.04, fully updated. Stock hardware for a 15" gazelle except for an additional SSD.

Not sure what else to include or try. Thanks.

ids1024, 2020-09-23:
We've seen reports of this, and we're trying to figure out what's going on. It's not happening on all gaze15 units; I'm running one on open firmware and the touchpad seems to be working fine.

In case you're not already aware, it should be easy to switch back to proprietary firmware: https://support.system76.com/articles/transition-firmware/

ids1024, 2020-09-23:
Okay... looks like `firmware-cli` had an issue with argument parsing. https://github.com/pop-os/system76-firmware/commit/84180eea1223fe1ea5caf797944322f4efe3f1df fixes that, and should be available as an update through apt shortly.

ids1024, 2020-09-23:
The touchpad wouldn't be listed in `lsusb` or `lspci`, since it's connected by i2c. `dmesg` might have some useful information.

caputmat, 2020-09-23:
Excellent, I'll await the update and revert. If there is anything I can do to help resolve this, please let me know because I'd love to use S76's firmware.
I should mention (I apologize, I've never opened an issue before and I'm no expert by any means) that the touchpad worked after the initial FW flash but after a power cycle last night, it stopped working.

jackpot51, 2020-09-23:
Can you provide dmesg output before reverting, please?

You can then update to system76-firmware 1.0.18 and it will be able to revert you

caputmat, 2020-09-24:
Unfortunately I reverted before I read this... I'm sorry.

jackpot51, 2020-09-24:
No problem

crawfxrd, 2021-02-10:
For testing, here is 2021-01-19_d6d37c0 for gaze15: [gaze15-76ec.zip](https://github.com/system76/firmware-open/files/5961875/gaze15-76ec.zip)

caputmat, 2021-02-11:
> For testing, here is 2021-01-19_d6d37c0 for gaze15: [gaze15-76ec.zip](https://github.com/system76/firmware-open/files/5961875/gaze15-76ec.zip)

Ok, I'd love to test. How exactly do I go about flashing this?

crawfxrd, 2021-02-11:
Updated my comment with instructions.

LearnLinuxTV, 2021-02-11:
I flashed my Gazelle with the firmware mentioned above using a USB drive as described.
My touchpad is still working.

crawfxrd, 2021-02-11:
@LearnLinuxTV I believe you have the 15" 1660 Ti variant, correct? That is the same variant I have and it did not have the touchpad issue.

@caputmat appears to have the 15" 1650 variant.

jackpot51, 2021-02-11:
15" 1650: @ids1024 
15" 1660 Ti: @bflanagin, @crawfxrd 
17" 1650: @jacobgkau

jackpot51, 2021-02-11:
@bflanagin @crawfxrd @ids1024 @jacobgkau Please let me know when you have flashed the firmware, and please let me know if touchpad works on first boot, through a suspend cycle, through a reboot cycle, and through a shut down and power on cycle

LearnLinuxTV, 2021-02-11:
> @LearnLinuxTV I believe you have the 15" 1660 Ti variant, correct? That is the same variant I have and it did not have the touchpad issue.

Yes, I have the 15" version with the 1660Ti.

ids1024, 2021-02-11:
Updated and reporting `Version: 2021-01-19_d6d37c0` in dmidecode. No magic smoke or bootloops yet... oh and the touchpad works. :laughing: But I didn't have an issue with it with the Open Firmware I was previously running.

bflanagin, 2021-02-11:
Running on 2021-01-19_d6d37c0:

Touch pad tests:
* Initial boot: works
* Through suspend: works
* Through reboot: works
* Through shutdown and power cycle: works

jacobgkau, 2021-02-11:
Flashed the firmware (`2021-01-19_d6d37c0`) on 17" 1650. Touchpad works:
- [X] First boot
- [X] Through a suspend cycle
- [X] Through a reboot cycle
- [X] Through a shut down and power on cycle

Re: @bflanagin, my only bootable drive in the system was an encrypted drive (I had two other drives installed, but they weren't bootable before flashing.) I'm still able to boot from the encrypted drive. It sounds like you had more than one bootable drive in the system before flashing?

bflanagin, 2021-02-11:
@jacobgkau, you are correct.

caputmat, 2021-02-11:
My touchpad doesn't work at all, not even after first boot. For reference, I do have the 15" 1650 variant. Please let me know if I can provide any information to help resolve this, or if there is anything else I can try. Thanks!

ids1024, 2021-02-11:
@caputmat I wonder why you would have this issue with the previous firmware, and this one, but the 15" 1650 model I have doesn't.

Maybe `dmesg` has some relevant information.

jackpot51, 2021-02-11:
@caputmat what disk and memory do you have? Do you need help transitioning to the previous firmware?

caputmat, 2021-02-11:
I have 8gb ram, booting off an unencrypted m.2 (256gb) with a secondary 1tb ssd, also unencrypted. I will probably need help going back if I can't use the regular cli utility, however I can leave the open firmware installed to help troubleshoot if need be.

jackpot51, 2021-02-11:
To revert back, follow the same instructions as from https://github.com/system76/firmware-open/issues/129#issuecomment-777073137 but using one of these files depending on GPU:

- 1650/1650 Ti:  [gaze15-NH50DB.zip](https://github.com/system76/firmware-open/files/5968130/gaze15-NH50DB.zip)
- 1660 Ti:  [gaze15-NH5xDC.zip](https://github.com/system76/firmware-open/files/5968126/gaze15-NH5xDC.zip)

crawfxrd, 2021-02-11:
In addition to `dmesg`, the cbmem console will also be useful.

Clone coreboot if you don't have it already:
- As a submodule of this repo
- Or from the [coreboot GitHub mirror](https://github.com/coreboot/coreboot)

Then build and run cbmem
```
cd  util/cbmem
make
sudo ./cbmem -1 > ~/cbmem.txt
```

caputmat, 2021-02-11:
OK, dmesg: 
https://pastebin.com/PKb53mQN

and cbmem:
https://pastebin.com/tckQWkuv

jackpot51, 2021-02-11:
@caputmat since we haven't replicated the issue, we are wondering if you would help us by creating a support ticket, and mentioning that "Jeremy Soller sent me regarding the gaze15 touchpad issue." Feel free to flash back to proprietary firmware if you want the touchpad restored in the meantime.

caputmat, 2021-02-11:
Ok, will do.

crawfxrd, 2021-02-11:
@caputmat Can you remove all peripherals and reboot, then provide the logs again?

@ids1024 @jacobgkau @bflanagin can you provide dmesg output as well?

bflanagin, 2021-02-11:
@crawfxrd 

[dmesg.txt](https://github.com/system76/firmware-open/files/5968302/dmesg.txt)

ids1024, 2021-02-11:
dmesg: https://gist.github.com/ids1024/dbdb453305bd37ca47007acc69f8ef61

caputmat, 2021-02-11:
Here is teh new dmesg output:
https://pastebin.com/FSsisdJA

I had my Logitech unifying receiver plugged in when I flashed the firmware... could that be an issue? Never thought about it. Would it be worth retrying? or flashing to proprietary and back with the receiver unplugged?

jacobgkau, 2021-02-11:
dmesg: [dmesg-gaze15-1650.txt](https://github.com/system76/firmware-open/files/5968317/dmesg-gaze15-1650.txt)

caputmat, 2021-02-11:
Forgot I had an sd card in. Here is an updated dmesg.
https://pastebin.com/sLDGyx3u

crawfxrd, 2021-02-11:
> I had my Logitech unifying receiver plugged in when I flashed the firmware... could that be an issue?

It shouldn't, it's just easier to see the touchpad device without the USB devices being initialized/enumerated.

Your dmesg reports:
```
[    1.812021] input: ETPS/2 Elantech Touchpad as /devices/platform/i8042/serio1/input/input6
```

All of ours report:
```
[    4.685686] input: PNP0C50:00 06CB:CD65 Mouse as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-PNP0C50:00/0018:06CB:CD65.0001/input/input19
[    4.685802] input: PNP0C50:00 06CB:CD65 Touchpad as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-PNP0C50:00/0018:06CB:CD65.0001/input/input20
```

caputmat, 2021-02-11:
Well. I definitely have no idea what that means. Anything I can or should do?

jackpot51, 2021-02-11:
I believe we have found that a small number of units shipped with a compatible but slightly different touchpad. We will need to make adjustments for the other touchpad variant in @caputmat's computer. Please let me know when you have created a support ticket, so we can discuss more in private.

caputmat, 2021-02-11:
I have created the ticket already. I'll await further instructions.

jackpot51, 2021-02-11:
@caputmat did you revert to proprietary firmware yet? You don't need to be stuck with a broken touchpad anymore, we have all the logs we need

jackpot51, 2021-02-11:
I found your ticket, I will let you know more details soon.

0181505050, 2021-02-15:
amazing work, does this mean its safe for me to flash that gaze15-NH5xDC.zip 15 "1660TI firmware? or do think I should wait? I do have a single encrypted boot drive and a unencrypted secondary drive.

0181505050, 2021-02-15:
Ok, I just test 2021-01-19_d6d37c0 for gaze15: gaze15-76ec.zip on my 15" gazelle with a 1660TI and the touchpad no longer works. is there anything you would like to do to help solve this issue?

jackpot51, 2021-02-23:
@0181505050 follow the instructions https://github.com/system76/firmware-open/issues/129#issuecomment-777755310 to revert

0181505050, 2021-02-25:
Yea i have done. all back to normal.

0181505050, 2021-03-07:
I don't have the firmware updater working on my system, Would you mind uploading it here again when its ready and ill install via USB? many thanks.

jackpot51, 2021-03-19:
The output indicates that some gaze15's use the ELAN touchpad we use on a number of products. The output of this command from one of these systems, running the proprietary firmware, could help us:

`cat /sys/bus/i2c/devices/i2c-1/i2c-ELAN0412:00/firmware_node/path`

jackpot51, 2021-03-19:
@caputmat or @0181505050 could you help us with output from this command? Also, are you willing to try new firmware that will have touchpad settings for ELAN touchpads?

caputmat, 2021-03-19:
Sure thing. See below:
```
matt@gazelle:[~] $ cat /sys/bus/i2c/devices/i2c-1/i2c-ELAN0412:00/firmware_node/path
\_SB_.PCI0.I2C0.TPD0
```
I would be more than willing to try new firmware. Thanks.

jackpot51, 2021-03-19:
Great, we should have something to try soon.

jackpot51, 2021-03-19:
Okay @caputmat, please follow the instructions in the first comment on https://github.com/system76/firmware-open/pull/181

caputmat, 2021-03-19:
I'm replying from a Gazelle with a working touchpad and open firmware!
The dmesg log after installing the new image: https://pastebin.com/SngusV3u

Awesome work guys, I am extremely grateful. I'll let you know if anything comes up in the meantime.

jackpot51, 2021-03-19:
Awesome @caputmat ! We will release this officially soon

crawfxrd, 2021-03-22:
Firmware with ELAN touchpad settings released in 2021-03-19_af2fbdf.

closed 2021-03-22 by crawfxrd