system76/firmware-open #130: Beta test darp6 open EC firmware with MTRR allocation fix

jackpot51, 2020-09-23:
We have a new firmware update for the darp6 that does the following:

- Transition to open EC
- Enable i2c-hid touchpad interface
- Enable ACPI backlight
- Add firmware configuration information
- Add UEFI TPM2 support
- Add ACPI thermal interface
- Fix allocation of memory type range registers

While we were able to reproduce the bug in https://github.com/system76/firmware-open/issues/127 and fix it, I would like to make sure we did not miss something. So, I am asking for any users who are willing to try out this firmware update before we release it to do so, using these instructions:

- Ensure that you are on firmware version `2020-05-20_de5eab8` prior to testing
- Download [darp6-open-ec.zip](https://github.com/system76/firmware-open/files/5271282/darp6-open-ec.zip)
- Extract darp6-open-ec.img from darp6-open-ec.zip and flash to a USB stick using Popsicle or Etcher
- Boot from the USB stick and follow the instructions. When the machine shuts down, boot it back up. It will take about 30 seconds to train memory before the screen turns on
- Check that OS boots and you can log in
- If this is not the case, follow steps in https://github.com/system76/firmware-open/issues/127#issuecomment-696759847 to revert
- Comment if it successful or not on this issue

matrix8967, 2020-09-23:
@jackpot51 @leviport 

This worked for me just as described.

`de5eab8` --> `904cee4` :heavy_check_mark: 
[image]

I've got another `darp6` that I'll run through this all again later tonight or tomorrow (gotta put a drive in it) 

Any others who test this moving forward -- watch the output carefully and it could be helpful to grab a picture when the

`system will shut off in 5 seconds` message appears. I'm not sure if there's a faster/easier way -- and you'll be heading into a possible soft brick.

See attached.

matrix8967, 2020-09-23:
@jackpot51  @leviport 

:grimacing: :wave: _Heeey guys_

Sorry to be a pill. 

Just a quick weirdness I'm noticing that's persisting through reboots:

This thing is staying HOT. `s-tui` shows cpu temps hovering around `65C`

During the upgrade I was using a 65W USB-C charger. It's been plugged in since then - but it's been unplugged for about 15 min and the CPU Temps are still really high. 

Actual CPU Usage is pretty low and normal. I'm not doing anything really nuts -- getting ready to flast UbuntuTouch on my old OnePlus. 

I'm gonna shut this thing down completely - let it sit for a bit, and I'll report back if something changes.

Let me know if there's a preferred thermal metric you prefer. Attaching a screenshot showing Temps + low usage.

[image]

jackpot51, 2020-09-23:
@matrix8967 does the chassis feel hot to you?

matrix8967, 2020-09-23:
@jackpot51 Yes - very. That's how I first noticed it (reaching for the numpad.) which has the CPU just on the other side of it.

That's when I checked temps. ~65C seems pretty accurate all things considered.

jackpot51, 2020-09-23:
Is the CPU fan running? Can you grab the output of `sensors`?

jackpot51, 2020-09-23:
What location on the chassis is hot?

matrix8967, 2020-09-23:
CPU fan wasn't kicking on (bottom left corner in the screenshot) 

I'll boot it up and grab `sensors` and while I'm in there I'll just "smash and grab" journalctl / dmesg etc for later.

Chasis is hot around the CPU (Centered around the `NumLK / Backspace` keys.)

jackpot51, 2020-09-23:
Okay, I think this is just down to the fan curve on the open EC firmware running silent up to 70 C CPU temperature. We might adjust this based on feedback. This curve works well on the Lemur Pro, but might need to start the fan at lower temperatures on the Darter Pro

https://github.com/system76/ec/blob/master/src/board/system76/common/peci.c#L46

matrix8967, 2020-09-23:
@jackpot51 Awesome - that makes sense. It was a very new development so I wanted to holler just in case.

I've started from a cold-start without the 65W charger plugged in.

As of right now it's not really moving past 45C. (but It's also just sitting at a desktop...) 

Still warmer than it usually is.

Is there documentation I can glance at real quick to start adjusting the fan-curve so that we can have _heated_ debates about which is better? :smirk:

I'd be a liar if I said I knew all about this level of `C/Rust`. [Quick glance shows](https://github.com/system76/ec/blob/master/doc/debugging.md) I'll need to:

* adjust the curve :point_up: 
* compile
* flash
* monitor.
* Rinse / Repeat?

bryanpaget, 2020-09-24:
Success! 
[image]

rchdz, 2020-09-24:
Successfully upgraded to 2020-9-22, both sticks of 16GB installed back in and boots normally.

npmenard, 2020-09-24:
The update didn't work for me, here is my [image]

jackpot51, 2020-09-24:
@npmenard it looks like another key than Enter was pressed when the prompt happened. Can you try again, and make sure that you don't press any keys except for Enter?

npmenard, 2020-09-24:
@jackpot51 Yes were right Itried again and it worked. However on my system suspend on disk and keyboard backlight have stopped working. I imagine the later may be a driver issue, for the former when I open the lid the sreen lit up but stay black I have to hard reboot to get back into Linux. Linux doesn't log any "wake-up" event. Reverting back to de5eab8 solves both problems

jackpot51, 2020-09-24:
@npmenard you need system76-acpi-dkms for keyboard backlight. I'm not sure what is going on with suspend. I'm thinking these are Arch issues. What kernel are you running?

npmenard, 2020-09-24:
@jackpot51 I am running 5.8.10-arch1-1 I agree with you it's probably an Arch issue I will try PopOS later tonight to see if i can replicate the it.
Also in the previous versions of BIOS/EC the keyboard backlight was cycling through all colors while the BIOS was loading, with this version this doesn't happen but I am not sure if it is related.

jackpot51, 2020-09-24:
We don't have keyboard color cycling in open EC prior to OS initialization. It wasn't deemed tasteful. If you have system76-acpi-dkms, you will get the keyboard backlight working in the OS.

npmenard, 2020-09-24:
@jackpot51 For some reason I had blacklisted system76_acpi, I can now control the keyboard backlight. Suspend is still broken however I will try to investigate this further! Thanks again

jackpot51, 2020-09-24:
I tried a newer 5.8 kernel as well and saw suspend issues.

leviport, 2020-09-28:
@jackpot51 were these the sorts of suspend issues you were seeing? I ran into it twice over ~30 manual suspends. Every now and then when I suspend, it looks like the login screen just goes bonkers. I can still log in and everything goes back to normal though.

[image]

leviport, 2020-09-28:
Update: I've seen the same GDM issue on the old proprietary EC. If that is the only suspend issue, we can safely say it's not a regression that comes from open EC.

jackpot51, 2020-09-29:
No, I've seen on the 5.8 kernel it not resuming from suspend. Screen remains blank

leviport, 2020-09-29:
Alright, I'll keep looking then. I haven't seen that happen yet on groovy.

leviport, 2020-09-29:
After at least 50 manual suspends and 150 automated suspends with `fwts`, I still am not seeing the suspend issue on darp6 running groovy with the 5.8.0-7618 kernel.

z00nk, 2020-10-21:
Hi I am not sure if this is the right place to report this issue, but I have not been able to figure out a solution. Since the firmware upgrade to 904cee4, I have been experiencing the system not being able to charge to 100 percent, it reaches 98% and the machine starts emitting a high frequency buzzing sound, until I unplug the charging cable. The machine also gets very hot when charging, reaching 76C.  
I am not sure if this related to the firmware upgrade, but I have only begun to experience this following the upgrade. The only time I have been able to get it to charge to 100% is only when the system is in suspend mode. Thank you for reading.

jackpot51, 2020-10-21:
> Hi I am not sure if this is the right place to report this issue, but I have not been able to figure out a solution. Since the firmware upgrade to [...]

Which model is this? Can you revert to proprietary EC?

z00nk, 2020-10-21:
> Which model is this? Can you revert to proprietary EC?

Darp6. any ideas how do I do so?

matrix8967, 2020-10-21:
I've experienced the "Coil-Whine" sound since I've had my Darp, but only when using the USB-C to supply power, and not when using the barrel jack.

I've not paid attention to if the battery is ACTUALLY charging to 100%, the GUI says so but I'll monitor my own usage over the next day or so and report back.

@z00nk I believe you can find that information in the parent post of this thread - [which links here](https://github.com/system76/firmware-open/issues/127#issuecomment-696759847) for instructions for reverting.

z00nk, 2020-10-21:
Hi all, I ran `sudo system76-firmware-cli schedule --proprietary` and the firmware has reverted to the de5eab8 version. I am happy to report that on 2 of the occasions of charging (using the barrel jack), temperature has stayed constant at around 42C, and system is able to reach 100% with the LED indicator turning green. I will continue to monitor and perhaps use USB Type C to test this out. Thanks to all who have helped :)

matrix8967, 2020-12-11:
Sorry it took me so long to come back to this. :weary: 

I finally managed to get a drive for my other `Darp6`. This one has `16GB` RAM (`2x 8GB DIMMs`), but is identical outside of that. 

Firmware Upgrade went off without a hitch -- temperatures are still noticeably higher than when using the default EC Firmware, but as @jackpot51 pointed out here (https://github.com/system76/firmware-open/issues/130#issuecomment-697964352) - it's probably just the fan curve. I'll try to dig more into it next week.

As for the the suspend issue mentioned here (https://github.com/system76/firmware-open/issues/130#issuecomment-700685521) I'm seeing that _very_ consistently on my other `Darp6`.

```
Nov 12 19:07:26 $HOST systemd[1]: Reached target Sleep.
Nov 12 19:07:26 $HOST systemd[1]: Starting Suspend...
Nov 12 19:07:26 $HOST systemd-sleep[6000]: /usr/lib/systemd/system-sleep/displaylink.sh: line 7: /tmp/PmMessagesPort_out: No such file or directory
```

So - seems like (for me) the `displaylink` driver is throwing a fit about that missing file. Even when trying to create the file - the issue persists.

It's also worth noting that the display woke up when I plugged in a Thunderbolt/USB-C -> HDMI display.

I couldn't recreate the suspend issue using `fwts`. Let me know if I need to move that into a separate issue @leviport 

The root of this issue thread (`de5eab8` --> `904cee4`) looks :+1: aside from the Fan Curve.

jackpot51, 2021-05-27:
The darp6 open EC firmware was released, issues solved

closed 2021-05-27 by jackpot51