system76/firmware-open #144: galp5 (and probably lemp10) not charging through USB-C connected to VP2771 display

leviport, 2020-12-16:
I see lemp9 start charging when I connect it to these displays with USB-C, but galp5 does not. I've heard similar reports about lemp10, so I think it's doing the same thing. 

The monitor still works properly as an external display, it just doesn't charge. The plain 65W USB-C charger I have also works as expected. Other USB-C docks may have the same problem.

Dygear, 2020-12-16:
Will verify the lemp10 once I have it in hand.

Dygear, 2020-12-22:
Yeah. Having the same issue on the lemp10.

leviport, 2020-12-22:
Thank you for verifying! It's good to know that it's not only the Viewsonic display.

Dygear, 2020-12-23:
It's actually not charging at all though the USB-C port. Currently plugged in with a power adaptor and it's not charging at all. Only the barrel plug works at the moment.

leviport, 2020-12-23:
Hmm, what is the wattage of your USB-C charger? We had the galp5 and lemp10 charging off of plain 65W USB-C chargers, just not from monitors that are supposed to charge over USB-C.

Dygear, 2020-12-23:
It was a Meagoes 12V USB-C Charger (5V @ 3A). Looks like that is not enough to provide power to the unit. Weird as the Macbook Pro (15") will take a charge from that. But very, very slowly. I will have to move over to the [Satechi 72W Type C PD Car Charger](https://satechi.net/products/72w-type-c-pd-car-charger-adapter?variant=29681251582040) for when I'm on the go.

I've just plugged in my 15" Macbook Pro charger, and that is charging it without issues from that port.

cipherboy, 2021-02-27:
Just as a different datapoint, I just got the `lemp10` model and tested it with a newer revision of a listed [Lenovo dock](https://www.lenovo.com/us/en/accessories-and-monitors/docking/universal-cable-docks-thunderbolt/Thunderbolt-Dock-Gen-2-US/p/40AN0135US) and it appears to charge my laptop just fine. I started at 60% and am now at 65%, an hour left to charge according to the status info. Running Ubuntu 20.10. 

Is there any debugging info I can provide that'd help?

Dygear, 2021-03-02:
After the update to `2021-02-04_be04aea` still does not charge over USB-C using the monitor mentioned above (LG 27UD88-W).

curiousercreative, 2021-05-01:
FWIW, my galp5 is charging nicely connected to a DELL U2720Q. I never tried before today, so perhaps it's recently been fixed or perhaps this specific device is more compatible

hfavisado, 2021-10-03:
I'm seeing the same behavior in my `lemp10` connected to an LG 35WN75C.

cotsuka, 2021-10-19:
Confirming I have this issue as well. Monitors and USB devices work, but no USB-C charging. I'm using the Lenovo Thunderbolt 3 Dock Gen 2 with a `lemp10`.