system76/firmware-open #145: lemp10 & galp5: splash screen stops showing & EFI boot entries can't be written

jacobgkau, 2020-12-28:
I started experiencing an issue on the galp5. While troubleshooting, I started using the lemp10 in its place, but it later started having the issue too. We have one lemp10 customer who's reported the issue to support so far.

After doing a certain number of OS reinstalls, the system is no longer able to write new EFI boot entries. Once this starts happening, the System76 logo splash screen stops displaying and I cannot access the EFI menu with ESC (I also cannot access the systemd-boot menu with spacebar), and I can only boot into the default OS/device. When attempting to install Pop!_OS from a USB drive, `efibootmgr` fails:

```
[INFO distinst:crates/chroot/src/command.rs:108] running "chroot" "/tmp/distinst.Go3DK7ZWfOtc" "bootctl" "install" "--path=/boot/efi" "--no-variables"
[INFO distinst:crates/chroot/src/command.rs:108] running "chroot" "/tmp/distinst.Go3DK7ZWfOtc" "efibootmgr" "--create" "--disk" "/dev/nvme0n1" "--part" "1" "--write-signature" "--label" "Pop!_OS 20.10" "--loader" "\\EFI\\systemd\\systemd-bootx64.efi"
[ERROR distinst:src/installer/state.rs:37] configuring bootloader error: command failed with exit status: exit code: 5
[ERROR distinst:src/installer/mod.rs:298] errored while installing system: command failed with exit status: exit code: 5
[INFO distinst:ffi/src/installer.rs:188] Install error: command failed with exit status: exit code: 5
```

If the Pop!_OS drive is the only bootable drive in the system, then the system boots into it without the boot entry, but this doesn't work for some other operating systems (Arch Linux had an error installing GRUB, and was unable to boot.) When attempting to run `efibootmgr` manually, I get `Could not prepare Boot variable: No such file or directory`. `bootctl install` gives `Failed to write 'LoaderSystemToken' EFI variable: No such file or directory`. (GRUB's error message was slightly different, and I thought I saw "No space left on device" at least once.)

Clearing the CMOS does not resolve the issue, although from what I saw, https://github.com/system76/coreboot/pull/43 was not incorporated yet in the firmware for these models (the existing boot entries stayed there after resetting.) Removing the old (no longer valid) boot entries manually had no effect on installing the new boot entry. Attempting to re-flash current firmware with a USB drive shows a blank screen in the firmware updater:

[image]

I was eventually able to flash firmware with updated `coreboot` (with the changes to clear `SMMSTORE` on CMOS reset) on the lemp10 while booted into Pop!_OS, but after doing that, that machine is just showing a completely blank screen when powered on (so I'm not going to repeat that on the galp5.)

leviport, 2020-12-29:
This looks exactly like what the UMA galp5 did to me before I swapped it with one Jeremy had.

crawfxrd, 2021-01-05:
I tried

```
for i in {1..200}; do
    sudo efibootmgr --create --disk /dev/nvme0n1 --part 1 --write-signature --label "foo$i" --loader "\\EFI\\systemd\\systemd-bootx64.efi"
done
```

It eventually starts returning

> Could not prepare Boot variable: Invalid argument

Deleting Boot variables seems to work, but no new Boot variables can be created. BootOrder cannot be modified either.

crawfxrd, 2021-01-11:
I started seeing `EINVAL` when deleting variables:

```
$ sudo efibootmgr -B -b 0119
Could not delete variable: Invalid argument
```

But coreboot logs show it is "deleted" (appended new list with entry removed):

```
coreboot-2021-01-06_ad2d818-dirty Mon Dec 14 22:04:44 UTC 2020 smm starting (log level: 7)...
SMI_STS: PERIODIC PM1 APM 
Appending into SMM store
FMAP: area SMMSTORE found @ 610000 (262144 bytes)
used smm store size might be 0xcc9c bytes
used size looks legit
open (cc9c, 33364) for writing
```

And efibootmgr/efivar no longer shows the deleted entry.

Need to dig more into the EFI libs to see where it's getting `EINVAL` from.

Still have not seen `EIO` using this method.

renau, 2021-01-14:
I have the same issue in a lemp10. It does not allow to add more efibars in the NVRAM. I even unplug the batter/clock but the same was there.

I also tried booting and using the chattr -i to allow removing the variables, still complains about lack of space for NVRAM which is not logical because there are very few bars. I think that it is a "garbage collect" problem at the NVRAM leaving empty space (or fragmentation) and it does not allow to create new efi (NVRAM) vars.

I can "survive" until the patch, but stuck with default boot option (I can change the /boot/loader/loader.conf to select kernel updates if needed)

jbonomi, 2021-01-14:
Is it the case that once affected by this issue the firmware cannot be updated using software tools? That's my current assumption but I would really, really love to be wrong!

crawfxrd, 2021-01-14:
@renau Removing the CMOS battery will not have any effect as system76/coreboot#43 is not released in any firmware. SMMSTORE is append-only, so the only way to reclaim space is to erase the entire region in the ROM and rewrite only the needed vars (e.g., what [firmware-smmstore](https://github.com/system76/firmware-smmstore) does).

@jbonomi At the least, anything that can clear the SMMSTORE region in the ROM from the running system will work (e.g., `scripts/flash.sh` to flash a new ROM).

crawfxrd, 2021-01-14:
It may be useful to provide a dump of the SMMSTORE region for comparison. This can be done by running `scripts/smmstore.sh`. It will create a 256k file called `smmstore.rom`.

crawfxrd, 2021-01-14:
I was able to reproduce the issue using the SMMSTORE ROM provided by Jacob on galp5 and oryp6 (although I still get `EINVAL` in user space).

```
sudo ./libs/intel-spi/target/release/examples/read
sudo ./coreboot/util/cbfstool/cbfstool ./read.rom write -r SMMSTORE -f ~/SMMSTORE.rom
sudo ./libs/intel-spi/target/release/intel-spi ./read.rom
sudo reboot
```

oryp6 coreboot+edk2 boot log: [debug.log](https://github.com/system76/firmware-open/files/5816609/debug.log)

crawfxrd, 2021-01-15:
A custom flash utility has been made to reflash the current firmware to work around this issue. These should only be used on galp5 and lemp10 units experiencing this issue.

- Galago Pro: [galp5-76ec.zip](https://github.com/system76/firmware-open/files/5822010/galp5-76ec.zip)
- Lemur Pro: [lemp10-76ec.zip](https://github.com/system76/firmware-open/files/5822011/lemp10-76ec.zip)

To use:

- Extract the ZIP to a FAT32 volume on a GPT partitioned USB drive
- Plug in the USB drive
- Boot the machine

If the machine does not boot into the firmware update utility, remove all internal drives and retry.

Things to be aware of:

- It will flash automatically, without user input
- It will remove all custom EFI variables (such as boot  entries made by boot managers)

jbonomi, 2021-01-15:
Thanks @crawfxrd, that worked on my galp5. I'll keep an eye out for updates.

adeptg, 2021-01-17:
Thanks @crawfxrd, that worked on my galp5!

ZeddieXX, 2021-01-17:
Strange.  I experienced this with my first GALP5, had it RMA'd.  After that I read about this, so I limited OS installs - was booting fine until just now.  I didn't do any new OS installs but I am  now experiencing the no System 76 logo splash and cannot go into the firmware.

Prior to this, the splash screen shows including the message to hit ESC to go to coreboot, BUT hitting ESC does nothing.  It's like it doesn't even read the keyboard (both the built-in and the external USB keyboard).  Frustrated, instead of doing a reboot (which I was doing the first few times), I shut it down.  Then I turned it back on, but this time, no splash screen at all!

I'm going to try this fix, but I was surprised to see this happen to my GALP5 which was working fine until now (and no new OS install in between reboot sessions).

renau, 2021-01-18:
I updated the bios and it worked like a charm.

crawfxrd, 2021-01-19:
At the point where firmware-smmstore should run, galp5 logs:

```
Error: Image at 00075DB9000 start failed: Device Error
```

crawfxrd, 2021-01-19:
`SMMSTORE_CMD_CLEAR` was broken when SMMSTOREv2 added in coreboot, so the store was never compacted on boot by firmware-smmstore. The fix has been submitted upstream as [CB:49724](https://review.coreboot.org/c/coreboot/+/49724) and cherry-picked to our fork.

This issue will be fixed in the upcoming firmware release.

ZeddieXX, 2021-01-30:
Okay, this is the 2nd time it happened - I am having this issue right now and I didn't change OS. Simply rebooting, shutting down, starting, etc. I also use an eGPU on occasion (TB3). I don't know if that would add to the issue (I don't see why it should).

Right now, I am getting the issue where it won't respond to ESC key, but logo is showing. If it is like the last time, eventually the logo will completely go away.

leviport, 2021-02-01:
>Right now, I am getting the issue where it won't respond to ESC key

Just on the built-in keyboard? Have you tried a USB keyboard yet?

ZeddieXX, 2021-02-01:
> > Right now, I am getting the issue where it won't respond to ESC key
> 
> Just on the built-in keyboard? Have you tried a USB keyboard yet?

Yes I have.  It does not respond to ESC on USB keyboard nor the built-in keyboard.

leviport, 2021-02-01:
Does unplugging and re-plugging the charger before booting have any effect?

crawfxrd, 2021-02-01:
Keyboard won't work in this state. I didn't dig into it too much, but edk2 fails to register the console protocol, resulting in all input (keyboards) and outputs (screen) to not work.

crawfxrd, 2021-02-08:
Fixed in 2021-02-04_be04aea for lemp10.
Fixed in 2021-01-19_d6d37c0 other models.

closed 2021-02-08 by crawfxrd

crawfxrd, 2021-02-10:
Some users have reported having this issue on oryp6. (I don't know how; It's not affected by the same bug as galp5/lemp10.) So here are the files for oryp6 as well.

- Oryx Pro: [oryp6-76ec.zip](https://github.com/system76/firmware-open/files/5959839/oryp6-76ec.zip)

vansweej, 2021-02-20:
Hi,
I am experiencing the same problem on my oryp6. I downloaded the zip and installed as instructed. Rebooting and pressing esc made me come in the boot selection menu, but I do not see the usb stick there, It also never sees a regular linux stick. I can only boot the default pop os

crawfxrd, 2021-02-20:
To confirm, you are _not_ able to flash to 2021-01-19_d6d37c0, and are still on 2020-10-19_d5deed5?

What is the command you are running and the exact message you are seeing?

> pressing esc made me come in the boot selection menu

Is that in the firmware menu or systemd-boot?

If you can reach the BIOS boot menu then you are not experiencing the issue that affected lemp10/galp5. If you are instead entering the systemd-boot menu then you won't see USB drives.

Additionally, drives must have an EFI system partition with `\EFI\BOOT\BOOTX64.EFI` present in order to boot them (so legacy boot is not supported).

>  but I do not see the usb stick there

Are you powering on the machine with the USB already plugged in? There is a known issue that edk2 will not recognize some USB flash drives if they are plugged in during power on. If you instead plug them in after entering the BIOS menu, then edk2 will correctly initialize and enumerate it.

vansweej, 2021-02-20:
> To confirm, you are _not_ able to flash to 2021-01-19_d6d37c0, and are still on 2020-10-19_d5deed5?

Yes I am
> 
> What is the command you are running and the exact message you are seeing?
error I am getting is : error in system76-firmware client: calling Schedule method failed: "failed to add boot entry: exit code: 5 
after trying to run the upgrade from the settings panel
> 
> > pressing esc made me come in the boot selection menu
> 
> Is that in the firmware menu or systemd-boot?

I have never seen a systemd-boot on this machine.

> If you can reach the BIOS boot menu then you are not experiencing the issue that affected lemp10/galp5. If you are instead entering the systemd-boot menu then you won't see USB drives. [...]

according to instructions the stick has a GPT style partition table. One partition is created and fat32 formatted, settings are boot, esp for this partition. Then the contents of the zip is copied to it. Which contains the directory and file you mentioned above.

> Are you powering on the machine with the USB already plugged in? There is a known issue that edk2 will not recognize some USB flash drives if they are plugged in during power on. If you instead plug [...]

Normally I plug my stick in before power on. That is what i always do when booting linux installers. I tried plugging it in after boot to BIOS menu and the stick is never detected.

crawfxrd, 2021-02-20:
Can you clone this repo, install the dependencies (`./scripts/deps.sh`), and then run `./scripts/smmstore.sh`?

This will read the current firmware image to `read.rom` (16 MiB), extract the SMMSTORE region to `smmstore.rom` (256 KiB), then parse and print the region data.

I do not need the output it prints, but if you can provide the `smmstore.rom` image I can investigate this further.

vansweej, 2021-02-22:
running the ./scripts/deps.sh gives me this error: ./scripts/deps.sh: line
42: syntax error near unexpected token `then'

I am a rust developer and saw you installed a rust version for this too.
Shouldn't I do this as a separate user, to not touch my "production" rust
environment ?

Jan

crawfxrd, 2021-02-22:
> Shouldn't I do this as a separate user, to not touch my "production" rust environment?

It's not necessary, as rustup manages multiple toolchain installations.
However, it was changing the default toolchain to nightly. This has been fixed.

vansweej, 2021-02-23:
smmstore.rom as asked

crawfxrd, 2021-02-23:
You will need to upload the file through the UI. Email attachments do not work.

vansweej, 2021-02-23:
does this works ?

[smmstore.zip](https://github.com/system76/firmware-open/files/6032043/smmstore.zip)

crawfxrd, 2021-02-24:
I am able to reproduce the issue using your SMMSTORE ROM. It appears to be caused by corrupted `ErrOut` variables.

You will need to remove the internal drives so that the USB is the only bootable device.

vansweej, 2021-03-13:
Hi,
Just to notify you I managed to successfully flashed the bios of my Oryx
Pro.
Thx a lot for the support.

Jan

crawfxrd, 2025-07-23:
Create a new issue with details. You do not have the issue reported here.