system76/firmware-open #148: galp5 w/o dGPU should still use GPU fan

curiousercreative, 2021-01-19:
The fan still cools a heatsink connected to the CPU and is effective. I understand that for some, thermal performance is not essential and therefore this would be great as an easily enabled feature rather than a default perhaps.

ZeddieXX, 2021-02-01:
I would love to know if this is being reviewed to be added to the mainline firmware for the next update.  Otherwise, I have two questions:

1. If we compile and flash ourselves, how would this affect the firmware update notification mechanism in Pop!_OS since this is not an official firmware version?
2. What are the step-by-step instructions for people who don't have experience with git, compiling a firmware, and flashing firmware?
3. Do we have to repeat #2 every time a new official firmware update comes out if the firmware update notification is broken with custom firmware?  (Assuming this will not be merged into the official firmware updates).

curiousercreative, 2021-10-20:
fixed in https://github.com/system76/ec/pull/177

closed 2021-10-20 by curiousercreative