system76/firmware-open #149: laptop fan speed changes should be smoother

curiousercreative, 2021-01-19:
With my new galp5 i7-1165G7, fan will abruptly blare to 100% when under occasional load. Under consistent but varied loads, fan may jump between each fan point every so often. It would be nice to smooth these fan ramps out to avoid the unpleasant audible stepping in fan speeds.

curiousercreative, 2021-01-27:
rofl @ https://github.com/gdoteof/system76-galapago-pro-fan-unfucker

jackpot51, 2021-01-27:
This code would only work with the proprietary EC, which shipped on previous models and was far worse.

ZeddieXX, 2021-02-01:
@jackpot51 - Will this be incorporated into the mainline code and be offered as an update through standard channels?  I am not comfortable compiling and flashing the firmware.  Firstly, I don't know if this will break the mechanism that Pop!_OS uses to determine if your firmware version is out-of-date and needs an upgrade, since this will not be an official build of the firmware.

https://github.com/system76/ec/pull/139

curiousercreative, 2021-02-01:
@ZeddieXX I can't say whether this will be accepted upstream, but my guess is that the firmware updater won't care about what we're doing here. It seems that the flashing we're doing here isn't updating the firmware (which consists of multiple components) but just a single firmware component (the embedded controller). I was similarly hesitant to flash the EC, but it wasn't so bad. 
These instructions for flashing from the [doc for keyboard mapping](https://github.com/system76/ec/blob/master/doc/keyboard-layout-customization.md) is what I followed, but first you need to clone [this repo](https://github.com/system76/firmware-open) and [install the dependencies](https://github.com/system76/firmware-open/blob/master/scripts/deps.sh). Once you've got that going, you can fetch my branch, switch to it and flash your EC. Here's more of a step-by-step (no warranty!):
1. Clone this repo
    ```bash
    git clone git@github.com:system76/firmware-open.git
    ```
1. Install dependencies (they stalled at some point for me, but it installed enough)
    ```bash
    cd firmware-open
    ./scripts/deps.sh
    source ~/.cargo/env
    ```
1. Fetch my branch and switch to it
    ```bash
    cd ec
    git fetch git@github.com:curiousercreative/ec.git galp5
    git checkout galp5
    ```
1. Configure the build for our galp5, you can optinally add a custom keyboard mapping by editting this new "config.mk" file
    ```bash
    echo 'BOARD?=system76/galp5' > config.mk
    ```
1. Now we can try flashing it. At this point, you can close all applications except your terminal. You'll be prompted for your password, your fans will go wild and it'll shutdown.
    ```bash
    make flash_internal
    ```
1. After it shuts down, power it back up and your should have smoother fans.

Find me on https://chat.pop-os.org/ as curiouser if you need more help :)

ZeddieXX, 2021-02-04:
I have a problem with the very first line.

`zeddie@pop-os:~/temp$ git clone git@github.com:system76/firmware-open.git
Cloning into 'firmware-open'...
The authenticity of host 'github.com (140.82.114.4)' can't be established.
RSA key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'github.com,140.82.114.4' (RSA) to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
`

curiousercreative, 2021-02-04:
@ZeddieXX https://docs.github.com/en/github-ae@latest/github/authenticating-to-github/generating-a-new-ssh-key-and-adding-it-to-the-ssh-agent

kepelrs, 2021-04-14:
Hi,

I own a lemp10 and share similar issues as @curiousercreative. While in my case the fan does not jump quite to 100%, it remains at 0% until the laptop reaches 70C, at which point the fan starts to repeatedly turn on and off. This all can be caused by very light usage (eg. browsing), and there's nothing I can do about it (power saver does not fix it).

I am of the opinion that everyone would benefit if system76 implemented a way for users to control the fan via the OS. A few points why I feel that way:
1. Most people (even some power users) will be turned off by the work involved in having to flash their firmware multiple times in order to experiment with the fan curve
2. Everyone has different needs and preferences, it is unlikely that system76 will design a fan curve that pleases every use case for each laptop model/configuration; therefore
3. If the fan could be controlled at OS level:
    1. power users could design new or migrate their own fan scripts from other laptops, without having to flash repeatedly
    2. community could develop GUI tools to allow less code savy customers to adjust fan behavior to their liking
    
I read somewhere that system76 had plans in this direction, wondering if that is still the case?

MilesBHuff, 2021-04-20:
@kepelrs There's an issue for it here:  https://github.com/pop-os/system76-acpi-dkms/issues/9

closed 2021-06-14 by jackpot51