system76/firmware-open #152: lemp9: ACPI BIOS Error (bug) notice at boot

metaphys, 2021-02-10:


bflanagin, 2021-02-10:
Looking into it now. Do you have any external hardware connected to your Lemur? Docking station,external monitor,etc.

metaphys, 2021-02-10:
I have an external keyboard that is nearly always plugged in. (Ouch I don’t remember but I might very well have let it connected while performing the update. Not smart…)
yeah the system react correctly to ac being disconnected/reconnected

bflanagin, 2021-02-10:
@metaphys nah, I don't think having it connected during the firmware update matters, I was more concerned about any external displays given whats being reported by the system. 
I'll keep testing but I'm not seeing the error on my lemp9. When does the message appear on your machine?

metaphys, 2021-02-10:
this happen right after kernel selection in grub and before grub ask for my partition password.

metaphys, 2021-02-10:
I’m having this on debian with the kernel from the ubuntu repo. Are there maybe some specific linux parameter that should be pass through grub?

crawfxrd, 2021-03-22:
Were these fixed with the 2021-03-11 release?

metaphys, 2021-03-22:
absolutely, thank you very much!

closed 2021-03-22 by crawfxrd