system76/firmware-open #153: Oryx Pro 6 cannot boot anymore into PopOS

elleven11, 2021-02-12:
Hello I just updated to the 2021-01-19_d9d37c0 firmware and I cannot boot anymore into PopOS.
I can boot however my arch installation on my other nvme (i use both nvme slots on my oryx pro) and it works like expected. And I can also boot into my windows install.
I also noticed I couldn't boot into my live kali usb but could boot into my live arch usb.
I can also mount the nvme drive that popos is on and chroot into it. I am very confused about this.
How can I revert back to the previous firmware?

Thank you.

elleven11, 2021-02-12:
In the meanwhile, I was able to manually load the EFI file of the OS using an EFI shell, so at least I am able to somewhat use my computer. 
It seems that the OS is not finding the grubx64.efi file.

crawfxrd, 2021-02-12:
Is the Windows install on the same drive as Pop!_OS?

Do you have `\EFI\BOOT\BOOTX64.EFI` on the drop with Pop!_OS?

elleven11, 2021-02-12:
Yes I do, but this shouldn't be a problem usually. And even if i remove the efi file it does not find it. i think the problem is that the bootloader does not find grubx64.efi files but it does find bootx64.efi files.
```
[elleven:/boot/EFI]$ pwd
/boot/EFI
[elleven:/boot/EFI]$ ls
arch  Boot  kali  Microsoft  popos
```

elleven11, 2021-02-12:
I use the same partition as the EFI one for all of my distros.

crawfxrd, 2021-02-12:
> I also noticed I couldn't boot into my live kali usb 

Does this have an EFI system partition with `\EFI\BOOT\BOOTX64.EFI`?

> Yes I do, but this shouldn't be a problem usually.

Our edk2 fork looks for this file to determine if it is a valid ESP; it is required in order for the drive to be enumerated in the firmware menu.

If you add the volume in NVRAM (such as by using efibootmgr) it will show up in the firmware boot menu.

Otherwise, if you boot the grub menu it should be listed there.

---

> How can I revert back to the previous firmware?

You can create a support ticket and they should be able to provide a ZIP to install the previous firmware image.

Or, if you are comfortable with building and flashing from this repo, you can do it yourself; The previous firmware for oryp6 was d5deed5.

elleven11, 2021-02-13:
Yes, and that BOOTX64.EFI seems to be the windows one.
Good one, I will add it in the NVRAM.
But i think that this should be an issue, I have been doing this setup for years now on many systems and I have never encountered a problem.

elleven11, 2021-02-13:
Problem solved! Idk why the entry got deleted from the nvram. The firmware updater deleted it maybe?
But thank you soo much @crawfxrd

closed 2021-02-13 by elleven11