system76/firmware-open #155: darp6 cannot be left alone

bryanpaget, 2021-02-13:
Ever since I updated the firmware (2021-01-19_d6d37c0) on my darp6 I have run into the following problem:

If I leave my computer alone for an extended period of time, the fan kicks up and the computer becomes stuck in a weird zombie state. I have to manually power it down.

I completely disabled suspend but it still does this.

Please advice on how to troubleshoot. 

Much love, Bryan.

bryanpaget, 2021-02-17:
I'd love to test out a different OS to see if it's hardware vs OS related but I can't seem to boot from anything other than the internal NVMe drive. Not sure when that became an issue.

jacobgkau, 2021-02-17:
@bryanpaget I'm sorry about the trouble with your Darter Pro. For System76 hardware, opening a [support ticket](https://support.system76.com/) is the most reliable way to get support.

> I can't seem to boot from anything other than the internal NVMe drive.

What happens when you try to boot from something else? Do you see the System76 splash screen when booting the machine? Do you see the `Press ESC for Boot Options/Settings` text at the bottom of the screen? Are you able to enter the boot options and nothing else shows up, or are you not able to enter the boot options?

bryanpaget, 2021-02-18:
I am able to enter the boot options but only the NVMe is available. This was not a problem in the past. I think my laptop is no longer under warranty but I posted here thinking it was something related to the firmware update.

bryanpaget, 2021-02-21:
> If I leave my computer alone for an extended period of time, the fan kicks up and the computer becomes stuck in a weird zombie state. I have to manually power it down.

Actually I think this was an issue with the Brave web browser. I've stopped using it. I think Brave was somehow causing the laptop to overheat.

bryanpaget, 2021-02-23:
Ooops nevermind. It l

bryanpaget, 2021-02-24:
I think it's just overheating because the web is so demanding. Closing issue since it's not the computer's fault.

jackpot51, 2021-02-24:
Hey @bryanpaget, it would be helpful if we could identify why it is overheating and fix it in firmware. Since you are on System76 EC it is possible to run a script to log the power and temperature values:

```
git clone https://github.com/system76/ec.git ec
cd ec
./scripts/power.sh
# Replicate the issue and provide the power.csv file that is generated
```

bryanpaget, 2021-02-26:
Here we go. I also attached some plots from Jamovi.
[power.zip](https://github.com/system76/firmware-open/files/6050590/power.zip)

bryanpaget, 2021-02-26:
Oh, I forgot to mention: I booted the computer, opened Firefox (with 10 tabs open) and left the laptop alone, plugged in (USB C power) for 17 minutes.

bryanpaget, 2021-02-26:
How does the firmware control fan speed as a function of laptop heat?

jackpot51, 2021-02-26:
How, specifically, is determined by https://github.com/system76/ec/blob/master/src/board/system76/common/peci.c

bryanpaget, 2021-02-26:
I'm going to try to understand this code... but let's be clear I'm just a data scientist / statistician... 

Maybe I'll start by asking some questions. I see some numbers in this code related to temperature and fan speed. Just wondering how were these numbers derived? If I collect enough data can I train an ML algorithm, something simple like an SVM, to find optimal fan speeds? Sorry if I sound ignorant, I'm interested in this stuff but I'm also out of my (current) element.

jackpot51, 2021-02-26:
I don't think the problem is the EC fan code. It sounds like there's an OS crash that causes the CPU to spin and heat up causing the system to lock up and the fan to spin at max

bryanpaget, 2021-02-26:
Oh, what about the issue where I can't boot from anything other than the built-in NVMe drive? Is that firmware related?

jackpot51, 2021-02-26:
That certainly could be. And it could be related to the hang. What is the output of:

```
efibootmgr -v
```

bryanpaget, 2021-02-26:
```
bryan@darterpro2019:~$ efibootmgr -v
BootCurrent: 0001
Timeout: 2 seconds
BootOrder: 0000,0001
Boot0000* UiApp	MemoryMapped(11,0x840000,0xffffff)/FvFile(462caa21-7614-4503-836e-8ab6f4662331)
Boot0001* Samsung SSD 970 EVO Plus 500GB 	PciRoot(0x0)/Pci(0x1d,0x4)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-55-91-B4-C9-83)N.....YM....R,Y.

```

bryanpaget, 2021-02-26:
Is the data useful? I can continue collecting more.

jackpot51, 2021-02-26:
Have you tried doing a CMOS reset by turning the system off an unplugging the CMOS battery?

bryanpaget, 2021-02-26:
I have not tried that. I can do that.

bryanpaget, 2021-02-26:
Dude that worked. I can boot from usb thumb drives now.

jackpot51, 2021-02-26:
Great! I think the NVRAM was hosed at some point. Let me know if that fixes the hanging with full fan issue

bryanpaget, 2021-02-26:
I remember having to reset different kinds of RAM on my Mac back in the day, is there a way to do that with Linux instead of opening the case?

jackpot51, 2021-02-26:
@bryanpaget technically yes, but I think it would have taken more time to do so. We would make edits to the flashing program, rebuild the firmware, and flash it with the edited flashing program that clears the NVRAM.

bryanpaget, 2021-02-26:
Awesome. I'll let you know.

bryanpaget, 2021-03-01:
Three days later:

I am able to boot into Recalbox and play all sorts of games with no freezing. So the hardware is totally good!

I still get the "slow/missing keys at boot when decrypting my internal NVMe", as mentioned in [156](https://github.com/system76/firmware-open/issues/156). So there may still be something funky about the firmware? Not sure. I have not tested it enough but I think the missing/slow keys issue is exacerbated by heat.

Sometimes my GNOME session still freezes. I think this may purely be a software issue (probably 100% my fault). I'll continue to investigate.

curiousercreative, 2021-03-09:
Should this be considered closed?

bryanpaget, 2021-03-12:
Yes. I recently rolled back to Pop!_OS 20.04 and my system is no longer freezes. The freezing must have been a software only problem. Thank you again @jackpot51 for helping me reset the NVRAM.

bryanpaget, 2021-03-17:
Actually, I

bryanpaget, 2021-03-17:
No, I will do that now.

bryanpaget, 2021-03-17:
Alright, still running memtester but loop 1 finished with:

```
Loop 1/5:
  Stuck Address       : ok         
  Random Value        : ok
  Compare XOR         : ok
  Compare SUB         : ok
  Compare MUL         : ok
  Compare DIV         : ok
  Compare OR          : ok
  Compare AND         : ok
  Sequential Increment: ok
  Solid Bits          : ok         
  Block Sequential    : ok         
  Checkerboard        : ok         
  Bit Spread          : ok         
  Bit Flip            : ok         
  Walking Ones        : ok         
  Walking Zeroes      : ok         
  8-bit Writes        : ok
  16-bit Writes       : -FAILURE: 0x7fbd7ffcff3f78e8 != 0x7fbd7ffcff3f78f8 at offset 0x2c43da998.

```

jackpot51, 2021-03-17:
That is not good news, it means there was a bit flip. How much RAM do you have? I can probably tell you which DIMM to remove based on the failed address. You can work with support to get replacement parts, let them know I sent you and link to this issue.

bryanpaget, 2021-03-17:
Oh! It's good news to me in the sense we are zeroing in on something! Nothing feels worse than not knowing!

I have 32GB of RAM.

jackpot51, 2021-03-17:
32 GB on two DIMMs? This looks like the first DIMM, since that address is at around 11 GB.

bryanpaget, 2021-03-17:
Yep.

```  
*-memory
       description: System Memory
       physical id: 9
       slot: System board or motherboard
       size: 32GiB
     *-bank:0
          description: SODIMM DDR4 Synchronous 2667 MHz (0.4 ns)
          product: GKE160SO102408-2666
          vendor: Unknown (4589)
          physical id: 0
          serial: 16161616
          slot: Channel-0-DIMM-0
          size: 16GiB
          width: 64 bits
          clock: 2667MHz (0.4ns)
     *-bank:1
          description: SODIMM DDR4 Synchronous 2667 MHz (0.4 ns)
          product: GKE160SO102408-2666
          vendor: Unknown (4589)
          physical id: 1
          serial: 16161616
          slot: Channel-1-DIMM-0
          size: 16GiB
          width: 64 bits
          clock: 2667MHz (0.4ns)
```

Okay. I'll pop it out and see if that resolves the freezing. It's out of warranty so maybe I'll just live with 16GB, probably fine. Not really sure why I got 32GB. haha

jackpot51, 2021-03-17:
Should probably let the memory test finish, just to make sure the second DIMM is okay

bryanpaget, 2021-03-17:
Good point. I was just going to say this test does not preclude the second DIMM from being faulty.

bryanpaget, 2021-03-18:
memtester finished without any other errors, i.e.

```
Loop 2/5:
  Stuck Address       : ok         
  Random Value        : ok
  Compare XOR         : ok
  Compare SUB         : ok
  Compare MUL         : ok
  Compare DIV         : ok
  Compare OR          : ok
  Compare AND         : ok
  Sequential Increment: ok
  Solid Bits          : ok         
  Block Sequential    : ok         
  Checkerboard        : ok         
  Bit Spread          : ok         
  Bit Flip            : ok         
  Walking Ones        : ok         
  Walking Zeroes      : ok         
  8-bit Writes        : ok
  16-bit Writes       : ok
```
and so on for loops 2, 3, 4 and 5.

I'll remove the first DIMM. According to the picture below, the DIMMs are numbered, is that correct?

[image]

jackpot51, 2021-03-18:
Yep, remove RAM1

bryanpaget, 2021-03-21:
I'm back from running experiments. I have good news! 

## Experiments

I'd put one stick in one slot, boot up and then run `memtester 13G 5` and wait. 

At first it looked like there was a problem with the slot since the "Bad DIMM" would perform just fine in slot 2 (see Experiment 2). 

Since I don't know much about hardware I figured I'd run the experiment multiple times.

The worst was Experiment 5 when the machine wouldn't even boot! I think I gave it around 2 minutes to boot but it just sat there with a black screen. 

After experiment 5 I started wiggly the DIMMs around in the slots to make sure they were seated correctly and that wiggling seems to fix the problem. 

Experiment 9 seems to indicate the problem is gone. I'm going to leave the "Bad DIMM" for good luck.

|Experiment|Slot 1   |Slot 2   |Results   |
|----------|---------|---------|----------|
|1         |Bad DIMM |Empty    |FAIL      |
|2         |Empty    |Bad DIMM |OK        |
|3         |Good DIMM|Empty    |OK        |
|4         |Empty    |Good DIMM|OK        |
|5         |Bad DIMM |Empty    |Can’t Boot|
|6         |Empty    |Bad DIMM |OK        |
|7         |Bad DIMM |Empty    |OK        |
|8         |Empty    |Bad DIMM |OK        |
|9         |Good DIMM|Bad DIMM |OK        |

I didn't record the exact output of `memtester` just whether or not there was an error.

Maybe there was dust in slot 1 and by rotating the DIMMs around I inadvertently cleaned it? Does this sound reasonable?

jackpot51, 2021-03-21:
Not sure. If it freezes again, try removing the "Bad DIMM". Otherwise, awesome.

bryanpaget, 2021-03-29:
Perfection. No freezing. If things go funny again I'll just get new RAM. Thanks for the help!. :-)

closed 2021-03-29 by bryanpaget