system76/firmware-open #157: galp5 4k @ 120Hz HDMI TV capped at 30Hz

jackpot51, 2021-02-15:
HDMI 2.0 should support 4k @ 60 Hz, HDMI 2.1 should support 4k @ 120Hz. Confirm that the HDMI port is configured correctly

leviport, 2021-02-15:
Let me know if you need anything tested. I have a 4K 60Hz display and an HDMI 2.1 cable, as well as the UMA galp5 and the 1650Ti galp5.

jackpot51, 2021-02-15:
@leviport please check what framerate you can select with those two units

leviport, 2021-02-15:
I can't select 60Hz on either.

1650Ti:
[image]

UMA
[image]

I'm going to try to confirm that this setup will do 60Hz over HDMI at all. It is an older display, and my HDMI 2.1 cable wasn't particularly expensive.

jackpot51, 2021-02-15:
@leviport @jacobgkau @thomas-zimmerman I believe this issue also affects the darp7. But not the lemp10. Can this be confirmed?

I will prepare a firmware update that should fix both

jacobgkau, 2021-02-15:
Confirmed, lemp10 lets me use a 4k display at 60Hz, darp7 and galp5 currently limit to 30Hz. Testing the firmware updates now.

closed 2021-02-16 by jackpot51