system76/firmware-open #160: Oryp6: nvidia-smi hangs while in compute mode

leviport, 2021-02-18:
In Pop!_OS 20.10, if I switch to the Compute graphics mode and reboot, running `nvidia-smi` hangs completely. This is with the current Nvidia 460.39 driver.

jackpot51, 2021-02-26:
I'm pretty sure this will be fixed by https://github.com/pop-os/ubuntu-drivers-common/pull/8

chriscardillo, 2021-03-18:
Hi - has this been incorporated into a firmware update yet? I am having trouble following.

leviport, 2021-03-18:
This is unrelated to firmware, but yes, that PR was merged and `nvidia-smi` doesn't seem to be hanging in compute mode on oryp6 any longer.

jacobgkau, 2021-03-18:
I'm afraid the oryp7 is still having trouble with compute mode. While `nvidia-smi` doesn't hang, I'm getting this output with accompanying dmesg text:
```
system76@pop-os:~$ nvidia-smi
Unable to determine the device handle for GPU 0000:01:00.0: Unknown Error
```

```
[  117.004655] nvidia 0000:01:00.0: can't change power state from D3cold to D0 (config space inaccessible)
[  117.004693] snd_hda_intel 0000:01:00.1: can't change power state from D3cold to D0 (config space inaccessible)
[  117.005518] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x23:0x56:624)
[  117.005624] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[  117.006598] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x23:0x56:624)
[  117.006694] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[  117.384630] snd_hda_codec_hdmi hdaudioC1D0: Unable to sync register 0x4f0800. -5
[  117.384644] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD buf size -1
[  117.384650] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD buf size -1
[  117.384654] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD buf size -1
[  117.384659] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD buf size -1
[  117.384664] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD buf size -1
[  117.384668] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD buf size -1
[  117.384673] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD buf size -1
[  117.384677] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD buf size -1
[  117.384683] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD buf size -1
[  117.384687] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD buf size -1
[  117.384692] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD buf size -1
[  117.384696] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD buf size -1
```

I believe the full hanging stopped with an NVIDIA driver update, but this does still need attention, at least on oryp7.

chriscardillo, 2021-03-18:
This is still an issue on oryp6 in compute mode.

Now it cannot find the device anymore.

Yuri6037, 2021-06-28:
I've just tried compute mode on Oryx Pro 7 result is much much worse than what is said here: complete entire system lock up after about 1 minute.

therealartifex, 2021-08-20:
I am also seeing the "can't change power state from D3cold to D0" on Pop_OS 21.04 with nvidia 470.57 drivers. I can no longer switch to compute graphics.

crawfxrd, 2021-11-10:
Should be fixed by just waiting longer for GPU init before setting power management.

You can try without building ubuntu-drivers-common by doing the following:

- Switch to compute mode
- Reboot
- Disable GPU manager
 ```
sudo systemctl disable gpu-manager.service
```
- Reboot
- Set power management after login
```
echo auto | sudo tee /sys/bus/pci/devices/0000\:01\:00.0/power/control
```

crawfxrd, 2022-05-16:
Should be fixed with updated coreboot in fd9d73bdb9c8.

crawfxrd, 2022-12-19:
Fix released for oryp7 in 2022-11-21_b337ac6.

closed 2022-12-19 by leviport

leviport, 2023-08-23:
@devoidfury New releases are being tracked in https://github.com/system76/firmware-open/issues/340