system76/firmware-open #162: lemp9: unable to undervolt

lead4good, 2021-02-18:
Using this [undervolt tool](https://github.com/georgewhewell/undervolt), I am able to set the temperature target as well as the powerlimit, however any attempt to undervolt core, gpu etc. fails.
```
$ sudo ./undervolt.py  --core -50 --cache -50
ERROR:root:Failed to apply core: set -49.8046875, read 0.0
```
System is stock PopOS 20.10

lead4good, 2021-02-18:
I've read that many hardware manufacturers have started disabling undervolt due to the plundervolt vulnerability. How does this relate to coreboot? Could it be that the newer FSP/Microcode disables undervolting by default?

crawfxrd, 2022-07-31:
Closing in favor of #201, which has more discussion.

closed 2022-07-31 by crawfxrd