system76/firmware-open #174: Coreboot/Bios/Tianocore Password

varac, 2021-03-12:
Hi there,                                                                                                                                                                                                        
                                                                                                                                                                                                                     I'm happy with my lemp9 - the only thing I wonder about is how to secure the initial boot with coreboot/tianocore(?) with a password. Is this possible ? If not, please implement this.                                                  
Thread model is rather a random bypasser trying to boot from usbstick i.e. than a more sophisticated attack.                                                                                                                                                                                                                                                                                                                          

Thanks !

JBombNZ, 2023-08-15:
Is there an update on this? I brought a brand new Servel servw13 and i am shocked that i cant set a password.....

crawfxrd, 2023-08-15:
No. While this feature is a valid request, no one is going to work on it.

jacobgkau, 2023-08-15:
Just to elaborate, the reason it's not a priority is because BIOS/UEFI boot passwords are easy to bypass by simply moving the storage drive into a different computer or enclosure (or by resetting the firmware/firmware settings.) The files aren't actually protected unless you're using encryption on your storage drive, which Pop!_OS offers by default. And if you're using encryption on the drive, then an attacker who boots from a USB drive won't be able to access your data anyway.

ilikenwf, 2023-09-05:
> Just to elaborate, the reason it's not a priority is because BIOS/UEFI boot passwords are easy to bypass by simply moving the storage drive into a different computer or enclosure (or by resetting [...]

In addition these aren't stored on disk, they're stored in the flash itself.

cmmh, 2023-09-25:
Without a bios password, the security posture of my organization won't allow use of that machine.  This really is a critical feature.  Pretty disappointed with @crawfxrd 's assertion that no one is going to work on it.

alexispurslane, 2024-12-01:
Although locking the BIOS does not protect your drive if it is unencrypted, if it is encrypted and you have secure boot enabled, it is a crucial part of building a chain of trust, as without it, as said before, secure boot is useless. I would greatly appreciate if this was worked on eventually as well.

ilikenwf, 2024-12-01:
100% agreed.

crawfxrd, 2026-08-27:
MrChromebox has ported UserAuthenticationDxe from Dasharo to his fork.