system76/firmware-open #182: oryp7: NVIDIA TGP limited/features missing

jacobgkau, 2021-03-19:
- On oryp7 w/ Open Firmware and Pop!_OS, the GPU is limited to ~80W as reported by `nvidia-smi -q`. (This should be the board power, as opposed to the GPU chip power.)
- With proprietary firmware and Pop!_OS, it's still limited to ~80W (it may average a few watts higher from what I've seen.)
- With proprietary firmware on Windows, the NVIDIA driver reports 105W as the maximum graphics power (and I'm able to get board power up that high when benchmarking with FurMark.) Dynamic Boost 2.0 and Whisper Mode 2.0 are listed as supported.
- With Open Firmware on Windows, I'm not able to open the NVIDIA control panel (it says `You are not currently using a display attached to an NVIDIA GPU.`) According to what I heard from @jackpot51, the NVIDIA driver still reports 105W as the maximum graphics power, but Dynamic Boost 2.0 and Whisper Mode 2.0 are not supported. I'm only able to get board power up to ~80W when benchmarking with FurMark (GPU chip power ~50W.)

The FurMark (Windows) FPS is ~80 on Open Firmware and ~115 on proprietary firmware.

Starting this here because it seems there's definitely a firmware component, not sure if more will be needed in Pop!_OS if/when this is improved in the firmware.

oceanio, 2021-04-08:
I have the same issue.  

According to the output of `nvidia-smi -q -d PERFORMANCE`
```
Clocks Throttle Reasons
        Idle                              : Not Active
        Applications Clocks Setting       : Not Active
        SW Power Cap                      : Active
        HW Slowdown                       : Not Active
            HW Thermal Slowdown           : Not Active
            HW Power Brake Slowdown       : Not Active
        Sync Boost                        : Not Active
        SW Thermal Slowdown               : Not Active
        Display Clock Setting             : Not Active 
```

I've tried `nvidia-smi -pl 115`, and it outputs `Changing power management limit is not supported for GPU: 00000000:01:00.0.`

According to https://www.anandtech.com/show/15692/nvidia-details-dynamic-boost-tech-and-advanced-optimus
> By dynamically allocating power between the CPU and the GPU, a system should be able to eke out a bit more performance by making better-informed choices about where to allocate power.

So I tried to set CPU power limit to spare some power for dGPU, but power draw still restricted to 80W due to "SW Power Cap". I've also tried several other linux distribution and kernels, and other brand notebooks, none can break the power cap.

SpookySkeletons, 2021-08-12:
Same issue here with the Oryx Pro 6 RTX 2070

SpookySkeletons, 2021-09-03:
@jackpot51
The difference in performance is really stark and the locked TDP is a bit of a handicap.
Is there any word on what is missing to allow the drivers to properly throttle the card?
I can only assume from messing around a bit with OVMF passthrough there's supposed to be ACPI tables to signal to nvidia's driver that these cards are compatible but I would like to know if s76 has any leads.

Angel418, 2022-04-20:
Has there been any progress on this? It's still an issue for me.

crawfxrd, 2022-04-20:
We got documentation from NVIDIA. @jackpot51 started working on fixing GC6 (system76/coreboot#97). Nothing has been done for GPU Boost.