system76/firmware-open #183: gaze15 keyboard input and battery monitor issue on firmware 2021-03-19_af2fbdf

bflanagin, 2021-03-23:
Issue is hard to duplicate due to its seemingly random occurrence. However, once triggered the systems battery monitor can no longer estimate the time left on the battery. At the same time the input from the keyboard becomes delayed to the point where the user would be forced to power off the laptop in order to regain both battery monitoring and keyboard. 

Here is the `journalctl -b0` output: 
[keyboard_issue.txt](https://github.com/system76/firmware-open/files/6190905/keyboard_issue.txt)

The system wasn't under load when the issue happened.

jackpot51, 2021-03-23:
Can you add this to the kernel parameters:

acpi.aml_debug_output=1

bflanagin, 2021-03-24:
acpi.aml_debug_output=1 fails using `sysctl -w` with:

```
sysctl: cannot stat /proc/sys/acpi/aml_debug_output: No such file or directory
```
This is with 5.11 however the input error happened on the older kernel as well.

jackpot51, 2021-03-24:
Use `sudo kernelstub -a acpi.aml_debug_output=1` and then reboot

Serious-Table, 2021-08-13:
I've experienced this same thing on a gaze15. This has typically happened under some sort of heavy load, such as compiling Unreal Engine 4 or most recently, running No Man's Sky. I've seen this resolved by leaving the laptop for about 10 minutes, then shutting it off and rebooting. Keyboard becomes responsive again and battery estimate is displayed.

This has happened under Nvidia Graphics; I haven't tried to replicate under Hybrid yet.

Kernel is 5.13.9-xanmod1

Serious-Table, 2021-08-13:
Following the advice of support, I used the same command as recommended by @jackpot51:

`sudo kernelstub -a acpi.aml_debug_output=1`

After entering the command into Terminal and rebooting, I've attempted to replicate the issue by playing through No Man's Sky over the last hour. Thus far, I haven't experienced the issue again. I will continue to test with both No Man's Sky and other games that may require shader compilation to see if I can cause the same issue again.

Serious-Table, 2021-08-17:
[system76-logs.zip](https://github.com/system76/firmware-open/files/7000768/system76-logs.zip)

I was able to replicate this issue by simultaneously: 
* Downloading and installing Final Fantasy XIV via the XIVLauncher from Lutris, and
* Installing dotnet48 and vcrun2019 into a new prefix.

Attached are the logs generated by the System76 Driver application while the issue was occurring; I was able to bypass the problem by plugging in an external USB keyboard, which allowed the external keyboard to type normally; while the keyboard was plugged in, the on-board keyboard was still experiencing the issues described. Battery level was also displayed as "Estimating..." during this time.

bflanagin, 2021-08-20:
I was finally able to replicate the issue on my gaze15 as well. 
[keyboardIssue-08202021.txt](https://github.com/system76/firmware-open/files/7023250/keyboardIssue-08202021.txt)

electron-farmer, 2026-05-16:
## Additional reproduction on gaze15 — confirmed thermal trigger + specific ACPI methods identified

**Hardware:** System76 gaze15
**BIOS:** 1.07.11a (2020-12-02)
**OS:** Pop!_OS 24.04 LTS
**Kernel:** 6.18.7-76061807-generic

---

I've hit this issue twice today (2026-05-15), both times resulting in a hard lockup requiring a battery pull (more severe than the delayed keyboard described above — complete freeze, shutdown hangs at the bottom of the stack even after systemd reaches poweroff).

### What the kernel logs show

The failure is a cascade of `AE_TIME` timeouts on `\_SB.WMI.SCMD` and `\_SB.WMI.WMBB`, firing every 2–3 seconds for ~2 minutes before the system becomes unresponsive:

```
May 15 23:22:41 kernel: ACPI Error: AE_TIME, Returned by Handler for [EmbeddedControl] (20250807/evregion-301)
May 15 23:22:41 kernel: ACPI Error: Timeout from EC hardware or EC device driver (20250807/evregion-311)
May 15 23:22:41 kernel: Initialized Arguments for Method [SCMD]:  (3 arguments defined for method invocation)
May 15 23:22:41 kernel:   Arg0: Integer 0000000000000000
May 15 23:22:41 kernel:   Arg1: Integer 0000000000000067   ← fan control command
May 15 23:22:41 kernel:   Arg2: Buffer(4) F9 FF 00 F0
May 15 23:22:41 kernel: ACPI Error: Aborting method \_SB.WMI.SCMD due to previous error (AE_TIME)
May 15 23:22:41 kernel: ACPI Error: Aborting method \_SB.WMI.WMBB due to previous error (AE_TIME)
```

This repeats every 2–3 seconds. `Arg1=0x67` is the WMI fan control command. The EC is being hammered with fan speed requests it can't respond to within the ACPI timeout window, and once it falls behind it never recovers.

i915 DRM vblank misses also appear simultaneously — consistent with the EC monopolizing IRQ resources.

### Confirmed thermal trigger

I have an OTel → SignOz monitoring pipeline shipping `node_hwmon_temp_celsius` and `node_load*` from this machine. The data from the crash window is unambiguous:

| Time (local) | CPU temp peak (coretemp) | thermal_zone0 | load1 |
|---|---|---|---|
| T-3min | 74°C | 68°C | 2.35 |
| T-2min | **84°C** | **84°C** | 3.34 |
| T-1min | **87°C** | **84°C** | 2.68 |
| T-0 (crash) | **86°C** | **84°C** | 3.63 |
| T+1min | drops to 65°C | 82°C | 3.50 ← last metric before OTel died |

Cores were pinned at 84–87°C for 3+ minutes before the EC storm started. No disk I/O — pure CPU workload. The system76 WMI stack was escalating fan control SCMD calls to try to manage thermals, and the EC firmware couldn't keep up.

The sudden CPU temp drop at T+1min (86°C → 65°C) is consistent with thermal throttling kicking in at the same time as the EC storm, rather than the system actually cooling — the EC was already failing to respond when the throttle happened.

### Connection to earlier comments

This matches the pattern described by @serioustable ("typically happened under some sort of heavy load, such as compiling Unreal Engine 4 or running No Man's Sky") — the common thread is sustained high CPU/GPU thermal load. The EC gets overwhelmed managing fan speeds via `SCMD 0x67` at high temps and stops responding to ACPI.

### Re: `acpi.aml_debug_output=1`

I'll add this kernel parameter now and report back with AML debug output on the next occurrence. Given this has happened twice today it should reproduce soon.