system76/firmware-open #189: oryp7 GPU temperature reports incorrectly

jackpot51, 2021-03-26:
As reported in https://www.reddit.com/r/System76/comments/mdexdx/hybrid_graphics_no_spinning_gpu_fan/

I have confirmed that if I toggle the GPU power on and off repeatedly, sometimes the GPU gets into a state where it reports back 0 temperature when it is still on. This only appears to happen on the oryp7, not the previous oryp6 model.

jackpot51, 2021-03-26:
The easiest way to reproduce this:

- Launch Steam
- Close the Steam window
- Wait for GPU temp to go to zero
- Open Steam Library, and potentially launch a game

lucserre, 2021-04-08:
This may or may not also be a problem, but I think even in the 'integrated graphics' mode, the right side fan doesn't turn on at all. Since the heat pipes are linked together on both CPU and GPU, I believe it might be a better idea to have both fans turn on at a lower speed rather than only one fan at a high (louder) speed.  Thanks for reading.

curiousercreative, 2021-05-01:
@lucserre yep, that update to fans should be coming soon: https://github.com/system76/ec/pull/177

MilesBHuff, 2021-05-07:
@curiousercreative [found](https://github.com/system76/ec/pull/190#issuecomment-834724959) that GPU temps are always broken in `compute` mode on a galp5.
I was able to [confirm](https://github.com/system76/ec/pull/190#issuecomment-834750085) this is also the case on an oryp7:
[image]

Raikiri, 2021-06-30:
May or may not be related, but LibreHardwareMonitor (windows) reports incorrectly the temperature of my second SSD. Both SSD's are almost identical (Samsung EVO 960 and 970, i think), but the second one always reports 0C.

crawfxrd, 2024-07-20:
Needs repro on latest release.