system76/firmware-open #196: AMD support

DemiMarie, 2021-04-15:
This could very well already be implemented, but if not, here is a feature request 🙂.

crawfxrd, 2021-04-15:
We don't have any AMD-based boards with coreboot support (ref system76/coreboot#53).

DemiMarie, 2021-04-15:
IIRC 3mdeb is working on this.

chuck-r, 2021-11-23:
Just got my pang11 in today and was disappointed that it's not Coreboot. Guess I didn't read the fine print well enough. One of my main reasons for purchasing a System76 laptop was because of core boot, so I'm a bit disappointed.

I've seen posts on Reddit that it might be in the works. Is this true? If so, I'd be willing to be a tester/contributer potentially -- I have the skills and tools to unbrick so a broken build doesn't scare me :)