system76/firmware-open #199: lemur10 last firmware upgrade broke sleep mode

alexgallotta, 2021-04-15:
I have a lemp10 lemur pro, I update the firmware few days ago.
After the reboot and firmware installation, every time I put the laptop in sleep mode while it is connected to a USB-C hub (plugged with a monitor and mouse+keyboard) the fan starts spinning continuously after 10 seconds or so.
When I resume and login, the fan stops.
I tried to run fwudp for update history, but got nothing
```
fwupdmgr get-history                                                                             1 ↵
No history
````

kepelrs, 2021-04-16:
I have the same issue on the lemp10.

curiousercreative, 2021-04-30:
Same issue on galp5 when connected to Dell U2720Q. Possibly related, system doesn't resume from hibernation properly anymore.

curiousercreative, 2021-04-30:
Likely culprint: https://github.com/system76/firmware-open/commit/236914e8e8f8b0acf3f14116e66ba04c7befc6e5

jackpot51, 2021-04-30:
Are these Thunderbolt hubs?

curiousercreative, 2021-04-30:
Notably, when the same display and cable are connected to my galp5 on the opposite side, suspend works as expected. On that side, the display doesn't receive a signal and there's no PD.

jackpot51, 2021-04-30:
Very interesting. I had noticed some issues with suspend when using Thunderbolt devices. Usually eGPUs. This was expected due to the Thunderbolt device itself keeping the CPU from going to s0ix.

But I haven't seen a non-Thunderbolt USB-C device keep the system from suspending. If you have any other USB-C devices please test those as well, so I can know if it is specific to some form of hardware. So far it looks like DisplayPort monitors are used on both @curiousercreative and @alexgallotta USB-C ports

jacobgkau, 2021-04-30:
I can recreate this when connected to the ViewSonic VP2771 via USB-C, and also when connected to a hard drive enclosure via USB-C (non-Thunderbolt).

jackpot51, 2021-04-30:
If updating the microcode caused the issue, @jacobgkau can you test with the old microcode firmware?

jackpot51, 2021-04-30:
Yeah, I don't think this is a microcode issue. Probably introduced in kernel updates

curiousercreative, 2021-04-30:
@jackpot51 I rolled back to 5.8 from the advice here, but no difference. https://www.reddit.com/r/System76/comments/mylbea/fan_keeps_spinning_when_suspend_lemur_pro/

curiousercreative, 2021-04-30:
note, uname -r shows kernel 5.8 while kernelstub -p shows 5.11...

curiousercreative, 2021-05-01:
I tried unplugging every USB peripheral from the monitor one by one until I was left with just the monitor and that still results in fans on suspend. Interestingly enough though, just having my USB headset plugged into the same TB3/USB4 port blocks suspend, but instead of blank screen and fans, suspend seems to exit immediately and return to lock screen. Here's some output from kern.log for the headset:
```
Apr 30 21:26:54 galp-curiouser kernel: [ 1280.642203] usb 3-6: new full-speed USB device number 25 using xhci_hcd
Apr 30 21:26:55 galp-curiouser kernel: [ 1280.865659] usb 3-6: New USB device found, idVendor=18d1, idProduct=5033, bcdDevice= 0.20
Apr 30 21:26:55 galp-curiouser kernel: [ 1280.865664] usb 3-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 30 21:26:55 galp-curiouser kernel: [ 1280.865665] usb 3-6: Product: Pixel USB-C earbuds
Apr 30 21:26:55 galp-curiouser kernel: [ 1280.865667] usb 3-6: Manufacturer: Google
Apr 30 21:26:55 galp-curiouser kernel: [ 1280.865668] usb 3-6: SerialNumber: xxxxx
Apr 30 21:26:55 galp-curiouser kernel: [ 1280.888300] input: Google Pixel USB-C earbuds as /devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.3/0003:18D1:5033.001A/input/input79
Apr 30 21:26:55 galp-curiouser kernel: [ 1280.946792] hid-generic 0003:18D1:5033.001A: input,hidraw2: USB HID v1.11 Device [Google Pixel USB-C earbuds] on usb-0000:00:14.0-6/input3
Apr 30 21:26:55 galp-curiouser kernel: [ 1281.083722] input: Google Pixel USB-C earbuds as /devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.3/0003:18D1:5033.001B/input/input80
Apr 30 21:26:55 galp-curiouser kernel: [ 1281.146856] hid-generic 0003:18D1:5033.001B: input,hidraw2: USB HID v1.11 Device [Google Pixel USB-C earbuds] on usb-0000:00:14.0-6/input3
Apr 30 21:26:59 galp-curiouser kernel: [ 1285.273126] wlp0s20f3: deauthenticating from 20:c0:47:e5:ac:aa by local choice (Reason: 3=DEAUTH_LEAVING)
Apr 30 21:27:04 galp-curiouser kernel: [ 1289.927632] rfkill: input handler enabled
Apr 30 21:27:04 galp-curiouser kernel: [ 1289.942331] PM: suspend entry (s2idle)
Apr 30 21:27:04 galp-curiouser kernel: [ 1290.425696] Filesystems sync: 0.483 seconds
Apr 30 21:27:09 galp-curiouser kernel: [ 1290.993058] Freezing user space processes ... (elapsed 0.002 seconds) done.
Apr 30 21:27:09 galp-curiouser kernel: [ 1290.995517] OOM killer disabled.
Apr 30 21:27:09 galp-curiouser kernel: [ 1290.995518] Freezing remaining freezable tasks ... (elapsed 0.024 seconds) done.
Apr 30 21:27:09 galp-curiouser kernel: [ 1291.019573] printk: Suspending console(s) (use no_console_suspend to debug)
Apr 30 21:27:09 galp-curiouser kernel: [ 1292.013717] ACPI: EC: interrupt blocked
Apr 30 21:27:09 galp-curiouser kernel: [ 1292.117996] ACPI: EC: interrupt unblocked
Apr 30 21:27:09 galp-curiouser kernel: [ 1292.674600] mei_me xxxxx: hbm: dma setup response: failure = 3 REJECTED
Apr 30 21:27:09 galp-curiouser kernel: [ 1293.224023] mei_hdcp xxxxxxx (ops i915_hdcp_component_ops [i915])
Apr 30 21:27:09 galp-curiouser kernel: [ 1295.023746] OOM killer enabled.
Apr 30 21:27:09 galp-curiouser kernel: [ 1295.023750] Restarting tasks ... done.
Apr 30 21:27:09 galp-curiouser kernel: [ 1295.039495] thermal thermal_zone1: failed to read out thermal zone (-61)
Apr 30 21:27:09 galp-curiouser kernel: [ 1295.122449] video LNXVIDEO:00: Restoring backlight state
Apr 30 21:27:09 galp-curiouser kernel: [ 1295.141777] PM: suspend exit
Apr 30 21:27:09 galp-curiouser kernel: [ 1295.394975] rfkill: input handler disabled
```

curiousercreative, 2021-05-01:
on kernel 5.8, suspend with USB-C display doesn't work, but it also doesn't spin the fans. 5.11 spins fans

alexrelis, 2021-05-01:
- Computer: galp5
- Kernel/OS: Linux alex-pc 5.10.0-6-amd64 #1 SMP Debian 5.10.28-1 (2021-04-09) x86_64 GNU/Linux

I might be having a similar issue to you folks. When I press "Fn + Suspend (moon icon)" to suspend, my system's screen will turn off as if it were attempting to enter s0ix but then the screen turns back on. After the suspend fails, the fan spins on constantly and a kworker process consumes a significant amount of CPU.

`    PID     USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
    1396345 root      20   0       0      0      0 R  78.4   0.0   2:18.99 kworker/0:3+kacpid`

I asked an expert and they told me to check my `/sys/firmware/acpi/interrupts/`.

`alex@alex-pc:~$  grep . -r /sys/firmware/acpi/interrupts/         
/sys/firmware/acpi/interrupts/gpe2F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe23:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe5D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe51:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe1F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe13:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe4D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe41:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe7B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe0F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe03:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe3D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe31:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe6B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe2D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe21:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe5B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe1D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/ff_pwr_btn:       0  EN     enabled      unmasked
/sys/firmware/acpi/interrupts/gpe78:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe11:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe4B:       0     STS invalid      unmasked
/sys/firmware/acpi/interrupts/gpe0D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe68:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe01:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe3B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe58:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe2B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/ff_rt_clk:       0         disabled     unmasked
/sys/firmware/acpi/interrupts/ff_pmtimer:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe48:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe1B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe76:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe38:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe0B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe66:       0     STS invalid      unmasked
/sys/firmware/acpi/interrupts/gpe28:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe56:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe18:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe46:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe74:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe08:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe36:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe64:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe26:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/error:       0
/sys/firmware/acpi/interrupts/gpe54:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe16:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/sci:134480049
/sys/firmware/acpi/interrupts/gpe44:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe7E:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe72:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe06:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe34:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe6E: 5774379  EN     enabled      unmasked
/sys/firmware/acpi/interrupts/gpe62:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe24:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe5E:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe52:       0         invalid      unmasked
[... 60 lines cut ...]
/sys/firmware/acpi/interrupts/sci_not: 1211254
/sys/firmware/acpi/interrupts/gpe35:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe6F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe63:       0         invalid      unmasked
[... cut ...]
/sys/firmware/acpi/interrupts/gpe61:       0  EN     enabled      unmasked`

He then told me to manually disable gpe69.

`root@alex-pc:~#  echo "disable" | tee /sys/firmware/acpi/interrupts/gpe69
disable`

This temporarily stops the excessive CPU usage, but it happens again when I attempt to suspend again. Not sure if this changes anything, but a Logitech wireless receiver is connected to the USB-A and my smartphone is connected to the USB-C Thunderbolt port. I hope that there's a way to fix this.

curiousercreative, 2021-05-01:
@alexrelis welcome and thanks for the detailed report! Does suspend work fine with no devices connected on a fresh boot? Does it work with just the USB-A device or just the USB-C device?

demiangz, 2021-05-01:
same exact issue with the lemp10. When connected to monitor through thunderbolt. kernel 5.11. Did happened in the previous one as well.

alexrelis, 2021-05-02:
@curiousercreative You're welcome! After a fresh boot, while charging and not charging, suspend works fine with nothing plugged in, everything plugged in, the smartphone via USB-C plugged and not the mouse via USB-A plugged in, and the mouse via USB-A plugged in and not the smartphone via USB-C plugged in, but there is a caveat! With my smartphone connected via the Thunderbolt USB-C port, it seems like the status light on the right of the machine doesn't blink like it normally does in s0ix. This doesn't happen on the non-Thunderbolt USB-C port. At the same time, I still believe the machine is suspending in some form, since the screen doesn't turn back on unless I press a key on the keyboard or the power button. So far, my computer has been on for 6 hours and I haven't been experiencing the issue. I have a Thunderbolt dock (this Ikling one: https://www.amazon.com/Adapter-Charging-Ethernet-Compatible-Thunderbolt/dp/B07TVSLDND/ref=sr_1_4?keywords=usb+c+hub&qid=1569187498&s=electronics&sr=1-4) that I will test to see if that's what's causing my issue when I get the chance. I'm curious to see what causes this.

atinybeardedman, 2021-05-05:
Having this issue with lemp9 as well. However I'm noticing it without devices connected. I shut the lid and the fans keep going and it's real hot. Started after I updated firmware as well.

alexgallotta, 2021-05-08:
Since it seems that it will take a while to fix, can we get the firmware version before 2021-04-07_236914e?
I see no tags in this repo, if you could tag the previous firmware it would be easier for us to flash it

curiousercreative, 2021-05-08:
@alexgallotta in your firmware update app, I believe you can lost previous firmware releases. While they're not tagged in this repo (probably because releases are device specific), you can `git checkout 236914e && git submodule update`  for the release that you mention. Basically, what comes after the date is a commit hash (abbreviated).

alexgallotta, 2021-05-09:
Thanks @curiousercreative for the explanation.
I don't see the previous firmware in the firmware manager UI
[image]
Following your indication I should checkout the commit `be04aea` for the previous version and if the issue continue I would try the original one `0e80285`.
I will try and report back if it solved our issue!

curiousercreative, 2021-05-09:
> Following your indication I should checkout the commit `be04aea` for the previous version and if the issue continue I would try the original one `0e80285`.
Yep, you got it.

alexgallotta, 2021-05-09:
I tried to flash the `be04aea` version, issue was still there. I also tried `0e80285`, same result.
I upgraded back to the current version(`236914e`), problem still there obviously. I noticed that, during the upgrade, there was something about "preserving old" something, so maybe there is some flipped bit that is kept the way it is now, no idea.
If anyone feels like trying the firmware downgrade too, it is pretty easy checking those commits and running the steps in the readme (I also `rm- rf build/` every time)

curiousercreative, 2021-05-09:
@alexgallotta perhaps you can attempt a kernel rollback. Do you have 5.8 installed still? Do you know how to set with kernelstub? -i and -k args

alexgallotta, 2021-05-09:
PopOS give the option to run the previous kernel (it was 5.8 when I tried) and I still had the issue.
Yes, it happens only when a USB-C is connected.
I played around yesterday and the behavior is:
- sleep laptop while USB-C is connected, fan not running -> fan spins after 10 seconds of sleep
- I unplug USB-C (laptop still sleeping) -> fan stops after 5-10 seconds
- I plug the USB-C again (laptop still sleeping)  -> fan spins again after 5-10 seconds
- I wake the laptop -> fan stops

demiangz, 2021-05-09:
> PopOS give the option to run the previous kernel (it was 5.8 when I tried) and I still had the issue.
> Yes, it happens only when a USB-C is connected.
> I played around yesterday and the behavior [...]

yes, exactly what I am experiencing

thomas-zimmerman, 2021-05-12:
@care has a customer that hit this as well. He has logs for:
```
A1: Started lemp10 with Sonnet box disconnected
A2: Hot-plug Sonnet box:  Neither the box or the eGPU are recognized, but the laptop does charge through the USB-C
A3: Hot-unplug Sonnet box, no apparent changes from lsusb.
A4: Tried to suspend laptop by closing screen lid. Power light stays solid green and if I open lid screen turns on but I cannot use the keyboard or click with mouse. Had to forcefully restart using the power button.

B1: Starting Lemur with Sonnet box already connected. System correctly recognized the box, the NVidia RTX8000 is set in compute mode, and the additional USB ports are available on the box. The laptop is also charging through USB-C.
B2: Hot-unplugging the Sonnet box correctly removed the items associated with the box from lsusb output.
B3: Tried to suspend laptop by closing lid. Power light stays solid green, but if I open screen lid I am able to type my password and unlock the computer to regain access to my gnome session.
```
Support case 59643

bryanspears, 2021-06-18:
I've recently encountered this with a new Darter Pro (darp7) using both Pop OS and Arch

Latest firmware (2021-04-07_236914e)
Latest kernel as well (5.11.0-7614-generic)

u-f-0, 2021-06-27:
I am encountering this as well on Lemur Pro (lemp10) running both Arch and Pop OS. 

Latest firmware (2021-04-07_236914e)
Arch kernel: 5.12.13-arch1-2

Like i have flashed both the latest and the previous firmware version with the same results. Please let me know if i can help out to get this resolved in any way!

julienlarena, 2021-07-04:
I don't know if it is the same issue, but I just experienced that on my lemp10: lid down, on AC power, the fan starts spinning with interruption and I can;t wake up from suspend, This is the first time it happens, but I usually only shuts the lid down when on battery. I said I was not sure if it was the same issue since it is not on USB-C but on the barrel power chord.

mvysin, 2021-07-09:
Hi. I'm experiencing this on lemp10 as well -- the behavior described [here](https://github.com/system76/firmware-open/issues/199#issuecomment-835848164) matches identically.

I'm running:
* 2021-04-07_236914e. I also tried rolling back the microcode update 236914e and had equally unhelpful results.
* Currently running Ubuntu 21.04 kernel 5.11.0-7620-generic, have also tried a 5.8 kernel, 5.12 kernel, and even older. Have also tried Pop OS. Same results.

I followed the [Linux S0ix Troubleshooting Guide](https://01.org/blogs/qwang59/2020/linux-s0ix-troubleshooting) and discovered that:
* booting the kernel with `initcall_debug` doesn't report any failures during the suspend or resume process.
* turbostat reports PC10 residency and graphics RC6 residency (can share the output if helpful)
* The counter at `/sys/kernel/debug/pmc_core/slp_s0_residency_usec` increments when in the 'bad suspend' state, which would seem to indicate the CPU reached S0ix
* The command `turbostat --show Pkg%pc2,Pkg%pc3,Pkg%pc6,Pkg%pc7,Pkg%pc8,Pkg%pc9,Pk%pc10,SYS%LPI rtcwake -m freeze -s 60` gives the following output:

    ```
    Pkg%pc2 Pkg%pc3 Pkg%pc6 Pkg%pc7 Pkg%pc8 Pkg%pc9 Pk%pc10 SYS%LPI
    1.10    0.86    0.00    0.00    0.00    0.00    96.75   0.00
    1.10    0.86    0.00    0.00    0.00    0.00    96.75   0.00
    ```
    which suggests good PC10 behavior (similar, I suppose, to the second bullet point).

Speculation:

Based on the LED behavior, it would appear that the EC is reading [C10_GATE_N](https://github.com/system76/ec/blob/master/src/board/system76/common/power.c#L567) high, yet that seems to contradict the `turbostat` output and s0ix counter. So I'm not sure if either that loop is not being reached (and the EC is busy elsewhere?) or if the pin is actually staying high. I may try to add debugging statements to get more information, but [neither debug option](https://github.com/system76/ec/blob/master/doc/debugging.md) is something I have available at the moment, so I'm hoping this can repro for someone who has this setup already.

mvysin, 2021-07-12:
On a hunch I replaced [`C10_GATE_N`](https://github.com/system76/ec/blob/master/src/board/system76/common/power.c#L609) in the LED loop with `SLP_S0_N` (and sped up the blink rate to verify my change worked at all), and it works -- I see the increased blink rate when I suspend -- both with and without USB-C peripherals attached.

Separately while playing with the fan curves, I noticed the sound from the fan behavior in this bug is consistent with 50% duty cycle [as set here](https://github.com/system76/ec/blob/master/src/board/system76/common/peci.c#L164) when the PECI fails. That code is also controlled by `C10_GATE_N`.

So I think I can refine what I'm seeing (and hopefully consistent with the OP) as: when USB-C monitor (+ power? I cannot test them in isolation yet)  is connected, `C10_GATE_N` remains inactive despite `SLP_S0_N` becoming active. When nothing is attached to USB-C then both signals become active and the EC works as I expect. In both cases it would appear that the cpu is entering sleep mode, but something about the USB-C connection is preventing the power signals controlled by C10_GATE from being removed.

hansmerkl, 2021-07-26:
I am seeing the same issue on my galp5. When the device goes into suspend and my USB-C hub is connected, the fan spins up after a few seconds. 

Adding logs as requested in https://github.com/system76/firmware-open/issues/222

sudo dmesg > dmesg.txt
sudo journalctl -r > journal.txt

[dmesg.txt](https://github.com/system76/firmware-open/files/6879579/dmesg.txt)
[journal.txt](https://github.com/system76/firmware-open/files/6879580/journal.txt)

hansmerkl, 2021-08-01:
Just wondering: Is there any hope for this to get fixed? It's a pretty annoying issue.

deifactor, 2021-09-03:
I am, perversely, experienceng this in a *different way*. Using a Plugabje UD-CA1a, starting with the dock attached, with no monitor plugged in:

1. I hit suspend. The laptop suspends (verified by looking at the `slp_r0_residency_usec` counter) but the LED stays solid. Fans aren't on.
2. Unplug the dock. The fans start spinning.
3. Plug the dock back in. Fans stop spinning.

If I have a monitor plugged into the HDMI output of the dock, then although the LED still stays solid, the fans never turn on even after unplugging and replugging.

One note: when the machine is docked and sleeping, I get a consistent clicking noise from the CPU which I assume is coil whine. I do get a small amount in normal use, but this is *much* louder and more consistent. The pattern is also different: "bad sleep" clickes about 4-5 times a second, but "good sleep" pauses occasionally.

... and just as I was typing to verify this, I did another docked-sleep test and got a blinking power LED, no clicking, and no fan noise even after unplugging and replugging. What.

hansmerkl, 2021-09-20:
One more observation: I booted Manjaro and Ubuntu from a USB drive and they showed the same behavior.

Seibz, 2021-10-20:
If Pop!_OS installs have been updated to the more recent 5.13 kernel, this issue may be resolved.

curiousercreative, 2021-10-20:
> One thing to keep in mind is that at least for my machine things worked fine with kernel 5.8
Have you tried to repro the issue now running 5.8? I recall my problems persisting.

>How can we convince System76 to take a look at this? I find this a pretty annoying issue which almost makes me regret buying the galp5.
Have you opened a support ticket? Maybe I'm in the minority here, but because I only have a dock at my in-laws and I'm affected by this bug so infrequently it's pretty back burner. If this was affecting me daily, I bet I would have been motivated to push the ball forward. In general, S76 engineers are busy, but maintaining activity on an issue and helping with reproduction steps or troubleshooting to better define and identify the source of the problem goes a long way.

crawfxrd, 2021-10-20:
It is the firmware. But I wasn't able to determine *what* was causing it.

What I saw was `C10_GATE#` constantly asserted/deasserted, which results in:

- Linux reporting some PC10 residency time
- Power LED stays green because it's less that 1s
- *Presumably*, CPU constantly being put in/out of C10 causes a temperature spike, which turns on the fan

If you want EC logs during suspend, you will need a Mega 2560 and set up [EC debugging](https://github.com/system76/ec/blob/master/doc/debugging.md#parallel-port).

hansmerkl, 2021-10-20:
> It is the firmware. But I wasn't able to determine _what_ was causing it.
>
> What I saw was `C10_GATE#` constantly asserted/deasserted, which results in:
>
> * Linux reporting some PC10 residency time
> * [...]

Do you work for System76? Sounds like this kind of issue needs a lab with proper equipment.

curiousercreative, 2021-10-21:
@crawfxrd I'm new to all this, but I've got a spare [rock64](https://www.pine64.org/devices/single-board-computers/rock64/) that I'd be keen to tinker with. I don't see that or similar SBCs listed in your linked documentation, but do you think it's a worthwhile exploration or a total waste of time and won't work for this purpose? Looks to have i2c pins similar to the documented Adafruit Trinket M0.

crawfxrd, 2021-11-09:
[CB:59024](https://review.coreboot.org/c/coreboot/+/59024)

hansmerkl, 2021-11-15:
> [CB:59024](https://review.coreboot.org/c/coreboot/+/59024)

I am not an expert on coreboot so I wonder if this is related to the issue of this thread?

curiousercreative, 2021-11-19:
Just received a launch keyboard (USB-C device) today and this bug reproduces with just the keyboard plugged in. I'm now on 21.04 with kernel 5.13.0-7614.

hansmerkl, 2021-11-19:
> Just received a launch keyboard (USB-C device) today and this bug reproduces with just the keyboard plugged in. I'm now on 21.04 with kernel 5.13.0-7614.

I am still curious what changed from 5.8 kernel. It worked fine back then.

nkalkhof, 2021-11-20:
I've cherry picked [CB:59024](https://review.coreboot.org/c/coreboot/+/59024) along with [CB:58798](https://review.coreboot.org/c/coreboot/+/58798) and build coreboot based on current system76-branch for a lemp10. Suspend now works (cpu enters pc10) after a lenovo thunderbolt 3 dock is connected. After disconnecting the laptop is stuck at pc8 again! Reconnecting the dock doesn't bring s0ix back. This behaviour is consistant strating from kernel 5.12.

redoxeon, 2022-02-11:
@nkalkhof 
> I've cherry picked [CB:59024](https://review.coreboot.org/c/coreboot/+/59024) along with [CB:58798](https://review.coreboot.org/c/coreboot/+/58798) and build coreboot based on current [...]

I think this is what I'm experiencing, but I left the comment on the wrong issue. [here's the comment](https://github.com/system76/firmware-open/issues/282#issuecomment-1036501478)

Haven't connected my eGPU since I did this, but haven't been able to suspend since.

hansmerkl, 2022-03-26:
Any news on this? Is there hope that this will get fixed one day?

cement-head, 2022-05-17:
Yeah, okay same issue here on a galp5 with a fresh install of Pop_OS 22.04 5.17 kernel.  Must be related to the firmware issue(s).  Apparently on the other issue, there's a firmware update.

hansmerkl, 2022-05-31:
Seems with some of the latest updates the fan doesn't turn on anymore but the laptop still does't suspend properly. The laptop still draws around 0.2A when it goes to sleep.

epistax, 2022-06-01:
I'm seeing this issue on a Lemp10 running 22.04 / 2021-07-20_93c2809. In my case, my USB-C device is a PD charger. I see the issue only in conjunction with a bluetooth speaker. I posted my details here: [reddit](https://www.reddit.com/r/System76/comments/v23riq/hint_about_the_recent_firmware_fan_issue/)

thomas-zimmerman, 2022-06-01:
> I'm seeing this issue on a Lemp10 running 22.04 / 2021-07-20_93c2809. In my case, my USB-C device is a PD charger. I see the issue only in conjunction with a bluetooth speaker. I posted my details [...]

There are a couple fixes [pending](https://github.com/pop-os/linux/pull/140) that will land in a kernel update along with a new bluez release. In testing, active bluetooth connections will still block suspend unless [additional configuration changes are made](https://unix.stackexchange.com/questions/539762/how-do-i-suspend-sleep-while-bluetooth-is-active)

TobiasKaiser, 2022-06-19:
I am experiencing the same problem on a lemp10 with a 5.10.0 kernel and a 2022-04-26_650b984 firmware image:
When connected to a USB C dock (03f0:046b HP, Inc HP USB-C Dock G5), the fan turns on and stays on after entering suspend mode. This is quite annoying for me. Strange to see that there is no fix given that the problem is known for over a year now.

alexrelis, 2022-06-19:
System76 firmware team, is there an update on this issue? I BEG of you to patch this bug as it severely impacts my work as I need to use a docking station for multi-monitors, keyboard/mouse, microphone, and webcam. I don't know if these laptops should be advertised as Thunderbolt capable in its current state.

hansmerkl, 2022-06-19:
>

> System76 firmware team, is there an update on this issue? I BEG of you to patch this bug as it severely impacts my work as I need to use a docking station for multi-monitors, keyboard/mouse, [...]

Totally agree. This situation is ridiculous. It's pretty bad that such obvious flaws aren't being addressed.

cement-head, 2022-06-20:
Kernel 5.18 fixes a lot of issues...

curiousercreative, 2022-06-20:
> Kernel 5.18 fixes a lot of issues...

This is not a kernel issue, it's firmware.

nkalkhof, 2022-06-20:
> > Kernel 5.18 fixes a lot of issues...
> 
> This is not a kernel issue, it's firmware.

I totally agree. After more than one year the issue still has not being properly addressed. Support Tickets remain unanswered. Thunderbolt 4 for example does not work at all unless you use a patched kernel from pop!os. Very disappointing for devices in that price range!

curiousercreative, 2022-06-20:
> Thunderbolt 4 for example does not work at all unless you use a patched kernel from pop!os

Are you saying that Thunderbolt 4 doesn't work on other distros with these S76 laptops or are you saying that you are running a patched kernel on Pop!_OS to make TB4 work? TB4 works on my galp5 on Pop! 22.04 aside from this bug.

hansmerkl, 2022-06-20:
> I totally agree. After more than one year the issue still has not being properly addressed. Support Tickets remain unanswered. Thunderbolt 4 for example does not work at all unless you use a patched [...]

What patched kernel? TB seems to be working for me except the suspend/fan issues.

[17 more replies]

closed 2023-01-17 by crawfxrd